arendst/Tasmota #521: FYI: Sonoff S85 Wall Socket

corani, 2017-06-14:
While browsing Iteads Taobao shop, I came across a new device that might be of interest: the [Sonoff S85 Wall Socket](https://world.taobao.com/item/530530736067.htm).

I couldn't find it on their international page, though there is a [placeholder](http://sonoff.itead.cc/en/104-products/227-s85-wall-socket). 

[image]

[image]

Some marketing specs:
- Working voltage: AC 90-250V
- Working frequency: 50Hz
- Rated power: 10A/2200W
- Own power consumption: <0.2W
- Working environment: -20c - 60c,Humidity<80%
- App: ewelink
- Remote distance: infinity
- Size: 86*86*30mm

I couldn't find any information about chipset and flashability. The Taobao price is CNY 29.90 ($4.40) but international it'll probably be a bit more.

ferbulous, 2017-06-15:
Have you tried asking itead support about the chipset info?

kruebsam, 2017-06-15:
Already listed on BangGood: https://www.banggood.com/de/SONOFF-S85-10A-2200W-86-Type-Wireless-Remote-Intelligent-Remote-Control-Wall-Socket-p-1142283.html

1043717432, 2017-06-20:
Does this support write sonoff firmware?

corani, 2017-06-21:
> Does this support write sonoff firmware?

It might, but until we get confirmation about the chipset that's pure speculation.

I've tried reaching out to itead via various channels, but haven't heard back from them yet.

1043717432, 2017-06-21:
ESP8285

corani, 2017-06-21:
If it's ESP8285 it should probably work. Did you get this from Itead? Or is it a guess? ;)

corani, 2017-07-01:
Just got a response from itead: ESP8266 + 4M Flash

1043717432, 2017-07-01:
Blessings, victories! Expect!

tlc76, 2018-06-12:
Hi, are there any updates here? I've just received two S85 devices but I wouldn't want to brick them :)

@rflx18 : I can dump the original firmware and post it somewhere so that you can use your device with the crappy ewelink app - probably it would be better than nothing...

tlc76, 2018-06-14:
I have successfully installed the Tasmota firmware on the S85 socket. The memory is 1M not 4M as stated by Itead.
Settings: 1M no spiffs, 115200 serial speed and the rest are identical to the ones mentioned by Theo in the Tasmota Wiki.

The S85 smart socket behaves EXACTLY as the Sonoff Basic device.

closed 2018-08-13 by stale[bot]