arendst/Tasmota #526: Wrong PowerState at startup

ionciubotaru, 2017-06-18:
I set `PowerOnState  0` then manually turn the relay on.
After that preform software restart via MQTT, or web interface
After reboot the relay in physical ON, but the software return status OFF

The same problem is on  `PowerOnState  1`

@arendst please perform a `digitalRead()` after software restart and set the power state accordingly.
It is a good behavior that the relay state remain unchanged after software restart

arendst, 2017-06-19:
Please provide values fro savadata and savestate and full logging of the power on situation. What do you mean by "software return status OFF"? Logging may provide more info.

ionciubotaru, 2017-06-19:
Normal start

```

00:00:00 APP: Project sonoff_test Sonoff (Topic sonoff_test, Fallback DVES_0CD6BD, GroupTopic sonoffs) Version 5.1.7
00:00:00 Wifi: Connecting to AP2 Public in mode 11N as sonoff_test-5821...
00:00:04 Wifi: Connected
00:00:04 HTTP: Webserver active on sonoff_test-5821 with IP address 192.168.1.109
00:00:06 MQTT: Attempting connection...
16:46:04 MQTT: Connected
16:46:04 MQTT: tele/sonoff_test/LWT = Online (retained)
16:46:04 MQTT: cmnd/sonoff_test/POWER = 
16:46:04 MQTT: tele/sonoff_test/INFO1 = {"Module":"Sonoff Basic", "Version":"5.1.7", "FallbackTopic":"DVES_0CD6BD", "GroupTopic":"sonoffs"}
16:46:04 MQTT: tele/sonoff_test/INFO2 = {"WebserverMode":"Admin", "Hostname":"sonoff_test-5821", "IPaddress":"192.168.1.109"}
16:46:04 MQTT: tele/sonoff_test/INFO3 = {"Started":"Software/System restart"}
16:46:05 MQTT: stat/sonoff_test/RESULT = {"POWER":"OFF"}
16:46:05 MQTT: stat/sonoff_test/POWER = OFF (retained)

```

Now I press the button on GPIO0, then send restart command via serial console

```

16:46:11 MQTT: stat/sonoff_test/RESULT = {"POWER":"ON"}
16:46:11 MQTT: stat/sonoff_test/POWER = ON (retained)
16:46:30 CMND: restart 1
16:46:30 MQTT: stat/sonoff_test/RESULT = {"Restart":"Restarting"}
16:46:32 APP: Restarting

 ets Jan  8 2013,rst cause:1, boot mode:(3,7)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v09826c6d
~ld
…
00:00:00 APP: Project sonoff_test Sonoff (Topic sonoff_test, Fallback DVES_0CD6BD, GroupTopic sonoffs) Version 5.1.7
00:00:00 Wifi: Connecting to AP2 Public in mode 11N as sonoff_test-5821...
00:00:08 Wifi: Connected
00:00:08 HTTP: Webserver active on sonoff_test-5821 with IP address 192.168.1.109
00:00:10 MQTT: Attempting connection...
00:00:10 MQTT: Connected
00:00:10 MQTT: tele/sonoff_test/LWT = Online (retained)
00:00:10 MQTT: cmnd/sonoff_test/POWER = 
00:00:11 MQTT: tele/sonoff_test/INFO1 = {"Module":"Sonoff Basic", "Version":"5.1.7", "FallbackTopic":"DVES_0CD6BD", "GroupTopic":"sonoffs"}
00:00:11 MQTT: tele/sonoff_test/INFO2 = {"WebserverMode":"Admin", "Hostname":"sonoff_test-5821", "IPaddress":"192.168.1.109"}
00:00:11 MQTT: tele/sonoff_test/INFO3 = {"Started":"Software/System restart"}
00:00:12 MQTT: stat/sonoff_test/RESULT = {"POWER":"OFF"}
00:00:12 MQTT: stat/sonoff_test/POWER = OFF (retained)
16:46:51 MQTT: tele/sonoff_test/STATE = {"Time":"2017-06-19T16:46:51", "Uptime":0, "Vcc":3.016, "POWER":"OFF", "Wifi":{"AP":2, "SSID":"Public", "RSSI":74, "APMac":"30:B5:C2:61:15:B0"}}

```
MQTT publish POWER = OFF, on the web interface relay state is OFF, but the relay is phisically ON, and the led connected to relay is ON

```
16:52:31 CMND: savedata
16:52:31 MQTT: stat/sonoff_test/RESULT = {"SaveData":"ON"}
16:53:18 CMND: SetOption0    
16:53:18 MQTT: stat/sonoff_test/RESULT = {"SetOption0":"OFF"}
17:00:11 CMND: status 0
17:00:11 MQTT: stat/sonoff_test/STATUS = {"Status":{"Module":1, "FriendlyName":"Sonoff", "Topic":"sonoff_test", "ButtonTopic":"0", "Power":0, "PowerOnState":3, "LedState":1, "SaveData":1, "SaveState":0, "ButtonRetain":0, "PowerRetain":1}}
17:00:11 MQTT: stat/sonoff_test/STATUS1 = {"StatusPRM":{"Baudrate":115200, "GroupTopic":"sonoffs", "OtaUrl":"http://www.cometa.ro:80/vlaicu/s.ino.bin", "Uptime":0, "Sleep":0, "BootCount":7, "SaveCount":18}}
17:00:11 MQTT: stat/sonoff_test/STATUS2 = {"StatusFWR":{"Program":"5.1.7", "BuildDateTime":"2017-06-18T14:38:07", "Boot":4, "Core":"2_3_0", "SDK":"1.5.3(aec24ac9)"}}
17:00:11 MQTT: stat/sonoff_test/STATUS3 = {"StatusLOG":{"Seriallog":2, "Weblog":0, "Syslog":0, "LogHost":"domus1", "SSId1":"Clicknet-BB4A", "SSId2":"Public", "TelePeriod":300}}
17:00:11 MQTT: stat/sonoff_test/STATUS4 = {"StatusMEM":{"ProgramSize":421, "Free":580, "Heap":32, "ProgramFlashSize":1024, "FlashSize":1024, "FlashMode":0}}
17:00:11 MQTT: stat/sonoff_test/STATUS5 = {"StatusNET":{"Host":"sonoff_test-5821", "IP":"192.168.1.109", "Gateway":"192.168.1.1", "Subnetmask":"255.255.255.0", "DNSServer":"192.168.1.1", "Mac":"5C:CF:7F:0C:D6:BD", "Webserver":2, "WifiConfig":4}}
17:00:11 MQTT: stat/sonoff_test/STATUS6 = {"StatusMQT":{"Host":"m21.cloudmqtt.com", "Port":16237, "ClientMask":"DVES_%06X", "Client":"DVES_0CD6BD", "User":"ionut", "MAX_PACKET_SIZE":512, "KEEPALIVE":15}}
17:00:12 MQTT: stat/sonoff_test/STATUS7 = {"StatusTIM":{"UTC":"Mon Jun 19 15:00:11 2017", "Local":"Mon Jun 19 17:00:11 2017", "StartDST":"Sun Mar 26 02:00:00 2017", "EndDST":"Sun Oct 29 03:00:00 2017", "Timezone":2}}
17:00:12 MQTT: stat/sonoff_test/STATUS10 = {"StatusSNS":{"Time":"2017-06-19T17:00:12"}}
17:00:12 MQTT: stat/sonoff_test/STATUS11 = {"StatusSTS":{"Time":"2017-06-19T17:00:12", "Uptime":0, "Vcc":3.015, "POWER":"OFF", "Wifi":{"AP":2, "SSID":"Public", "RSSI":66, "APMac":"30:B5:C2:61:15:B0"}}}
```

arendst, 2017-06-19:
Verified. Will look into it...

ionciubotaru, 2017-06-19:
Great !
Please do not change this behavior, just update the relay state according to pin state.
Sometimes sonoff restarts after wifi loss or mqtt loss and it will be unpleasant if the light turns off and nobody understand what's happened.

arendst, 2017-06-22:
There are several items at play here and I do not like the way it works now...
1. With ``SetOption0 1`` the relay state is under control of Tasmota and after reboot (either planned or due to not having configured ``WifiConfig 4`` and a wifi error occurs) or power on the relay is returned to the last known state.
2. with ``SetOption0 0`` set the relay state is not saved to flash and Tasmota does not know what the relay state is after a reset or power on. The user is expected to set the relay in a known state. That currently the relay stays ON or OFF is a feature I cannot support.
3. with ``powerretain 1`` set the home automation tool is expected to provide the correct relay state when tasmota is reset or after power on.

I can only conclude that I cannot support your use case.

ionciubotaru, 2017-06-23:
I suggest this code to run in setup:

```
if ( SetOption0==0 ) {
  pinMode(relayPin, INPUT);
  relayState=digitalRead(relayPin);
  pinMode(relayPin, OUTPUT);
}

```

Sunrise17, 2017-07-10:
Dear @arendst,
During the AP mode (192.168.4.1) of Sonoff, i want to configure following (5) settings below. Unfortunately, after i have updated some of them, sonoff is restarting and leave from AP mode and connect my other wifi SSID. How can i avoid restarting during my configuration??
Please note that i have function to push all these links to the Sonoff in one time, but in due to restarting issue, sometimes i can succeed to configure three of them, sometimes two of them randomly.

http://192.168.4.1/cm?cmnd=friendlyname%20Ceiling%20Light
http://192.168.4.1/cm?cmnd=grouptopic%20Kitchen
http://192.168.4.1/cm?cmnd=ssid2%20AsusRouter
http://192.168.4.1/cm?cmnd=password2%20x5x5x5x
http://192.168.4.1/cm?cmnd=mqtthost%20192.168.1.4

i have tried to configure above items over MQTT but in AP mode (192.168.4.1), i could not succeed to establish connection over MQTT to Sonoff.

Sunrise17, 2017-07-10:
it will be fine to have setting to restart Sonoff one time after configure all.

davidelang, 2017-07-10:
see the troubleshooting page on the wiki for how to reset the config.

to set all the network settings with one restart, use the web page (there was 
another issue on this topic within the last week)

Sunrise17, 2017-07-10:
https://github.com/arendst/Sonoff-Tasmota/issues/593

Yes i saw following topic above and @arendst 's constructive reply about it. We are looking forward to see it in the further updates.

@arendst : I will consider JSON input in the future allowing more commands at once.
--

JSCSJSCS, 2017-08-14:
Restarting Sonoff Tasmota 5.2.2 wiht web interface resumes with Power state ON always.

I have just uploaded new Tasmota 5.2.2 to a new Sonoff. I am using Home Assistant as my Automation server, however I am not sure that is in play here. The problem I have is that after resetting the device with the Tasmota web interface, it comes back with power state ON even if it was OFF when the restart occurred and even if the Home Assistant last state was off.

After the restart, the web interface and Home Assistant are updated to show the ON state. I have changed many settings suggested here, but non seem to work. This is the current state:
With Tasmota user. config.h:

```
#define SAVE_DATA              1                 
#define SAVE_STATE             1        
#define MQTT_USE               1
#define MQTT_BUTTON_RETAIN     0                 
#define MQTT_POWER_RETAIN      1  
#define MQTT_SWITCH_RETAIN     1 
```
With Home Assistant configuration.yaml:

```
switch:
  - platform: mqtt
    name: "Fountain Pump"
    state_topic: "stat/pumpswitch/POWER"
    command_topic: "cmnd/pumpswitch/POWER"
    qos: 1
    payload_on: "ON"
    payload_off: "OFF"
    retain: true
```
My understanding of the above options, Tasmota is not only supposed to save the last state to flash (so as to recover it after a problem?), but also Home Assistant also has retain set, so I should be covered both ways.

davidelang, 2017-08-15:
please read the wiki page on poweronstate, it looks like your mqtt server is 
overriding what the tasmota firmware has saved.

ionciubotaru, 2017-08-29:
Fixed the poweronstate bug adding this code at the end of `setup`

`  for (byte i = 0; i < Maxdevice; i++) if (digitalRead(pin[GPIO_REL1 +i])) bitSet(power,i);`

ionciubotaru, 2017-08-29:
One small fix to activate the pulse_timer:
```
  // Issue #526
  for (byte i = 0; i < Maxdevice; i++) {
    if ((pin[GPIO_REL1 +i] < 99) && (digitalRead(pin[GPIO_REL1 +i]))) {
      bitSet(power, i);
      pulse_timer[i] = sysCfg.pulsetime[i];
    }
  }
```

closed 2017-08-31 by ionciubotaru

stefanbode, 2017-11-29:
As far as I understood the problem, this only happens with Setoption 0. Therefore is is really required to put the digitalRead FOR loop in a IF clause like suggested above. @arendst the current FOR loop can cause issues depending on the wireing. Additionally the STATUS Is not completely updated. The Webinterface still shows the state at boot and not the change caused by the digitalRead.