arendst/Tasmota #527: MQTT teleperiod cannot be set to less than 10

KillSwitchSIG, 2017-06-20:
Using a sonoffpow, and trying to set the teleperiod to 5 using the console. However it's setting it to 10 instead.

18:09:56 CMND: teleperiod
18:09:56 MQTT: stat/sonoffpow/RESULT = {"TelePeriod":"300"}
18:10:08 CMND: teleperiod 5
18:10:08 MQTT: stat/sonoffpow/RESULT = {"TelePeriod":"10"}
18:10:09 MQTT: tele/sonoffpow/STATE = {"Time":"2017-06-20T18:10:09", "Uptime":2, "Vcc":3.158, "POWER":"ON", "Wifi":{"AP":1, "SSID":"phoenix3", "RSSI":48, "APMac":"40:16:7E:58:75:C8"}}
18:10:09 MQTT: tele/sonoffpow/ENERGY = {"Time":"2017-06-20T18:10:09", "Total":0.907, "Yesterday":0.000, "Today":0.907, "Period":45, "Power":870, "Factor":1.00, "Voltage":228, "Current":3.739}
18:10:17 CMND: teleperiod
18:10:17 MQTT: stat/sonoffpow/RESULT = {"TelePeriod":"10"}
18:10:19 MQTT: tele/sonoffpow/STATE = {"Time":"2017-06-20T18:10:19", "Uptime":2, "Vcc":3.159, "POWER":"ON", "Wifi":{"AP":1, "SSID":"phoenix3", "RSSI":48, "APMac":"40:16:7E:58:75:C8"}}
18:10:19 MQTT: tele/sonoffpow/ENERGY = {"Time":"2017-06-20T18:10:19", "Total":0.909, "Yesterday":0.000, "Today":0.909, "Period":2, "Power":801, "Factor":1.00, "Voltage":227, "Current":3.455}
18:10:24 CMND: teleperiod 5
18:10:24 MQTT: stat/sonoffpow/RESULT = {"TelePeriod":"10"}

arendst, 2017-06-20:
As designed.

davidelang, 2017-06-20:
It used to be that you could set it very low (I set it to 1 one time), but if 
it's set too low, the system becomes unusable as it spends all it's time trying 
to send log data and can't do anything else (including responding to commands)

on some configs, I've gotten away with 5, but 10 is probably a safe minimum.

KillSwitchSIG, 2017-06-30:
ok i'm going to close it. I think I saw the wiki saying from 2 onwards that's why i thought it's a bug.

closed 2017-06-30 by KillSwitchSIG

arendst, 2017-06-30:
Updated wiki with correct value of 10. Thnx.