arendst/Tasmota #530: Power state returns blank

initrd, 2017-06-21:
Power status returns a blank string:

> 19:41:22 CMND: stat/sonoff1/POWER
> 19:41:22 MQTT: stat/sonoff1/RESULT = {"POWER":""}
> 19:41:22 MQTT: stat/sonoff1/POWER =  (retained)

When I power on using the web UI, result shows "POWER": "POWER"

> 19:44:40 CMND: stat/sonoff1/power
> 19:44:40 MQTT: stat/sonoff1/RESULT = {"POWER":"POWER"}
> 19:44:40 MQTT: stat/sonoff1/POWER = POWER (retained)

Relevant config:

> sonoff/user_config.h:#define MQTT_STATUS_OFF        "OFF"
> sonoff/user_config.h:#define MQTT_STATUS_ON         "ON"

> sonoff/user_config_override.h:#ifdef MQTT_POWER_RETAIN
> sonoff/user_config_override.h:#undef MQTT_POWER_RETAIN
> sonoff/user_config_override.h:#endif
> sonoff/user_config_override.h:#define MQTT_POWER_RETAIN    1

Observed after upgrading from 4.x to 5.x. Currently on 5.1.6. Is there some setting that's missing?

arendst, 2017-06-21:
Try a reset 1 to re-initialize settings.

initrd, 2017-06-22:
That worked perfect, thanks!

closed 2017-06-22 by initrd