arendst/Tasmota #531: Errror on compilation

sujitrp, 2017-06-21:
xdrv_ir_send:46: error: 'IRsend' does not name a type

 IRsend *irsend = NULL;

Removed all RobotIRSend lib still getting this error 

using Widows 7 
IDE :1.8.2
Board :armtronix NodMCU 4 relay board

davidelang, 2017-06-21:
it sounds as if you are missing the IR library on your system.
k

sujitrp, 2017-06-21:
this is the error 

Arduino: 1.8.2 (Windows 7), Board: "Generic ESP8266 Module, 80 MHz, 40MHz, DIO, 115200, 1M (64K SPIFFS), ck, Disabled, None"

xdrv_ir_send:46: error: 'IRsend' does not name a type

 IRsend *irsend = NULL;

 ^

D:\OS\Raspberrypi3\Sonoff\Sonoff-Tasmota\Sonoff-Tasmota-5.2.0\sonoff1vwithirerror\xdrv_ir_send.ino: In function 'void ir_send_init()':

xdrv_ir_send:50: error: 'irsend' was not declared in this scope

   irsend = new IRsend(pin[GPIO_IRSEND]);  // an IR led is at GPIO_IRSEND

   ^

xdrv_ir_send:50: error: expected type-specifier before 'IRsend'

   irsend = new IRsend(pin[GPIO_IRSEND]);  // an IR led is at GPIO_IRSEND

                ^

xdrv_ir_send:50: error: expected ';' before 'IRsend'

D:\OS\Raspberrypi3\Sonoff\Sonoff-Tasmota\Sonoff-Tasmota-5.2.0\sonoff1vwithirerror\xdrv_ir_send.ino: In function 'boolean ir_send_command(char*, uint16_t, char*, uint16_t, int16_t, char*, uint16_t)':

xdrv_ir_send:99: error: 'irsend' was not declared in this scope

           if      (!strcmp_P(protocol,PSTR("NEC")))     irsend->sendNEC(data, bits);

                                                         ^

xdrv_ir_send:100: error: 'irsend' was not declared in this scope

           else if (!strcmp_P(protocol,PSTR("SONY")))    irsend->sendSony(data, bits);

                                                         ^

xdrv_ir_send:101: error: 'irsend' was not declared in this scope

           else if (!strcmp_P(protocol,PSTR("RC5")))     irsend->sendRC5(data, bits);

                                                         ^

xdrv_ir_send:102: error: 'irsend' was not declared in this scope

           else if (!strcmp_P(protocol,PSTR("RC6")))     irsend->sendRC6(data, bits);

                                                         ^

xdrv_ir_send:103: error: 'irsend' was not declared in this scope

           else if (!strcmp_P(protocol,PSTR("DISH")))    irsend->sendDISH(data, bits);

                                                         ^

xdrv_ir_send:104: error: 'irsend' was not declared in this scope

           else if (!strcmp_P(protocol,PSTR("JVC")))     irsend->sendJVC(data, bits, 1);

                                                         ^

xdrv_ir_send:105: error: 'irsend' was not declared in this scope

           else if (!strcmp_P(protocol,PSTR("SAMSUNG"))) irsend->sendSAMSUNG(data, bits);

                                                         ^

exit status 1
'IRsend' does not name a type

Exception in thread "AWT-EventQueue-0" java.lang.IllegalStateException: cannot open system clipboard
	at sun.awt.windows.WClipboard.openClipboard(Native Method)
	at sun.awt.windows.WClipboard.setContentsNative(WClipboard.java:75)
	at sun.awt.datatransfer.SunClipboard.setContents(SunClipboard.java:106)
	at processing.app.EditorStatus.lambda$initialize$2(EditorStatus.java:365)
	at javax.swing.AbstractButton.fireActionPerformed(AbstractButton.java:2022)
	at javax.swing.AbstractButton$Handler.actionPerformed(AbstractButton.java:2348)
	at javax.swing.DefaultButtonModel.fireActionPerformed(DefaultButtonModel.java:402)
	at javax.swing.DefaultButtonModel.setPressed(DefaultButtonModel.java:259)
	at javax.swing.plaf.basic.BasicButtonListener.mouseReleased(BasicButtonListener.java:252)
	at java.awt.Component.processMouseEvent(Component.java:6533)
	at javax.swing.JComponent.processMouseEvent(JComponent.java:3324)
	at java.awt.Component.processEvent(Component.java:6298)
	at java.awt.Container.processEvent(Container.java:2236)
	at java.awt.Component.dispatchEventImpl(Component.java:4889)
	at java.awt.Container.dispatchEventImpl(Container.java:2294)
	at java.awt.Component.dispatchEvent(Component.java:4711)
	at java.awt.LightweightDispatcher.retargetMouseEvent(Container.java:4888)
	at java.awt.LightweightDispatcher.processMouseEvent(Container.java:4525)
	at java.awt.LightweightDispatcher.dispatchEvent(Container.java:4466)
	at java.awt.Container.dispatchEventImpl(Container.java:2280)
	at java.awt.Window.dispatchEventImpl(Window.java:2746)
	at java.awt.Component.dispatchEvent(Component.java:4711)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:758)
	at java.awt.EventQueue.access$500(EventQueue.java:97)
	at java.awt.EventQueue$3.run(EventQueue.java:709)
	at java.awt.EventQueue$3.run(EventQueue.java:703)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:76)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:86)
	at java.awt.EventQueue$4.run(EventQueue.java:731)
	at java.awt.EventQueue$4.run(EventQueue.java:729)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:76)
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:728)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:201)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

sujitrp, 2017-06-21:
\Arduino\libraries\IRremoteESP8266  this is only IR lib present in my PC

sujitrp, 2017-06-21:
Thanks issue solved I added IRremote & IRremoteESP8066 lib in lib folder

closed 2017-06-21 by sujitrp