arendst/Tasmota #532: Set wrong ip address

Jantokil, 2017-06-21:
Hi, newbie here. I made a mistake at flashing Tasmota on a basic Sonoff. Instead of leaving the Wifi ip adress at 0.0.0.0 I accidentally filled in my routers IP adress. So now when I turn on the Sonoff it conflicts with my router and vice versa. Is there a way to change the Sonoffs ip adress? Maybe with the USB to Serial adapeter, but I think I have to be on another network to avoid network conflicts.

davidelang, 2017-06-21:
yes, you can issue commands via the serial port to change the IP, but if it 
conflicts with your router, I would change the settings in your user_config.h 
and re-flash it (don't forget to change the CFG_HOLDER) variable as well)

kk

Jantokil, 2017-06-21:
Ok thanks! Will let you know if it worked. First I have to find a good workflow for flashing these Sonoffs a lot easier

closed 2018-05-08 by stale[bot]