arendst/Tasmota #533: Inverting ON/OFF logic

petrfaitl, 2017-06-21:
This pains me even to write, but I have just sourced a 4 relay board that uses LOW as ON and HIGH as OFF. And whilst I could just set PowerOnState to 1, so I don't flood my garden on reset, I think it would be great to have a setting in the user_config.h that deals with this situation.

I have found this in user_config `MQTT_STATUS_OFF        "OFF"` , but I don't really see anything different in the console, when I swap the values around.

Is it doable? Thanks.

arendst, 2017-06-21:
As it's not a Sonoff I suggest you select a wemos device and populate the relays yourself using the configuration webpage. By selecting option Relayx instead of RelayxI you should be able to control a straigth relay.

petrfaitl, 2017-06-21:
Thank you. Wasn't aware of that option.

closed 2018-05-08 by stale[bot]