arendst/Tasmota #535: Sonoff Touch - no respose after flashing

SebastianSienicki, 2017-06-21:
I'v just sucessfully flash sonoff touch with 5.1.7 version.
[image]

[image]
After reconnect there is no response on serial monitor.
I'v try to reflash it but i have no respose at all.
Any ideas?

SebastianSienicki, 2017-06-21:
I'v got it.
It was hardware problem - i had to re solder esp.

closed 2017-06-21 by SebastianSienicki

Jantokil, 2017-06-27:
I just flashed 3 Sonoffs (basic) in a row with succes but now I'm having the same issues. No response at all when connected to mains. How did you fix it exactly? I'm fairly new to this

SebastianSienicki, 2017-06-27:
First of all check if your soldering is fine, then check if you upladed correct Environment (for eg. basic is sonoff for TC is sonoff-touch-4ch and so on), then check if you listen on correct serial port with boud rate 115200. there is almost no possibility that 4 modules are broken one.
In one of my TC esp module had poor soldering i had to resolder it, then it start to work properly.

David-Aunta, 2020-04-11:
Hola SebastianSienicki ,,,podria tener un paso a paso de esto, no me esta dejando flashear