arendst/Tasmota #537: Configuration Seconds for On/Off Time delay

FabianHM, 2017-06-22:
I need a function, e.g. Homematic switch actor is present and I am very missing at sonoff.
The goal is to set the switching command by specifying seconds (within the configuration of the GUI) so that no further switching command is necessary if the IP connection is terminated.
Ideally, the value in seconds for the duration for which the device is to be switched on and automatically switched off, or the time duration after which a switching command is to be executed, would be ideal. 

Perhaps there is this function already and I have not found this, then a reference would be great.
Thanks,
Greetings Fabian

davidelang, 2017-06-22:
There is such a function, look for pulsetime 
https://github.com/arendst/Sonoff-Tasmota/wiki/Commands

FabianHM, 2017-06-23:
This works fine, Thank you!

closed 2017-06-23 by FabianHM