arendst/Tasmota #539: Electrodragon Issues - Code Uploads but no info on Serial Monitor or connect to Wifi

amitubale, 2017-06-22:
Hi - I am able to push to code to the elctrodragons( tried 2) but after update I do not get any info on the serial port. I am using the same user_config.h file I have used for more than a dozen sonoffs. 

Do i need to make any specific changed in the user_config.h file for the Electrodragons ?

After the update the 2 Red LED indicators stay lit and pressing button1 or 2 caused the blue(wifi?) to light up.

When googled the devices others show are the one with the pins in the middle of the board. the one i got has all pins right next to the mains input.
[image]

davidelang, 2017-06-22:
you should not need to do anything to get a serial console, but you will need to 
set the module type and define the GPIOs before the buttons and relays work 
properly

amitubale, 2017-06-23:
Hi David.  After the code push the device does not connect to wifi( the same code and wifi works on a sonoff). To troubleshoot I need the serial port data but nothing shows up.   So I am not even getting to the webpage stage to set the miodule. 
In short I don't have to modify the user conf file for  Electrodragon. 

Button 2 press at power up puts it in prof mode and the code is uploaded. But. No details on the serial port. I even tried button 1 press and power up and that one spews garbage via serial port. 

Someone please let me know how you have flashed the electrodragon.

davidelang, 2017-06-23:
garbage on the serial port usually means you have the wrong baud rate set.

amitubale, 2017-06-23:
the code uploads at 15200 speed and when monitoring i am using the same baud rate.
also checked electrodragon site for data sheet and instructions ans they too instruct to use 15200 - 2xxxx...but indicate 15200 is optimal. ...I got these elctrodragon..since i could am not able to micro solder on the sonoff and these have so many gpios exposed and a adc too (so i can do current sensing...i hope someone helps me out...

amitubale, 2017-06-23:
I was able to flash the espeasy code and able to see the serial port logs.
 looks like the new Sonoff-Tasmota does not work on these new electrodragon.
Sonoff-Tasmota - When i flash the 5.2.1 code both red leds on the board stay lit(could not find any info on the electrodraon support site or data sheet.
ESPeasy- With easpeasy flash the red led on the main board if off and the red lead on the power module is lit. - i hope this helps.

amitubale, 2017-06-23:
i am gogin to try version 5.1.0 sine someone had an issue(#449) and was able to resolve.
But it might also have been the older version of electrodragon hardware. will post my results soon.

amitubale, 2017-06-28:
anyone ahve the new electrodragon that looks like the picture above ? and let me know if you are able to flash the code ?
thanks

diplix, 2017-09-26:
i have a new electrodragon that looks like the one above and just flashed version 5.8.0 successfully. runs without any problems.

closed 2018-05-07 by stale[bot]