arendst/Tasmota #541: Switch2 for POWER1 and different Switchtopics

kisseler, 2017-06-23:
First of all Thanks, Theo, for your wonderful work!

I am trying for two days now to achieve the following procedure:
I use a Sonoff Basic module to switch a main light. Setting the GPIO14 to Switch1 allows to plug in a wall switch. That works fine. With GPIO3 I want to switch another Sonoff (S20) which is placed in the corner of the room. So I set GPIO3 to Switch2.

What worked at the beginning: I set switchtopic1 to the local Sonoff and switchtopic2 to the S20. That does not work anymore. As soon I set switchTopic2 some value, all switchTopic gain that value. I really don't know what is going wrong here, because at the beginning that worked fine.

And most important: What I was not able to achieve is: I would like the Switch at GPIO3 (Switch2) not to send the command "cmnd//POWER2:TOGGLE", because I receive a "command:unknown" message by the S20. Can I influence the message sent by Switch2 to be "cmnd//POWER:TOGGLE" (without number of the switch)?

Thank you very much for your help!

arendst, 2017-06-23:
I see what you're aiming for but it's impossible with Tasmota. I suggest to use a MQTT broker and configure aliases to do what you want.

I don't know either how you managed to have different SwitchTopic1 AND SwitchTopic2; there has always been but one switchtopic string and both commands would change the same string. Perhaps you used one ButtonTopic and one SwitchTopic?

kisseler, 2017-06-27:
Thanks! I was a bit confused by [this ](https://github.com/arendst/Sonoff-Tasmota/issues/88) thread. And I guess I defined one switchtopic and tested, then the other and tested - but not the first again.

I'll try to realized my idea by configuring aliases. Although it would be nice to keep a "local" switch on the device, that switches independently frmo MQTT-Server.

Is there any chance, the feature of individually configure the MQTT-topic of switch1 to 4?

PS: Thanks for your help! How can I buy you a coffee or a beer?

fabricioncosta, 2017-07-20:
Hi Kisseler, 
How to connect another swtich on GPIO3? Is it necessary to disable the serial communication on the device? Is it necessary to solder pull up resistor?

Thanks,

Fabrício

kisseler, 2017-07-21:
Hi Fabrício,
that's exactly the way I did that. You can disable the serial communication under Configuration Menu > Configure Logging. Using the gpio I connected a pull-up resistor directly on the pins
[image]
[image]
Greetings

closed 2018-05-07 by stale[bot]