arendst/Tasmota #542: Failed OTA upgrade

dariosm, 2017-06-23:
After upgrading 4 sonoffs, with `5.2.1` version, just one of the would not start, keeping led light blinking for a while. I had to disassemble it from the mains (it was hooked to a ceiling lamp), and started it with a serial monitor:
```
ets Jan  8 2013,rst cause:1, boot mode:(3,3)

load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d
v09826c6d
~ld
�
00:00:00 Cnfg: Use defaults
00:00:00 APP: Project sonoff Sonoff (Topic sonoff, Fallback DVES_F99DEF, GroupTopic sonoffs) Version 5.2.1
00:00:00 Wifi: Connecting to AP1 indebuurt1 in mode 11N as sonoff-7663...
00:00:07 Wifi: Connect failed as AP cannot be reached
00:00:07 Wifi: Connecting to AP2 indebuurt2 in mode 11N as sonoff-7663...
00:00:14 Wifi: Connect failed as AP cannot be reached
00:00:14 WPSconfig: Active for 1 minute

```

It seems like sonoff lost the configuration after upgrade. I should say that I followed the two step upgrade suggested in the wiki, that is, first upgrade to minimal and then normal firmware.

Is there anything I could have done wrong?
Do you think I could solved this without disconnecting from the light installation? I'm afraid next time I upgrade the modules, this could happen again.

arendst, 2017-06-23:
What was the version level you upgraded from?

As you can see from the log it has defaulted the settings but started WPSconfig which allow you to press your wifi routers WPS button and your device should configure itself for your wifi lan. After that you still have to configure other parts of the device but at least you did not have to disconnect it from your installation.

dariosm, 2017-06-23:
@arendst sadly my router does not have that functionality (it is an ISP modem-router).
I think I upgraded from `5.1.7`, quite sure about that but not 100%. I'm certain, though, that the other sonoff modules where running the same version, and were upgraded just fine, keeping configs and restarting with no issues.

arendst, 2017-06-23:
Do you happen to know what the version was BEFORE 5.1.7...

From my 19 devices one of them also failed the upgrade. It just blinks the blue led every 20 seconds. I still have to look into it what went wrong there...

arendst, 2017-06-24:
I found the problem in my case and will solve it in next release. 

Were you using the provided release firmware or did you happen to compile your own using esp8266/arduino 2.4.0-rc1?

dariosm, 2017-06-24:
I downloaded from here: https://github.com/arendst/Sonoff-Tasmota/releases

---
Lic. Darío Silva Morán

hubecker, 2017-06-25:
Same for me, After upgrading sonoff pow from 5.2.0 to 5.2.1 version, it failed to restart.
I tried to flash it again with arduino IDE, no change.
In the serial monitor i see nothing.
What to do now?

dariosm, 2017-06-25:
I flashed it again through serial and it worked, just lost the config no
big deal.

---
Lic. Darío Silva Morán

arendst, 2017-06-26:
@hubecker erase flash using esptool and reload using ide.

chadbone, 2017-12-15:
I am having this issue as well.  I am trying to upload via PlatformIO.  I did change the user_config.h (perhaps the second time) but it is for sure set now with my ssid and password.

The serial monitor reflects that it is trying to connect to indebuurt1 and 2.  I do see in the serial monitor that the "Project" has been changed to my setting.  My version is reporting as 5.10.0a.  

Any ideas on how I can correct this?  THANKS! your work is awesome.