arendst/Tasmota #544: sonor sc change DHT 11 to DHT 22

johannesbonn, 2017-06-24:
Hello,

is it possible to change the builded in DHT 11 sensor to a DHT 22 sensor? The connection should be the same, but I don't know if the tasmota software can work with the new sensor.

Have a nice weekend!

Johannes

arendst, 2017-06-24:
As the DHT11 is connected to the Arduino processor and not the Tasmota controlled esp8266 processor you will have to write your own software for a connected DHT22.

johannesbonn, 2017-06-25:
Are there any plans to write a Arduino software for that?

arendst, 2017-06-25:
Nope, not by me but he did http://tinkerman.cat/sonoff-sc-with-mqtt-and-domoticz-support/

johannesbonn, 2017-06-25:
But according to the web configuration there should be the possibility to add a DHT22 to gpio2. I can t find gpio2 on the board. Do you know if there is a soldering point for that on the sonoff sc board?

felixekman, 2017-06-25:
What you can quite easily do is go to Itead's wiki and download the source for the ATMEGA328 and modify it from DHT11 to DHT22 that way you keep the same functions and can use the same Sonoff firmware for all SC's. But it would be ideal to have an ATMEGA328 firmware that support more sensors and LED rings combined with support for the amazing tasmota firmware on the ESP.

johannesbonn, 2017-06-25:
Ok thank you, the Problem is that I am no programmer. Is my idea adding dht22 to gpio2 possible?

felixekman, 2017-06-25:
So in simple terms, the DHT11/22 is right now connected within the Sonoff SC to the ATMEGA328 processor not the ESP8266 that we are programming with this firmware. As such you can't address the pin you're looking for within the firmware here. The reason you're seeing it as an option is because this is an universal firmware and you could technically solder a DHT22 directly to the ESP8266 if you find a spare pin on the PCB and in the schematic. Take a look here if you want to get down that road yourself: https://www.itead.cc/wiki/images/f/f7/Sonoff-SC-Schematic.pdf otherwise there isn't much you can do with the Tasmota firmware as it does not flash touch the MCU that is connected to the sensors.

felixekman, 2017-06-25:
If someone where to write a new compatible firmware for the ATMEGA328 it would logically have a serial command (that is how the two MCUs communicate with each other) that allowed the ESP to send similar to an AT+SETUP command to the ATMEGA328 to say like AT+SETUP = S2R1M0
Where for example S1 equals DHT11 and S2 equals DHT22, R0 for no RGB led ring and R1 to enable it, and maybe M0/M1 to add motion sensor support. Then you could theoretically kindly ask @arendst to add a configuration page/support for the SC within the tasmota firmware to enable configuration via the web interface. But that is quite a lot of work and would need someone familiar with serial communication between MCUs (which is sadly not me).

johannesbonn, 2017-06-25:
Am I right that gpio2 is pin sda on the board?

johannesbonn, 2017-06-25:
That's it:-), try and glory! Pin sda is gpio2..... great software.

closed 2017-06-26 by johannesbonn

langestefan, 2018-11-08:
> Nope, not by me but he did http://tinkerman.cat/sonoff-sc-with-mqtt-and-domoticz-support/

Would flashing the ATMega328P with this software and replacing the DHT11 with a DHT22 still be compatible with tasmota?