arendst/Tasmota #545: 8ch relayboard support with esp8266

BigOrangeClown, 2017-06-24:
tried to change the code for wemos but ended up on switching multiple channels via mqtt
i am not firm in programming so a native support 4 that would be really nice....

arendst, 2017-06-25:
8ch relay board.... Is there only one type around? Provide at least schematics and/or esp8266 pin layout.

mrmin24, 2017-06-27:
yes please. stefanbode did do a version for the I2C expander, but there are some issues with pulsetimers not working correctly. 

Basically just 8ch DIO pins required inverted or non-inverted

BigOrangeClown, 2017-06-27:
Hey there, sorry for the missing information!
basically i just need 8 DOUT pins on an NodeMcu based baord ( pinout: https://goo.gl/images/5gdKa2 ) for the 8 DIN Pins on the relayboard ( similiar to that board:  https://goo.gl/images/wdG7wU )

thankfully

arendst, 2017-06-27:
What's wrong with having two Sonoff 4CH devices already supported?

BigOrangeClown, 2017-06-27:
I want to minimize the  clients in network as far as possible, would be glad about 16 relay / network user but right now there are 5x8ch boards already in use ( with own pretty simple code / wether nice than reliable )- just took tasmota for some sonoff for single actors and i am really satisfied by the performance and all the informations that are delivered by the device - so just thought about would be awsome if i could get this to my installed boards - really don´t want to disassemble half of my e - installation for adding  some new devices....

levonbragg, 2017-07-12:
I too would like to see 8 channel available.
Currently I use one of these [http://linksprite.com/wiki/index.php5?title=LinkNode_R8:_Arduino-compatible_WiFi_relay_controller](url). with ESPeasy
There are issues with this board on reboot that cause 2 of the relays to be on during boot up, so I only use 6 of them.

neographikal, 2017-07-17:
Same here :) Using two 4ch modules isn't practical cost wise and implementation wise (space confinements)

nisof, 2018-01-17:
Here is support for the LinkNode R8  

[sonoff_template.h.txt](https://github.com/arendst/Sonoff-Tasmota/files/1640299/sonoff_template.h.txt)

closed 2018-05-31 by stale[bot]