arendst/Tasmota #546: Sonoff sc air quality unit

johannesbonn, 2017-06-25:
Hello,

I thought the unit for airquality has the unit mg/m^3. My sensor shows me a value of 90, but I can remember that other sensors show values (indoor) about 0,2 or something like that. Which unit shows tasmota?

Thank you!

arendst, 2017-06-25:
Considering it's a crappy sensor it only reports values in percentages in 10% steps. Were 100% would mean good air. Mine almost always shows 100%...

For more info see iTead Sonoff Sc website informaton:
[image]

johannesbonn, 2017-06-26:
Thank you for your help. To change the output (10% steps) I think I have to flash the ATMEGA328 firmware. I read that the dust sensors output is Voltage in relation to mg/m^3.
Is your esp software compatible to the  ATMEGA328 firmware from xoseperez (https://github.com/xoseperez/sonoffsc)? He changed some basic parameters for the dust sensor/DHT11 to DHT22/clap monitor and led ring. I love your software and the interface and a connection of both softwares would be awesome!

arendst, 2017-06-26:
I don't think it's compatible with xoseperez. If I have a lot of time I might take a look ;-)

In the meantime you might try to get hold of the changes he made to the data interchange between esp8266 and the Atmega.

johannesbonn, 2017-06-26:
That would be nice! Have a nice evening!

firelook, 2017-09-10:
What exactly does air quality sensor measure ? NO2 ? CO2 ? What are the ranges for good, moderate unhealthy ? By the way, what does sound quality means ? How many decibels are considered quiet, normal or noisy according to this equipment ?

arendst, 2017-09-10:
See http://sonoff.itead.cc/en/products/residential/sonoff-sc

akadlec, 2018-01-23:
Hello. One question. Does anybody know the ranges for light, sound and dust channels? I would like to show same info on my dashboard. Is it lienar? 33% of maximum value? or not. I tried to search it on itead page, but without any success.

closed 2018-06-07 by stale[bot]