arendst/Tasmota #547: TimeZone < -1 issue

dariosm, 2017-06-25:
It seems like setting timezone to a value lower than -1 is always -1.
```
17:27:14 CMND: timezone 1
17:27:14 MQTT: stat/sonoff5/RESULT = {"Timezone":1}
19:27:21 CMND: timezone -1
19:27:21 MQTT: stat/sonoff5/RESULT = {"Timezone":-1}
17:27:23 CMND: timezone -3
17:27:23 MQTT: stat/sonoff5/RESULT = {"Timezone":-1}
```

By looking at the code, I've found it correct:
```
    else if (!strcmp_P(type,PSTR("TIMEZONE"))) {
      if ((data_len > 0) && (((payload >= -12) && (payload <= 12)) || (99 == payload))) {
        sysCfg.timezone = payload;
      }
      printf("<payload: %d><svalue: %d>\n",payload,svalue);
      snprintf_P(svalue, sizeof(svalue), PSTR("{\"Timezone\":%d}"), sysCfg.timezone);
```
But I should say `payload` is always `-1` for every value lower than `0`, so maybe `payload` can't take negative values in this function.

arendst, 2017-06-25:
It's a regression caused by setting a wrong default value for payload. Change line 919 into:
```
    int16_t payload = -99;               // No payload
```
which should solve the situation.

dariosm, 2017-06-25:
Working

closed 2017-06-26 by arendst