arendst/Tasmota #548: hlw_margin_chk for protection

ChenChiang, 2017-06-26:
hi,

currently this `hlw_margin_chk` is called each time in `every_second` loop inside `sonoff` file. 
What I am thinking is that why not calling this `hlw_margin_chk` inside `hlw` file directly? And disable the `MaxPowerHold` time?
It needs to react faster.
1). Currently it's checked at 1s interval. But it doesn't need 1s to burn things for large current (e.g: a surge current). 
2). `MaxPowerHold` time is not necessary. When it reaches `MaxPower` it needs to be switched off immediately. No need to do a counter down. 
3). Each time when it gets the energy data from `hlw`, this `hlw_margin_chk` should be called. (e.g: every 200ms, when it does the energy accumulation, it should check it.)

I am just thinking that it needs to act immediately when it detects something wrong. So this `hlw_margin_chk` shall be called each time when it gets the energy data from `hlw`.

arendst, 2017-06-26:
There is no need to react faster as a single measurment of power, current and voltage can take up to three seconds with low loads on HLW hardware. You'll need different hardware than a Sonoff Pow if you want faster reaction times.

The concept of MaxPowerHold was implemented for situations where users were not allowed to draw more power then allowed. After that they would have a few chances to lower power before it would power off.

The software is not a replacement for a simple fuse. If things could burn because of overload a fuse should have been implemented long before you went in production. In fact there is a fuse in the Pow.

ChenChiang, 2017-06-26:
For small current, it's fine. It won't do any damage anyway. For for large current, software should react much faster than the fuse. So don't need to each time open the casing and replace it. That's what I was thinking about.
I still didn't really see the point of `MaxPowerHold`. The user will only notice the load is drawing too much power only after it's switched off. Then he can change. As there is a `MaxPowerWindow` and Retry No.

davidelang, 2017-06-26:
In this case, the sonoff Pow takes several seconds to measure the current that's 
going through it, so a fuse will be far faster.

As for the problem of the user noticing the power consumption, remember that 
every time the power is measured, it's also sent out to the home automation 
software, so 'noticing the power draw' is a matter of software, not a person.

closed 2018-05-07 by stale[bot]