arendst/Tasmota #549: How to setup  Sonoff 4 CH for Home Assistant?

DNA99, 2017-06-26:
Hi,
Really new to this and I managed to setup the sonoff basic with the home assistant but when I try to setup the sonoff 4CH, it does not seem to work. What do I use as topic to access the 4 different switches?

davidelang, 2017-06-26:
it's very similar to the basic, but instead of POWER, you have POWER1, POWER2, 
POWER3, POWER4

DNA99, 2017-06-26:
Thanks for quick reply!
  - platform: mqtt
    name: "TV Switch"
    state_topic: "stat/sonoff3/POWER1"
    command_topic: "cmnd/sonoff3/POWER"
    qos: 1
    payload_on: "ON"
    payload_off: "OFF"
    retain: true

So I need to have 4 sets of this but with POWER2, POWER3, POWER4?

davidelang, 2017-06-26:
yes, (note you have to change command_topic as well as state_topic)

DNA99, 2017-06-26:
Thank you so much! It works now! Just need to locate the GPIOs so I can connect them to wall switches.

pprotschka, 2017-07-16:
hey there, sorry to hijack the support threat. but i also have the 4ch and want to use it with HA. How did you get the 4ch into programming mode? i am at this for 2 days now and keep getting errors on upload. :/ warning: espcomm_sync failed
error: espcomm_open failed
error: espcomm_upload_mem failed

DNA99, 2017-07-17:
You need to use esp 8285 not 8266.

closed 2018-05-07 by stale[bot]

wb441, 2018-06-25:
@pprotschka 
Probably not relevant anymore, but there is a link on the Tasmota wiki where to pull to ground.
Worked for me with the chip set to ESP8266 tho :)