arendst/Tasmota #551: S20/basic works on 3.3V, can't connect network when on Mains power

MattThawne, 2017-06-27:
New to this game, bear with me i am no pro with this stuff. I am pretty sure i followed the wiki correctly in the IDE setup and config options.

I have a couple of S20 smart sockets and a bunch of sonoff basics all flashed ok.
I am able to connect to the S20 socket's web page when it is powered by 3.3V through the headers.
When powered by mains there is no network connectivity, and the LED flashes for a bit then the relay clicks.
That is happening on both S20 devices.
Most of the sonoff basics are working normally, but I do have one that is showing the same issue as the S20 sockets
The S20s and sonoffs were tested with default f/w and Ewelink app before flashing.

I am appealing to the community for any assistance

davidelang, 2017-06-27:
most of the time when we get reports like this, the problem is a bad solder 
connection in the mains -> 3.3v power supply on the device.
k

MattThawne, 2017-06-27:
i get 3.3V between the Vcc and GND header pins when powered up on mains. I couldn't see any obvious bad solder joints, though the components are pretty small and i don't have a good magnifier.
I am a little reluctant to poke around for voltages on the back of the board when it's powered up on mains.
I'll try touch up some solder joints and see.
Thanks

davidelang, 2017-06-27:
usually it's not a case of the voltage not being 3.3v, but rather that the 3.3v 
is very noisy. You will not see this noise on a multimeter, but would need a 
scope to see it.
k

MattThawne, 2017-06-28:
i put an extra filter cap between Vcc_bar and GND. fixed the issue.
Thanks for the advice

closed 2017-06-28 by MattThawne

MichaelBremer, 2018-09-19:
Hello MattThawne,
what exactly do You mean with "filter cap"? And what is "between Vcc_bar and GND"? The pins at the headers?