arendst/Tasmota #552: Enhancement: Rain flow sensor.

mrmin24, 2017-06-27:
Any chance there can be an option for a tipping bucket rain sensor. It just has a reed switch which toggles every x mm of rain. The x should be a user definable value. 

If it can then report rain for past hour / day  similar to the POW.

http://rebecca-ricks.com/http:/rebecca-ricks.com/categories/rain-gauge/

Thanks

arendst, 2017-06-27:
Why not use the counter option and calculate rain level yourself?

mrmin24, 2017-06-28:
MMMM, Did not even know about counter. Let me try it

mrmin24, 2017-06-30:
Hi,

The counter works nicely, but it is quite an effort to implement in the controller if you want the rain fall current, past 24 hours, week, etc.  I will close the ticket if you don't want to implement.

thanks

arendst, 2017-06-30:
As you noticed. it's quite an effort to implement in Tasmota too ;-)

closed 2017-06-30 by arendst

davidelang, 2017-06-30:
that sort of historical data should be maintained inthe home automation 
software, not in the firmware (remember, the firmware can be rebooted and loose 
track of anything it's counting)

mrmin24, 2017-06-30:
I suppose. I was hoping the Energy monitor code could be easily used to have a standard report on this, but thanks anyway

mpuff, 2020-05-25:
@

> The counter works nicely, but it is quite an effort to implement in the controller if you want the rain fall current, past 24 hours, week, etc. I will close the ticket if you don't want to implement.

Hello,

what settings do you use for the rain calculation?
i have also a tipping bucket rain sensor, tasmota shows me on and off when the bucket change, but o dont know how to calculate.