arendst/Tasmota #553: Group 2 or more sonoff touch switches (stairs, corridor)

tomekkepa, 2017-06-27:
Hi.
I would like to know if it is possible to group 2 or more switches to make something like in standard house when we talking about stairs solution (i'm on the ground floor, switching on light, going up to the 1st floor, switching off light or example 2: long corridor, switch on at the beggining to turn on the light, switch at the end to turn off and in the opposite way).
From what I see at the moment sonoff touch doesn't have so many input/output like standard stair/cross switches but if I install only one switch directly connected to bulb + other switch not connected to bulb but only attached to electricity to make it visible in network and I will pair/group them, Will it work? Or there is simpliest way?

davidelang, 2017-06-27:
do this in software in your home automation software. Have each of the switches 
send a message to your mqtt server and have your home automation software turn 
the light on if the two switches are the same state and off if they are 
different.

tomekkepa, 2017-06-27:
Could you tell me more about that?
I dont have yet home automation software but 99% i will choose homebridge + eventually home assistant.

tomekkepa, 2017-06-27:
If I understand well, i'm going to push phisically button on sonoff touch which send info to software which will send info to other switch to update status?

davidelang, 2017-06-27:
On Tue, 27 Jun 2017, tomekkepa wrote:

> If I understand well, i'm going to push phisically button on sonoff touch which send info to software which will send info to other switch to update status?

something like that.

by default, pushing the button on a sonoff switches the relay and sends a 
message, but you can configure it to just send a message instead.

once you have it configured to just send a message, then you can have that 
message generate any action that you want to have happen. How you do this is 
going to depend on what software you use. I don't use home assistant, I use 
openhab. Openhab has 'rules' that watch for actions and then make decisions. I 
assume home assistant has something similar.

tomekkepa, 2017-06-27:
What about delay? Are they acceptable, how many seconds between push and final effect?

davidelang, 2017-06-27:
that depends on your system, but with LED/CF lights there can be delays even 
when everything is hard-wired.

when you are using commercial cloud-based systems, you hit a button, the command 
goes out to the Internet, authenticates, is processed by the server, and then it 
sends a command to the device to turn on. There are far longer delays inherent 
to that sort of system than to a local system.

Try it.

ionciubotaru, 2017-06-28:
Below is a ruby script for your needs.
You can expand it to as many Sonoffs Touch as you need. When you press the button on any sonoff, all of them will TOGGLE.

```
#!/usr/bin/env ruby

require 'rubygems'
require 'mqtt'

last_loby2=''
last_loby1=''

loop do
        MQTT::Client.connect('localhost') do |c|
          c.get("#") do |topic,message|
                topic=topic.downcase
                if topic=="stat/t_loby1/power"
                  c.publish('cmnd/t_loby2/POWER', message) if last_loby2 != message
                  last_loby2=message
                elsif topic=="stat/t_loby2/power"
                  c.publish('cmnd/t_loby1/POWER', message) if last_loby1 != message
                  last_loby1=message
                end
          end
        end
end
```

tomekkepa, 2017-06-28:
Thanks very much

jormus, 2017-07-10:
@davidelang 
hi,
you say "by default, pushing the button on a sonoff switches the relay and sends a
message, but you can configure it to just send a message instead."

Can you point me out how to do that? i cant figure it out.... (sonoff touch, latest firmw)
thx in advance.
ps, i love you work, is there a way to buy you a beer?
dutch reply is fine for me ;-)

qorron, 2017-07-10:
there is another solution, without scripting, just via mqtt:
actually 2 simmilar solutions.
all use the "ButtonTopic"
if set, the button topic only sends out a
cmnd/sonoff/POWER ON/OFF
that gets distributed via the mqtt broker to all devices.
those in turn switch their relays on/off
so, all devices that share the same name e.g. sonoff will activate.
this happens almost instantly without any perceivable delay.
this works but isn't pretty as you have to have several devices with the same name.
alternative:
adhere to the group topic:
set the names to hallway1 hallway2 hallway3 etc.
and the group topic on every switch to hallway
also the buttontopic shall be set to hallway.
so, if a switch is pressed, it sends out cmnd/hallway/POWER ON/OFF [0]
and all the hallway switches react to that.

these solutions give you freedom on how you do your wiring. all lights can be hooked up to a single switch, or each switch controls one light or the lights are controlled by separate sonoff basic or POW modules hidden away.
or any combination of that.

this solution has one problem:
[0] in version:
 * 3.1.14 20170107
the changelog reads:
 * Add support for command TOGGLE (define MQTT_CMND_TOGGLE) when ButtonTopic is in use and not equal to Topic (#207)
hence in our better looking solution there is not ON/OFF corresponding to the status if the activated switch. instead a TOGGLE is sent out every time.
this works fine unless one of the switches misses a command due to whatever reason or it is touched twice hence triggering the relay directly without informing the other switches to do something.
then it is out of sync and has to be brought back.

once I get a round tuit, I'll write an issue that the buttontopic should also be checked against the grouptopic and the switch should only issue a TOGGLE command if the button topic does not equal to any topics set in the switch.

qorron, 2017-07-10:
@jormus in short this will be:
go to the web interface of each device, select console and do:

Topic hallway
ButtonTopic hallway

or if you don't like that your switches share the same name:

Topic hallway<switch suffix/number e.g. 1, 2, 3, east, west,..>
ButtonTopic hallway
GroupTopic hallway

and you are good to go.

davidelang, 2017-07-10:
look at buttontopic in the commands section.

@arendst is the one who deserves the beer, he's doing the programming, I'm just 
answering questions and helping.

David Lang

davidelang, 2017-07-10:
actually, having multiple devices with the same topic can cause problems 
(mosquitto may decide that the multiple devices are really one device and 
disconnect the first when the second connects)

use grouptopic hallway instead of setting the devices all to the same topic. 
That's what grouptopic was created for.
k

jormus, 2017-07-10:
@davidelang @arendst 
yes, got that right, this was indeed on option. But I want it different. I want to act my touch (sonoff touch) button to act as a button alone, without following the relay? I see switchmode 3/4 is what I need. 
(for single tap, I want long tap as well, what should send out on another topic, like "MQTT: cmnd/sonoff_touch1/longTap = TRUE"). Now, when I enter the command "switchmode", what should give me the current state, it says "MQTT: stat/sonoff_touch1/RESULT = {"Command":"Unknown"}"
I tried every version of the command I could think of, even resulting that the console stopt working, and i had to reboot the device :-)

So short version; 
short tap = cmnd1, 
long tap= cmnd2,
and relay should subscribe to cmnd/RELAY (ON/OFF)

ps thx for reply guys, but for now, I drink the beer while trying, and when it works, I send you one as well 👍 (Ik woon in belgie, dus goe bier in overvloed 🍻 )

davidelang, 2017-07-10:
switchmode works for switches, the doubletap only works for button0

what you are looking for (long tap) would require additional code.

jormus, 2017-07-10:
@davidelang 
Thats what I figured out. I'm happy to pay for the hours of work :-)
Because I do not see another option to get this running like 1 want..

lmamakos, 2017-07-11:
> actually, having multiple devices with the same topic can cause problems
(mosquitto may decide that the multiple devices are really one device and
disconnect the first when the second connects)

> use grouptopic hallway instead of setting the devices all to the same topic.
That's what grouptopic was created for.

There's no problem with having multiple devices subscribing to the same topic.  The problem you refer to is when there are duplicate client-ids - the MQTT broker sees the same client-id and decides that any existing connection is bogus since the client has re-connected.

closed 2018-05-07 by stale[bot]