arendst/Tasmota #554: Sonoff basic not responding after OTA update to 5.2.2

ghayne, 2017-06-27:
Did an OTA update today on a Sonoff basic to 5.2.2 and it looks like it's bricked itself. I have not tried to re-flash yet.

Garry

arendst, 2017-06-27:
What's the previous version?

ghayne, 2017-06-27:
5.2.1

arendst, 2017-06-27:
Is there any output on the serial interface?

ghayne, 2017-06-27:
I'll let you know, not at home at the moment.

aferraz, 2017-06-27:
Hi @arendst, same problem here upgrading from 5.1.5 to 5.2.2.
At the moment I can attach sonoff to serial but from syslog I can see a boot loop:
ESP-MQTT: tele/sonoff3/LWT = Online (retained)
ESP-MQTT: cmnd/sonoff3/POWER = 
ESP-Wifi: Connected
ESP-HTTP: Webserver active on sonoff3-3764 with IP address 192.168.1.72
ESP-MQTT: Attempting connection...
ESP-MQTT: Connected
ESP-MQTT: tele/sonoff3/LWT = Online (retained)
ESP-MQTT: cmnd/sonoff3/POWER = 
ESP-Wifi: Connected
ESP-HTTP: Webserver active on sonoff3-3764 with IP address 192.168.1.72
ESP-MQTT: Attempting connection...
ESP-MQTT: Connected
ESP-MQTT: tele/sonoff3/LWT = Online (retained)

Thanks
 Alessandro

davidelang, 2017-06-27:
so this shows that it is running, but it looks like the wifi is getting 
disconnected on a regular basis or that your DHCP server is unhappy and 
disconnecting things.

double check that you don't have duplicate IP addresses and/or duplicate names 
between multiple sonoff devices

aferraz, 2017-06-27:
@davidelang I don't think it's a DHCP or Wifi problem. The device is reachable for 3/4 seconds, in this period I managed to downgrade the firmware and the sonoff back to life.

davidelang, 2017-06-27:
see the troubleshooting page on the wiki, increase the debugging level and see 
what's happening in more detail.

aferraz, 2017-06-27:
@davidelang 
I try with anoter sonoff with "Mode Debug" and this is the log (syslog)
Jun 27 23:32:31 sonoff2.iltuoweb.net sonoff2-4702 ESP-Wifi: Checking connection...
Jun 27 23:32:31 sonoff2.iltuoweb.net sonoff2-4702 ESP-Wifi: Connected
Jun 27 23:32:31 sonoff2.iltuoweb.net sonoff2-4702 ESP-HTTP: Webserver active on sonoff2-4702 with IP address 192.168.1.71
Jun 27 23:32:32 sonoff2.iltuoweb.net sonoff2-4702 ESP-MQTT: Attempting connection...
Jun 27 23:32:32 sonoff2.iltuoweb.net sonoff2-4702 ESP-MQTT: Connected
Jun 27 23:32:32 sonoff2.iltuoweb.net sonoff2-4702 ESP-MQTT: tele/sonoff2/LWT = Online (retained)
Jun 27 23:32:32 sonoff2.iltuoweb.net sonoff2-4702 ESP-MQTT: cmnd/sonoff2/POWER = 
Jun 27 23:32:33 sonoff2.iltuoweb.net sonoff2-4702 ESP-RSLT: Receive topic cmnd/sonoff2/power, data size 2, data ON
Jun 27 23:32:33 sonoff2.iltuoweb.net sonoff2-4702 ESP-RSLT: DataCb Group 0, Index 1, Type POWER, Data ON (ON)
Jun 27 23:32:42 sonoff2.iltuoweb.net sonoff2-4702 ESP-Wifi: Checking connection...
Jun 27 23:32:42 sonoff2.iltuoweb.net sonoff2-4702 ESP-Wifi: Connected
Jun 27 23:32:42 sonoff2.iltuoweb.net sonoff2-4702 ESP-HTTP: Webserver active on sonoff2-4702 with IP address 192.168.1.71
Jun 27 23:32:43 sonoff2.iltuoweb.net sonoff2-4702 ESP-HTTP: Handle root
Jun 27 23:32:43 sonoff2.iltuoweb.net sonoff2-4702 ESP-MQTT: Attempting connection...
Jun 27 23:32:43 sonoff2.iltuoweb.net sonoff2-4702 ESP-MQTT: Connected
Jun 27 23:32:43 sonoff2.iltuoweb.net sonoff2-4702 ESP-MQTT: tele/sonoff2/LWT = Online (retained)
Jun 27 23:32:43 sonoff2.iltuoweb.net sonoff2-4702 ESP-MQTT: cmnd/sonoff2/POWER = 
Jun 27 23:32:43 sonoff2.iltuoweb.net sonoff2-4702 ESP-RSLT: Receive topic cmnd/sonoff2/power, data size 2, data ON
Jun 27 23:32:43 sonoff2.iltuoweb.net sonoff2-4702 ESP-RSLT: DataCb Group 0, Index 1, Type POWER, Data ON (ON)

In this case however after some reboots the item seem stable and working fine.
I make some tests with other units.

davidelang, 2017-06-27:
when you find one that's unstable, set the log level to 'more debugging'

aferraz, 2017-06-27:
@davidelang it's very strange but the trick is setting logging to "More debug".
I upgrade 6 sonoffs POW by setting logging level to debug and all work fine, after upgrade setting back log level to info and try some restart without problems

aferraz, 2017-06-28:
More info about the problem.
After a home power outage all sonoffs POW begun a reboot loop. Also in this case the solution was setting SysLog to 4

Regards

arendst, 2017-06-28:
I cannot reproduce. 

Please provide a complete logging with option 4 (More debug) for one device. Your previous logs are at least missing the Info messages and probably more.

See below the output of my syslog:
```
Jun 28 16:04:19 pow2 ESP-Wifi: Checking connection...
Jun 28 16:04:19 pow2 ESP-Wifi: Connected
Jun 28 16:04:19 pow2 ESP-mDNS: Initialized
Jun 28 16:04:19 pow2 ESP-HTTP: Webserver active on pow2.local with IP address 192.168.2.192
Jun 28 16:04:20 pow2 ESP-MQTT: Attempting connection...
Jun 28 16:04:21 pow2 ESP-mDNS: Query done with 0 mqtt services found
Jun 28 16:04:21 pow2 ESP-MQTT: Connected
Jun 28 16:04:21 pow2 ESP-MQTT: tele/pow2/LWT = Online (retained)
Jun 28 16:04:21 pow2 ESP-MQTT: cmnd/pow2/POWER = 
Jun 28 16:04:21 pow2 ESP-MQTT: tele/pow2/INFO1 = {"Module":"Sonoff Pow", "Version":"5.2.2", "FallbackTopic":"DVES_A01626", "GroupTopic":"sonoffs"}
Jun 28 16:04:21 pow2 ESP-MQTT: tele/pow2/INFO2 = {"WebserverMode":"Admin", "Hostname":"pow2", "IPaddress":"192.168.2.192"}
Jun 28 16:04:21 pow2 ESP-MQTT: tele/pow2/INFO3 = {"Started":"Power on"}
Jun 28 16:04:22 pow2 ESP-MQTT: stat/pow2/RESULT = {"POWER":"ON"}
Jun 28 16:04:22 pow2 ESP-MQTT: stat/pow2/POWER = ON
```
Or better still, open a console when the pow is up and running. Set weblog to 4. While keeping the console open remove and reapply power to your pow. Return the messages on the console:
```
00:00:00 Cnfg: Load from flash at FB and count 151
00:00:00 APP: Bootcount 28
00:00:00 APP: Project sonoff Pow2 (Topic pow2, Fallback DVES_A01626, GroupTopic sonoffs) Version 5.2.2
00:00:00 Cnfg: Save (1492 bytes) to flash at FB and count 152
00:00:00 Wifi: Checking connection...
00:00:00 Wifi: Attempting connection...
00:00:00 Wifi: Patch issue 2186
00:00:00 Wifi: Connecting to AP2 indebuurt2 in mode 11N as pow2...
00:00:01 Wifi: Checking connection...
00:00:01 Wifi: Attempting connection...
00:00:02 Wifi: Checking connection...
00:00:02 Wifi: Attempting connection...
00:00:03 Wifi: Checking connection...
00:00:03 Wifi: Attempting connection...
00:00:04 Wifi: Checking connection...
00:00:04 Wifi: Connected
00:00:04 mDNS: Initialized
00:00:04 HTTP: Webserver active on pow2.local with IP address 192.168.2.192
00:00:06 MQTT: Attempting connection...
00:00:07 mDNS: Query done with 0 mqtt services found
00:00:07 MQTT: Connected
00:00:07 MQTT: tele/pow2/LWT = Online (retained)
00:00:07 MQTT: cmnd/pow2/POWER = 
00:00:07 MQTT: tele/pow2/INFO1 = {"Module":"Sonoff Pow", "Version":"5.2.2", "FallbackTopic":"DVES_A01626", "GroupTopic":"sonoffs"}
00:00:07 MQTT: tele/pow2/INFO2 = {"WebserverMode":"Admin", "Hostname":"pow2", "IPaddress":"192.168.2.192"}
00:00:07 MQTT: tele/pow2/INFO3 = {"Started":"Power on"}
00:00:08 MQTT: stat/pow2/RESULT = {"POWER":"ON"}
00:00:08 MQTT: stat/pow2/POWER = ON
00:00:12 RTC: (UTC) Wed Jun 28 14:08:10 2017
00:00:12 RTC: (DST) Sun Mar 26 02:00:00 2017
00:00:12 RTC: (STD) Sun Oct 29 03:00:00 2017
16:08:13 MQTT: tele/pow2/STATE = {"Time":"2017-06-28T16:08:13", "Uptime":0, "Vcc":3.122, "POWER":"ON", "Wifi":{"AP":2, "SSID":"indebuurt2", "RSSI":96, "APMac":"00:22:6B:FE:8E:20"}}
16:08:13 MQTT: domoticz/in = {"idx":174,"nvalue":0,"svalue":"0;430.7"}
16:08:13 MQTT: tele/pow2/ENERGY = {"Time":"2017-06-28T16:08:13", "Total":0.431, "Yesterday":0.046, "Today":0.041, "Period":0, "Power":0, "Factor":0.00, "Voltage":203, "Current":0.000}
16:08:24 Wifi: Checking connection...
16:08:24 Wifi: Connected
```

aferraz, 2017-06-28:
@arendst with syslog 4 I don't have the problem. I don't know why. I'll take one pow and will link to serial console.

aferraz, 2017-06-28:
@arendst 
I do more tests. 
The problem could be the syslog host. I always use a dns name but in this version It would seem to cause problems if syslog is set to INFO, in the other case work.
I put syslog server ip and it work also with syslog INFO

this is the log from console when syslog is set to INFO with dns name os the syslogserver:
00:00:00 Wifi: Checking connection...
00:00:00 Wifi: Attempting connection...
00:00:00 Wifi: Patch issue 2186
00:00:00 Wifi: Connecting to AP1 TP-LINK_80666A in mode 11N as sonoff3-3764...
00:00:01 Wifi: Checking connection...
00:00:01 Wifi: Attempting connection...
00:00:02 Wifi: Checking connection...
00:00:02 Wifi: Attempting connection...
00:00:03 Wifi: Checking connection...
00:00:03 Wifi: Attempting connection...
00:00:04 Wifi: Checking connection...
00:00:04 Wifi: Attempting connection...
00:00:05 Wifi: Checking connection...
00:00:05 Wifi: Attempting connection...
00:00:06 Wifi: Checking connection...
00:00:06 Wifi: Connected
00:00:06 HTTP: Webserver active on sonoff3-3764 with IP address 192.168.1.72
00:00:06 RTC: (UTC) Wed Jun 28 17:34:45 2017
00:00:06 RTC: (DST) Sun Mar 26 02:00:00 2017
00:00:06 RTC: (STD) Sun Oct 29 03:00:00 2017

ghayne, 2017-06-29:
No output on the serial interface and it appears to be bricked, I cannot re-flash!

davidelang, 2017-06-29:
flashing has nothing to do with what software is on it. The flashing capability 
is built-in to the ESP8266 and cannot be changed/broken by whatever software you 
load on it.

So go back over your setup, check everything (make sure you don't have the 
serial wires reversed, that you are using the correct serial port, etc) and try 
again.

ghayne, 2017-06-29:
Thank you davidlang, I suspected as much.
Garry

terba, 2017-07-20:
@aferraz, @arendst: Similar to LogHost problem in #408

aferraz, 2017-07-20:
@terba I think it could be the same problem. I have the same problem in #408 and take the same solution, change loghost from dns name to ip address.

closed 2018-05-07 by stale[bot]