arendst/Tasmota #555: Newbie - help needed please

Blue407, 2017-06-27:
I have started following the excellent guides on here to get started with Arduino and SonOff.

I have followed all the instructions and pre-requisites, I'm using Arduino IDE.
Its configured in a portable folder as suggested and I have just downloaded all the mentioned libraries.

Thanks in advance for advice and help :)

Error I get when trying to compile:-

Arduino: 1.8.3 (Windows 10), Board: "Generic ESP8266 Module, Serial, 80 MHz, 40MHz, DIO, 115200, 1M (no SPIFFS), ck, Disabled, None"

xdrv_ir_send:46: error: 'IRsend' does not name a type

 IRsend *irsend = NULL;

 ^

D:\Arduino\arduino-1.8.3\portable\sketchbook\sonoff\xdrv_ir_send.ino: In function 'void ir_send_init()':

xdrv_ir_send:50: error: 'irsend' was not declared in this scope

   irsend = new IRsend(pin[GPIO_IRSEND]);  // an IR led is at GPIO_IRSEND

   ^

xdrv_ir_send:50: error: expected type-specifier before 'IRsend'

   irsend = new IRsend(pin[GPIO_IRSEND]);  // an IR led is at GPIO_IRSEND

                ^

xdrv_ir_send:50: error: expected ';' before 'IRsend'

D:\Arduino\arduino-1.8.3\portable\sketchbook\sonoff\xdrv_ir_send.ino: In function 'boolean ir_send_command(char*, uint16_t, char*, uint16_t, int16_t, char*, uint16_t)':

xdrv_ir_send:99: error: 'irsend' was not declared in this scope

           if      (!strcmp_P(protocol,PSTR("NEC")))     irsend->sendNEC(data, bits);

                                                         ^

xdrv_ir_send:100: error: 'irsend' was not declared in this scope

           else if (!strcmp_P(protocol,PSTR("SONY")))    irsend->sendSony(data, bits);

                                                         ^

xdrv_ir_send:101: error: 'irsend' was not declared in this scope

           else if (!strcmp_P(protocol,PSTR("RC5")))     irsend->sendRC5(data, bits);

                                                         ^

xdrv_ir_send:102: error: 'irsend' was not declared in this scope

           else if (!strcmp_P(protocol,PSTR("RC6")))     irsend->sendRC6(data, bits);

                                                         ^

xdrv_ir_send:103: error: 'irsend' was not declared in this scope

           else if (!strcmp_P(protocol,PSTR("DISH")))    irsend->sendDISH(data, bits);

                                                         ^

xdrv_ir_send:104: error: 'irsend' was not declared in this scope

           else if (!strcmp_P(protocol,PSTR("JVC")))     irsend->sendJVC(data, bits, 1);

                                                         ^

xdrv_ir_send:105: error: 'irsend' was not declared in this scope

           else if (!strcmp_P(protocol,PSTR("SAMSUNG"))) irsend->sendSAMSUNG(data, bits);

                                                         ^

exit status 1
'IRsend' does not name a type

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

Blue407, 2017-06-27:
With Verbose on:-

Using library pubsubclient at version 2.6 in folder: D:\Arduino\arduino-1.8.3\portable\sketchbook\libraries\pubsubclient 
Using library Ticker at version 1.0 in folder: D:\Arduino\arduino-1.8.3\portable\packages\esp8266\hardware\esp8266\2.3.0\libraries\Ticker 
Using library ESP8266WiFi at version 1.0 in folder: D:\Arduino\arduino-1.8.3\portable\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WiFi 
Using library ESP8266HTTPClient at version 1.1 in folder: D:\Arduino\arduino-1.8.3\portable\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266HTTPClient 
Using library ESP8266httpUpdate at version 1.1 in folder: D:\Arduino\arduino-1.8.3\portable\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266httpUpdate 
Using library ArduinoJson at version 5.11.0 in folder: D:\Arduino\arduino-1.8.3\portable\sketchbook\libraries\ArduinoJson 
Using library ESP8266WebServer at version 1.0 in folder: D:\Arduino\arduino-1.8.3\portable\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WebServer 
Using library DNSServer at version 1.1.0 in folder: D:\Arduino\arduino-1.8.3\portable\packages\esp8266\hardware\esp8266\2.3.0\libraries\DNSServer 
Using library ESP8266mDNS in folder: D:\Arduino\arduino-1.8.3\portable\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266mDNS (legacy)
Using library Wire at version 1.0 in folder: D:\Arduino\arduino-1.8.3\portable\packages\esp8266\hardware\esp8266\2.3.0\libraries\Wire 
Using library IRremoteESP8266 at version 2.0.3 in folder: D:\Arduino\arduino-1.8.3\portable\sketchbook\libraries\IRremoteESP8266 
Using library NeoPixelBus_by_Makuna at version 2.2.8 in folder: D:\Arduino\arduino-1.8.3\portable\sketchbook\libraries\NeoPixelBus_by_Makuna 
Using library SPI at version 1.0 in folder: D:\Arduino\arduino-1.8.3\portable\packages\esp8266\hardware\esp8266\2.3.0\libraries\SPI

Blue407, 2017-06-27:
Instructions in WIKI say to install IRremoteESP8266 v1.02, and the link in the WIKI no longer works.
It seems the owner upgraded the v2.0.3, as can be sen by my version above.

I;m looking to find the older version...  If I can, I just hope it isn't automatically updated by Arduino IDE...

arendst, 2017-06-27:
Supported versions are in this repository in lib folder.

Blue407, 2017-06-27:
Thanks, will give that a go :)

How can I stop the Arudino IDE from updating to newer libraries from the web?
Is this often a problem?

arendst, 2017-06-27:
It normally does NOT update libraries unless you tell it to do so. Another option is disabling Check for Updates on startup in the IDE settings.

I prefer a dedicated esp8266 arduino environment which doesn't interfere with "normal" arduino stuff. See wiki how to set it up next to your arduino IDE.

Blue407, 2017-06-27:
You Guys are brilliant, compiled perfectly now 👍 

I have this already in a portable folder, is that what you meant?

arendst, 2017-06-27:
Yep

sujitrp, 2017-07-14:
Blue407 can you write it down step by step how issue resolved .

idumzaes, 2017-07-28:
I'm having the same exact issue! Tried on 2 Macs and 2 PCs, same issue!
Please let us know the steps you took to solve this issue!!!

idumzaes, 2017-07-28:
OMG YES! I was able to get it complied successfully on one of my Macs.
What I did was...
1. Show Package Contents of Arduino.app
2. Go into Contents/Java/libraries
3. Rename RobotIRremote folder to IRremote
4. Compile Sonoff.ino

ulab, 2017-09-07:
You might also be able to just not use the IR stuff by disabling USE_IR_REMOTE in user_config.h.

jeylites, 2017-09-09:
And idea why this error is coming out 

/Users/jk/Documents/Arduino/Arduino-1.8.3/portable/sonoff/sonoff.ino:160:52: fatal error: PubSubClient.h: No such file or directory
 #include <PubSubClient.h>                   // MQTT
                                                    ^
compilation terminated.
exit status 1
Error compiling for board Generic ESP8266 Module.

closed 2018-05-07 by stale[bot]