arendst/Tasmota #557: Boot problems after flashing

Blue407, 2017-06-28:
I'm new to all this, so be gentle :)

I've successfully flashed my new SonOff basic unit.

Archiving built core (caching) in: C:\Users\Paul\AppData\Local\Temp\arduino_cache_353327\core\core_esp8266_esp8266_generic_UploadTool_esptool,CpuFrequency_80,FlashFreq_40,FlashMode_dio,UploadSpeed_115200,FlashSize_1M0,ResetMethod_ck,Debug_Disabled,DebugLevel_None_____feb531284d1d511ad1fbcc82bca91a75.a
Sketch uses 480716 bytes (46%) of program storage space. Maximum is 1023984 bytes.
Global variables use 43356 bytes (52%) of dynamic memory, leaving 38564 bytes for local variables. Maximum is 81920 bytes.
warning: espcomm_send_command: wrong direction/command: 0x01 0x00, expected 0x01 0x08
Uploading 484864 bytes from C:\Users\Paul\AppData\Local\Temp\arduino_build_792321/sonoff.ino.bin to flash at 0x00000000
................................................................................ [ 16% ]
................................................................................ [ 33% ]
................................................................................ [ 50% ]
................................................................................ [ 67% ]
................................................................................ [ 84% ]
..........................................................................       [ 100% ]

It boots and I get this in the serial console:-

00:00:00 APP: Project sonoff Sonoff (Topic sonoff, Fallback DVES_05706E, GroupTopic sonoffs) Version 5.2.2
00:00:00 Wifi: Connecting to AP1 indebuurt1 in mode 11N as sonoff-4206...
00:00:07 Wifi: Connect failed as AP cannot be reached
00:00:07 Wifi: Connecting to AP2 indebuurt2 in mode 11N as sonoff-4206...
00:00:14 Wifi: Connect failed as AP cannot be reached
00:00:14 WPSconfig: Active for 1 minute
00:00:33 APP: Restarting

 ets Jan  8 2013,rst cause:1, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v09826c6d
~ld

Now the very first time I flashed this unit, I used the files exactly as they were, just to make sure flashing worked.
The subsequent times, I had already changed the user_config.h file to connect to my local Wi-Fi.

So I do not understand why it keeps trying to connect to AP1 indebuurt1
I have done a search through all the libraries and this does not exist any longer...

So is it not really re-flashing?

Or am I missing something obvious?

The only thing I missed/couldn't do in the steps, was 

Best practice to implement is:

Open the webpage to your device
Perform option Backup Configuration
Upgrade new firmware using Firmware upgrade
If configuration conversion fails keep the webpage open and perform Restore Configuration

I never managed to connect to the web page successfully.

Thanks in advance for ideas and suggestions :) 👍

davidelang, 2017-06-28:
please read the troubleshooting page on the wiki. I think it explains your 
problem.

Blue407, 2017-06-28:
Excellent, I will give that a go tonight :)

It's quite difficult getting started with these, so many new things to learn and some of it is far from intuitive :) :)

closed 2018-05-07 by stale[bot]