arendst/Tasmota #559: The problem of The HC-SR501 PIR Motion Detection in sonoff

vaguecupid, 2017-06-28:
I want add HC-SR501 PIR Motion Detection to the sonoff, so I set GPIO14  to 09switch1, set GPIO1 to 10switch2, set SwitchMode to 0, set SwitchTopic to 1. I add the switch to GPIO14, and PIR to GPIO1.

the result is switch can work perfectly, the PIR can work,but console and MQTT always show（no “OFF”）:

> cmnd/sonoff/POWER2 = ON
stat/sonoff/RESULT = {"Command":"Unknown"}
cmnd/sonoff/POWER2 = ON
stat/sonoff/RESULT = {"Command":"Unknown"}

davidelang, 2017-06-28:
cmnd/ switches relays, not switches.
you would need to have two relays defined before power2 does anything

vaguecupid, 2017-06-28:
I test use the relay,but PIR(relay 2) can not work.

jhlweb, 2017-10-26:
set `Switchmode2 1` this lets switch2 follow the state and publishes it

EdelHert, 2017-12-06:
Thanks jhlweb, thats works for me. The only problem I am having still is that the switch does not stay on. It turns on for 36 seconds (PIR-delay), then turns off, even with motion, and turns on after 6 seconds again.. Any idea how I can make it to stay on until no motion is detected

closed 2018-05-07 by stale[bot]