arendst/Tasmota #561: MHZ-19

jiraff, 2017-06-29:
Hello. Can I add my СО2 sensor to the Tasmota? Connection on the TX TX or PVM. Here's how I connected to gpio14 on PWM:

#define pwmPin 14
#define LedPin 13

Int prevVal = LOW;
Long th, tl, h, l, ppm;

Void setup () {
   Serial.begin (9600);
   PinMode (pwmPin, INPUT);
   PinMode (LedPin, OUTPUT);
}

Void loop () {
   Long tt = millis ();
   Int myVal = digitalRead (pwmPin);

   // If you find a change
   If (myVal == HIGH) {
     DigitalWrite (LedPin, HIGH);
     If (myVal! = PrevVal) {
       H = tt;
       Tl = h - l;
       PrevVal = myVal;
     }
   } Else {
     DigitalWrite (LedPin, LOW);
     If (myVal! = PrevVal) {
       L = tt;
       Th = l - h;
       PrevVal = myVal;
       Ppm = 5000 * (th - 2) / (th + tl - 4);
       Serial.println ("PPM =" + String (ppm));
     }
   }
}

Thank you.

TD-er, 2017-06-30:
The problem with looking to the PWM-pin is that you do not take into account the "S" value computed by the sensor. This bit-shifted value proves to be an indicator for the stability of the measurement of the CO2 value.
The S-value is outputted via the serial line, along with the value.

And another thing, you use a scale of 5000, while most of these sensors are in the range of 400-2000 ppm.

jiraff, 2017-07-05:
I'm wondering what and where to put in the code Sonoff-Tasmota that would see the readings of mhz-19 in mqtt ...

closed 2018-05-07 by stale[bot]