arendst/Tasmota #562: Suggestion Sonoff Basic: Switch for a Time (x second)

Bambutscha, 2017-06-29:
Hello,

i have a Suggestion for the Sonoff Basic.
Is it possible for the Switchmode to say:
Switch on for a Special Time  xx Second ?

Idea behind is to Control with 2 Sonoff Basics and Amazon Alexa a "Awning, mawquee"

- Alexa,  Awning (Markise) off (aus) ! And the the Sonoff Switch for 10second the Switch and Switch them off by him self after 10 second. 

- Alexa,  Awning (Markise) on (ein) ! And the the Sonoff Switch for 10second the Switch and Switch them off by him self after 10 second.

It is nice, then the Relay brings not permanently 230V for the Awning, mawquee.

Is that possible ???

davidelang, 2017-06-29:
take a look at the pulsetime command

Bambutscha, 2017-06-30:
OK, i will test it later.
This it not possible to make them editable over the Webinterface?
it is a Little bit easyer to Play a Little bit with the time.

closed 2017-06-30 by Bambutscha