arendst/Tasmota #563: Problem: switch off after losing Wifi

0n3-70uch, 2017-06-30:
Hi,

My last comment was about the version 3... Now I'm back :-).

I notice a problem with my sonoffs connected to a MQTT server and switched via Domoticz. If the sonoff lose the connection to the MQTT server then the sonoff power off.
For example: if I reboot the MQTT server then the sonoff switch go to off. Same happend with a sonoff outside...

0n3-70uch, 2017-06-30:
Found another interesting thing:
Normally if I'm upgrading the Sonoffs then the devices go off for a second or less. But with the new firmware the devices go off and not on again. And the Domiticz status is also incorrect. Some switches are off but Domoticz show me an on.

arendst, 2017-06-30:
Sounds like you are using one of the many MQTT message retain options. Turn them off and try again.

0n3-70uch, 2017-06-30:
Hi Theo,

no the retain options are off. I test it with the original firmware with only these changes:
`
#ifdef WIFI_GATEWAY
#undef WIFI_GATEWAY
#endif
#define WIFI_GATEWAY    "192.168.2.1"

#ifdef WIFI_DNS
#undef WIFI_DNS
#endif
#define WIFI_DNS    "192.168.2.1"

#ifdef STA_SSID1
#undef STA_SSID1
#endif
#define STA_SSID1   "name1"

#ifdef STA_PASS1
#undef STA_PASS1
#endif
#define STA_PASS1   "password"

#ifdef STA_SSID2
#undef STA_SSID2
#endif
#define STA_SSID2   "name1"

#ifdef STA_PASS2
#undef STA_PASS2
#endif
#define STA_PASS2   "password"

#ifdef SYS_LOG_HOST
#undef SYS_LOG_HOST
#endif
#define SYS_LOG_HOST    "192.168.2.11"

#ifdef OTA_URL
#undef OTA_URL
#endif
#define OTA_URL     "http://192.168.2.11:8080/Updates/" PROJECT ".ino.generic.bin"

#ifdef MQTT_HOST
#undef MQTT_HOST
#endif
#define MQTT_HOST   "192.168.2.11"

#ifdef MQTT_USER
#undef MQTT_USER
#endif
#define MQTT_USER   "mosquitto"

#ifdef MQTT_PASS
#undef MQTT_PASS
#endif
#define MQTT_PASS   "password"

#ifdef MQTT_CLIENT_ID
#undef MQTT_CLIENT_ID
#endif
#define MQTT_CLIENT_ID    "Sonoff_%06X"

#ifdef MQTT_GRPTOPIC
#undef MQTT_GRPTOPIC
#endif
#define MQTT_GRPTOPIC   "Sonoffs"

#ifdef WEB_SERVER
#undef WEB_SERVER
#endif
#define WEB_SERVER    0

#ifdef WEB_PASSWORD
#undef WEB_PASSWORD
#endif
#define WEB_PASSWORD    "password"

#ifdef USE_IR_REMOTE
#undef USE_IR_REMOTE
#endif`

0n3-70uch, 2017-06-30:
Hi,

the restart of the Sonoff is back... But now I got a message:
> Software Watchdog

Seems thats something is wrong. Its an standard Sonoff with attached AM2301 at GPIO14. MQTT and Domoticz is active nothing else...

0n3-70uch, 2017-07-03:
Hi Im closing this issue now... seems that the wifi strenght is too low (8%). I will make some test in the future...

closed 2017-07-03 by 0n3-70uch