arendst/Tasmota #564: Update fail

0n3-70uch, 2017-06-30:
Hello again,

for me the update is not working fine since firmware 5.1.4. But I also added some Sonoff into the network. So its also possible that the update of all devices at the same time is not working anymore.

The LAN runs at 1000MBit/s but the update server (raspberry pi) allow only 100MBit/s.

If I use the update command for only one device its working fine most the time. But sometimes the same problem occurs.

Greetings
Fabi

ionciubotaru, 2017-07-02:
Your problems comes from wifi signal strength. Check the signal in the information page:
`AP1 SSId (RSSI)   Public (78%)`
If it is below 70% you will have problems during OTA. MQTT works fine even at 20%, but OTA needs a very good wifi signal.

0n3-70uch, 2017-07-02:
Hi, thanks for your help. I will check the WiFi strength and let infirm you later...

0n3-70uch, 2017-07-03:
Seems that there are two problems:

1) Bad wifi quality
2) too many devices at the same time ask for an update...

I close this issue now.

closed 2017-07-03 by 0n3-70uch