arendst/Tasmota #565: Sonoff 4CH Pro Support

erkgh, 2017-07-01:
Any chance that this "enhanced" model of the Sonoff 4CH will be supported?

scargill, 2017-07-01:
I would ask the same as I'm sitting in front of one of these Pro units and would like to relief it of it's original software.

arendst, 2017-07-01:
My pro is on it's way and should arrive any time now.

Looking at the schematics I do not expect big changes to my software. Most new features are controlled by a dedicated microcontroller and switches on the board. I have no control over these settings.

scargill, 2017-07-01:
I'd be quite happy with basic control and Alexa - even without Alexa as I can do that in Node-Red...

arendst, 2017-07-01:
From what I see in the schematics is that the esp8266 connections to relay 3 and 4 have been swapped. I do not yet know if the signals have been inverted as the relays are being controlled by the STM32 microcontroller.

I expect when the configuration switches are set correctly you might be able to use the current firmware configured as a Sonoff 4CH to control the relays while keeping in mind the swapped relays.

Watch this space for more info as soon as mine arrives (probably delayed by the NotPetya attack on the Dutch parcel delivery service)

erkgh, 2017-07-01:
Besides the extra hardware present in the PCB to support among others receiving RF-443Mhz signals, the huge improvement (imho) against the 4CH model is related to the relay part. No Common L wire as mains power for the PCB itself and the 4 relays at the same time, but seperate contacts for each channel  (supporting both normally open & normally closed contacts)

Good to notice that this new model has the attention of the developer (s) though. I am  looking forward to any useable firmware support

marksev1, 2017-07-03:
So will the code support also the RF-433Mhz control of it? So either mqtt via wifi or rf-control?

Justblair, 2017-07-08:
I got one now, pretty good unit....

https://youtu.be/SJNNNiWY9Gg

metraport, 2017-07-18:
Does anyone get the 4ch Pro in Programming mode ? A press on E-FW/Button0 at powering on the device doesn't brig the device in programming mode.

Justblair, 2017-07-18:
Yes... Weirdly enough I just posted a youtube about this...

https://www.youtube.com/watch?v=hOFvbdYkOII

bru1979, 2017-07-19:
@Justblair flashing the 4CH Pro using your instructions went smoothtly, thanks! 

BTW, were you able to control the relays with the buttons and rules? Also setting boot state for GPIOs seem to be ignored, relay 1 at GPIO12 is always on at boot. Maybe it's related to the dip switches?

Justblair, 2017-07-19:
Hi I am still working on that, I am getting similar issues myself.  It looks to me as if the sonoff is loosing the boot state as you suggest.  I managed to make it a bit better by adding 

`On System#Boot do
   GPIO,12,0
   GPIO, 5,0
  GPIO,4,0
  GPIO,15,0
  Relay1#Switch=0
  Relay2#Switch=0
  Relay3#Switch=0
  Relay4#Switch=0
endon

On Button1#Switch=0 do
  if [Relay1#Switch]=1
    GPIO,12,0

  else
    GPIO,12,1
   endif
Endon

On Button2#Switch=0 do
  if [Relay2#Switch]=1
    GPIO,5,0
  else
    GPIO,5,1
   endif
Endon

On Button3#Switch=0 do
  if [Relay3#Switch]=1
    GPIO,4,0
  else
    GPIO,4,1
  endif
endon

On Button4#Switch=0 do
    pulse,15,1,500
endon`

Initiating the relays at the beginning definitely improves things, but even then the relay state gets randomly lost.

With full debug on I am also getting what look like wtd errors (watchdog I guess?).  Something is definately up.

I am going to try using R148 and see if that it is more or less reliable.  

For my purposes I can use the device without the switches, but I would like to keep them as I want RF control.  I can easily achieve it using rfLink and MQTT however I want the remote control to work independently of a wifi connection for reasons.

Justblair, 2017-07-19:
Tried with R148a.  

https://www.letscontrolit.com/forum/viewtopic.php?p=16244#p16244

Initial thoughts... this does seem better.  However the relays begin in mixture of states.  However the Hardware page does seem to correctly set their boot states.  So I dont have to do that on system boot carry on at the beginning.  

It may be my imagination, but the buttons and rf seem more responsive

aequitas, 2017-07-22:
I got mine in the mail this week, @arendst I also noticed some delay in international mail lately. 

Flashed it with `v5.3.0` and everything seems fine. Like @Justblair mentioned the GPIO0 pin does not seem connected to the E-FW button, manually shorting the pin to GND did the trick.

Everything seems to work as expected, relay order seems to be fine. I'll post an update when I have a chance to install this somewhere useful later this or next week.

arendst, 2017-07-22:
@aequitas glad you were able to test it this far. Today I went to the postoffice to see if my package might have been received there. It appeared they were just about to return the package as it was sitting there for two weels already. I never received any notice. So much for dutch post nl...

I managed to retreive the present code using 2k2 resistor from gpio0 to gnd as indeed the fwr switch didn't work.

I  will start testing too asap.

arendst, 2017-07-24:
Did some testing leading to the following conclusions:

1. In Self-Locking mode (switches K5 set to 1 (not default), K6 set to 0 (default) and S6 set to 1 (default)) the device works fine with the current Sonoff 4CH Module selection.
2. In Interlock and Inching mode strange behaviour can be observed as the button signal levels fed to the esp8266 and controlled by the microcontroller are not as expected by my firmware.

I'm working on a version which allows Interlock and inching too but that will need special configuration by the user as he/she will loose the multipress and hold functionality for all buttons for this device.

I'm also wondering if I should support Interlock and Inching anyway as it needs fiddling with the switches and some settings contradict eachother.

bru1979, 2017-07-24:
@arendst +1 for inching, I need to simulate 3 active high push buttons (open, stop, close) for a garage door

aequitas, 2017-07-25:
For inching/interlocking I would prefer a software solution over the current hardware switches. As it offers better configuration and doesn't require access to the hardware.

I mostly choose this product for the individual switches (compared to the 4ch which are all wired to mains) and not for the RF or special modes, I think some others might have the same reason as well.

bru1979, 2017-07-25:
I'd rather prefer the hardware solution because in case of a "soft" failure (network/controller issues, etc) the NO contact condition would alway be restored in hard inching mode and my door wouldn't be stuck. that's just my two cents of course.

aequitas, 2017-07-25:
@bru1979 thats what I meant, as a firmware feature.

arendst, 2017-07-25:
Implemented in version 5.4.0 just released. 

Select Module ``Sonoff 4CH Pro`` (All the way down the list) and fool around with the jumper switches (while power off as suggested by ITEAD).

As the buttons are fed through the microcontroller to the esp8285 I have no control over multi-press and hold and therefore these options are not available for the Sonoff 4CH Pro. In fact inching and interlock use button presses to update esp8285 status!

Double press initiates 433MHz button search for some 10 seconds as iTEAD documents on their wiki page.

Let me know of any anomalies and have fun.

aequitas, 2017-07-25:
Latest firmware working as expected. I added some infos about this board to the wiki.

bru1979, 2017-07-26:
hard inching works flawlessly (K5/1-4 all on). thanks @arendst!

scropion86, 2017-07-26:
hi guys ,

as i understand that the full futures of the 4Ch pro seams not all will be working , so if i am going to get one what is the cons and pros between 4CH VS 4CH Pro assuming i will not use the official FW.

thanks

jeylites, 2017-07-27:
Not a 4Ch pro question, I'm just curious of the inching function works on Sonoff basic or dual?

arendst, 2017-07-27:
@jeylites inching as available on the Sonoff 4CH Pro is microcontroller based and not esp8266 based. 

Both Sonoff Basic and Sonoff Dual do not have this functionality although you can always use PulseTime to simulate inching on any Sonoff.

mcfarlde, 2017-09-04:
Hello,  Been playing with Hassio, MQTT and a 4CH Pro at Tasmota 5.6.1.  I see traffic in the 4CH Pros console.  I'm quite confused on the topics.   Are the friendly names available as topics for each relay or do we use the Client_ID as set in the MQTT options and reported as Topic on the information tab with POWER1, POWER2, etc... for Relay 1, Relay 2, etc...  If someone has 4CH Pro working in Hassio, I'd appreciate screen shots of the 4CH Pro MQTT configuration and the switch: definition in configuration.yaml that works with it.

savage872, 2018-02-04:
Can any one see the 4CH PRO working in a 2way (3way) light switch system and getting the bulb state right as well ? :)

closed 2018-06-19 by stale[bot]

nickdeacon, 2020-01-16:
I have re-written the firmware for the STM32F030 chip on the sonoff 4ch Pro, to create a PIR based security lighting system using the sonoff RF PIR units, and including a light sensor to ensure that switching only occurs at night. https://github.com/nickdeacon/PIR-Lighting-controller

joba-1, 2020-01-16:
Cool stuff! Thanks for the detailed description!
I currently try to learn how to program STM32 chips and succeeded on doing it with a Nucleo and an F7 based drone flight controller. I wasn't able to generalize on other chips yet (e.g. an L151 Lora node with a demo firmware but not much documentation). Your work will definitely help.