arendst/Tasmota #567: Hardware Support: BN-SZ01 WiFi Round Flush Mount LED Ceiling Light

jaysuk, 2017-07-01:
Are you planning to add support for the above product?

https://captain-slow.dk/2017/04/10/itead-bn-sz01-wifi-round-flush-mount-led-ceiling-light-part-2-hacking-it/

Its based on an ESP8285, the LED's are driven by a BP2838G connected to pin 12 and supports PWM.
Or is there enough functionality in the software for me to create a new profile?

arendst, 2017-07-01:
I have one lying around  but thought flashing would be a too big challenge...

After reading the article I will reconsider.

jaysuk, 2017-07-01:
Good News.
I plan for it to become my new porch light

jaysuk, 2017-07-01:
I have managed to flash it ok.
The LED is on GPIO 13.
Can't turn it on and off though.

davidelang, 2017-07-01:
did you configure GPIO13 to be a relay?

David Lang

On Sat, 1 Jul 2017, jaysuk wrote:

> Date: Sat, 01 Jul 2017 14:52:57 -0700
> From: jaysuk <notifications@github.com>
> Reply-To: arendst/Sonoff-Tasmota
>     <reply+0021b24f00d0ea3208c149eef32de0f8ee180215b894015392cf00000001156fdbb
>     992a169ce0e4d278c@reply.github.com>
> To: arendst/Sonoff-Tasmota <Sonoff-Tasmota@noreply.github.com>
> Cc: Subscribed <subscribed@noreply.github.com>
> Subject: Re: [arendst/Sonoff-Tasmota] Hardware Support: BN-SZ01 WiFi Round
>     Flush Mount LED Ceiling Light (#567)
> 
> I have managed to flash it ok.
> The LED is on GPIO 13.
> Can't turn it on and off though.
>
>

jaysuk, 2017-07-01:
no. pin 12 as PWM LED.

Here's my config

  { "Sonoff Round",       // Sonoff SC (ESP8266)
     GPIO_KEY1,        // GPIO00 Button
     0, 0, 0,
     GPIO_USER,        // GPIO04 Optional sensor (PWM3 Green)
     GPIO_USER,        // GPIO05 Optional sensor (PWM2 Red)
     0, 0, 0, 0, 0, 0, // Flash connection
     GPIO_PWM1,        // GPIO12 Cold light (PWM0 Cold)
     GPIO_LED1_INV,    // GPIO13 Blue Led (0 = On, 1 = Off)
     0,        //
     GPIO_USER,        // GPIO15 Optional sensor (PWM4 Blue)
     0, 0
  }

I think some more code may need modify

arendst, 2017-07-02:
Just flashed my BN-SZ01 with tasmota 5.2.3 and here are the results:
- Initial flash needs to be as esp8285 OR esp8266 with flash mode DOUT
- Select Sonoff Led
- Set initial color with ``Color 2000`` as the led is connected to the "Cold" led connection
- Control led with commands Dimmer 0..99 (Dimmer 100 results in flicker as noticed by https://captain-slow.dk/2017/04/10/itead-bn-sz01-wifi-round-flush-mount-led-ceiling-light-part-2-hacking-it/)

jaysuk, 2017-07-02:
That works ok for me.
Time to fit it to my porch

arendst, 2017-07-03:
When OTA or web upgrading your BN-SZ01 to version 5.2.4 follow the steps below to make sure that the device is NOT re-programmed to esp8266:
- Open a webpage to the BN-SZ01 and select Sonoff Touch as module
- After restart select upgrade from webpage and select ``sonoff.bin``
- After restart select module Sonoff BN-SZ as module and enjoy your device.

amitubale, 2017-07-03:
the flicker comment was by me and let me clarify that i completle overlooksed that this is a 220V device. here in US it 110. so please verify. back in May i was not able to get this code on the so used the precompolied code from captain-slow's steps and have easyesp.. i iwll again go though the tedious stepsto get he leads solders on the chip and grounding gpio0 to flash this code...I am not sure how to select esp8266 ..will check the wiki.

arendst, 2017-07-03:
@amitubale thnx for the info. I have 230V and it (used to) flicker only at full dim value where the Arduino PWM routine stops sending PWM and keep the signal steady.

In version 5.2.4 just release I make sure that the max PWM value is not reached and therefor the PWM signal is still being send to the led control chip and I see no flicker. YMMV but let me know.

jaysuk, 2017-07-03:
All upgraded here fine for me.
Good work

closed 2017-07-03 by jaysuk

rajil, 2017-09-24:
@amitubale Were you able to use it with 110V?

I flashed it Tasmota 5.8.0 and it works fine with 220V (used a step-up transformer). When i connect it to 110V the LEDs come on however the device doesnt make a DHCP request. I wonder if the ESP chip is actually powered on or not.

kvjajoo, 2018-01-25:
Does anyone know how to add an external dimmer switch to BN-SZ01. I have flashed it with tasmota and its working fine but since my parents are not techie enough I need to place a manual switch in place. I have been using sonoff t1 for rest of the thing but there is no point connecting BN-SZ01 to T1 as it will turn of BN-SZ alltogether. 

I have below AC dimmer switch and I was wondering if I can connect it to BN-SZ01. I am sure some firmware changes will be required as well. Can anyone please help ?

[image]