arendst/Tasmota #569: how change module in user_config.h

zhilly, 2017-07-01:
Hey there,

I've downloaded the last source code 5.2.3 20170630 and try to editing user_config.h from sonoff basic to ch4 module but i don't find this setting. because i want flash a 4ch module not the basic one.

someone can help me?

thx

ferbulous, 2017-07-01:
you can change to 4ch module later after you flash it
would be nice to know how to properly select the device type before flashing, anyone can help?

SwiftNick, 2017-07-01:
At the top of sonoff.ino (Line 47)

``` c++ 
#define MODULE                 SONOFF_BASIC // [Module] Select default model
```

it can be any of the models defined in sonoff_template.h

```c++
// Supported hardware modules
enum module_t {
  SONOFF_BASIC,
  SONOFF_RF,
  SONOFF_SV,
  SONOFF_TH,
  SONOFF_DUAL,
  SONOFF_POW,
  SONOFF_4CH,
  S20,
  SLAMPHER,
  SONOFF_TOUCH,
  SONOFF_LED,
  CH1,
  CH4,
  MOTOR,
  ELECTRODRAGON,
  EXS_RELAY,
  WION,
  WEMOS,
  SONOFF_DEV,
  H801,
  SONOFF_SC,
  MAXMODULE };
```

arendst, 2017-07-01:
No need to change anything in source or user_config.h

Just load the default firmware; open the webpage and configure desired module. This way the code is unchanged and any new version can be uploaded as is.

zhilly, 2017-07-02:
okey thx for all i had fear before i flash it.

closed 2017-07-02 by zhilly