arendst/Tasmota #570: Boot loop on saving values in configuration web page.

LordVanguard, 2017-07-01:
Greetings, 

I seem to have encountered an issue which I am able to replecate on multiple Sonoff units. 

I am using three different SONOFF basic WiFi units. I have flashed the Tasmota compiled binary to the Sonoff unit and all works on reboot. I can access the web pages etc, however, as soon as I make a change and hit save and it states that the device will reboot, the Sonoff unit just goes into a boot loop with the following displayed on the serial monitor

`
 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v09826c6d
~ld

 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v09826c6d
~ld
`

I can see that the watchdog resets the unit so something didnt quite stick. I can unplug the unit from power and then plug it back in with the unit just heading straight back into the boot loop. 

I saw that there was another thread in the issues page which stated that the solution was that there may have been a defective unit, however I have tested this with three Sonoff Basic Wifi units with exactly the same result. 

If I reflash the unit with the Tasmota compiled binary the unit comes back to life. 

Thoughts ? 

Any help would be received most welcome.

-L

arendst, 2017-07-01:
What did you change and save?

LordVanguard, 2017-07-01:
Apologies, I should have included that. 
I am using 5.2.2
The change I made was to the MQTT username, address, and password. 

I see that there is a 5.2.3 which I have literally just downloaded, and will test via an upload to the SONOFF unit in about 5 mins. I will report back if the issue persists with 5.2.3

Serious thanks for the speedy response though !!

arendst, 2017-07-01:
What are the exact changes to these MQTT parameters?

LordVanguard, 2017-07-01:
Changes were to change
- Host Ip with my MQTT broker IP £10.0.0.30
- Client ID to "SONOFF03_%06X"
- User to Username "sensor-test"
- Password to the 21 character password used for the MQTT broker all just letters and numbers

I have just flashed with 5.2.3 and have placed the information above into the arduino sketch (IDE 1.8.4) and flashed the SONOFF. Unit loaded. I attempted to change the Client Id to see if it would be just the one thing that caused the issue. I changed the Id from "SONOFF03_%06X" to SONOFF04_%06X" and hit save. It stated that the "Device will restart in a few seconds" 

Unit goes into the loop ... com port readout as follows. 

> ⸮
00:00:00 Cnfg: Use defaults
00:00:00 APP: Project sonoff03 Sonoff03 (Topic sonoff03, Fallback SONOFF_056054, GroupTopic sonoffs) Version 5.2.3
00:00:00 Wifi: Connecting to AP1 paddock in mode 11N as sonoff03-0084...
00:00:04 Wifi: Connected
00:00:04 mDNS: Initialized
00:00:04 HTTP: Webserver active on sonoff03-0084.local with IP address 10.0.0.16
00:00:05 MQTT: Attempting connection...
21:43:29 mDNS: Query done with 0 mqtt services found
21:43:29 MQTT: Connected
21:43:29 MQTT: tele/sonoff03/LWT = Online (retained)
21:43:29 MQTT: cmnd/sonoff03/POWER = 
21:43:29 MQTT: tele/sonoff03/INFO1 = {"Module":"Sonoff Basic", "Version":"5.2.3", "FallbackTopic":"SONOFF_056054", "GroupTopic":"sonoffs"}
21:43:29 MQTT: tele/sonoff03/INFO2 = {"WebserverMode":"Admin", "Hostname":"sonoff03-0084", "IPaddress":"10.0.0.16"}
21:43:29 MQTT: tele/sonoff03/INFO3 = {"Started":"Power on"}
21:43:30 MQTT: stat/sonoff03/RESULT = {"POWER":"OFF"}
21:43:30 MQTT: stat/sonoff03/POWER = OFF
21:43:37 MQTT: tele/sonoff03/STATE = {"Time":"2017-07-01T21:43:37", "Uptime":0, "Vcc":3.238, "POWER":"OFF", "Wifi":{"AP":1, "SSID":"paddock", "RSSI":100, "APMac":"08:60:6E:21:CE:98"}}
21:48:27 HTTP: MQTT Host 10.0.0.30, Port 1883, Client SONOFF04_%06X, User sensor-user, Password 331d46d9edc5b11266ef262fb0853fba, Topic sonoff03, FullTopic 
21:48:29 APP: Restarting

 ets Jan  8 2013,rst cause:1, boot mode:(1,6)

 ets Jan  8 2013,rst cause:4, boot mode:(1,6)

wdt reset

LordVanguard, 2017-07-01:
Its a temporary generated password so Im not concerned about it being here

LordVanguard, 2017-07-01:
If I unplug the unit and plug it back in this time it does come back to life. I can change the topic and it does save and come back up. I attempted to change the MQTT client ID again and it looped.

terminal readout as follows
`00:00:04 Wifi: Connected
00:00:04 mDNS: Initialized
00:00:04 HTTP: Webserver active on sonoff03-0084.local with IP address 10.0.0.16
00:00:06 MQTT: Attempting connection...
21:54:43 mDNS: Query done with 0 mqtt services found
21:54:43 MQTT: Connected
21:54:43 MQTT: tele/sonoff03/LWT = Online (retained)
21:54:43 MQTT: cmnd/sonoff03/POWER = 
21:54:43 MQTT: tele/sonoff03/INFO1 = {"Module":"Sonoff Basic", "Version":"5.2.3", "FallbackTopic":"SONOFF04_056054", "GroupTopic":"sonoffs"}
21:54:43 MQTT: tele/sonoff03/INFO2 = {"WebserverMode":"Admin", "Hostname":"sonoff03-0084", "IPaddress":"10.0.0.16"}
21:54:43 MQTT: tele/sonoff03/INFO3 = {"Started":"Power on"}
21:54:44 MQTT: stat/sonoff03/RESULT = {"POWER":"OFF"}
21:54:44 MQTT: stat/sonoff03/POWER = OFF
21:54:51 MQTT: tele/sonoff03/STATE = {"Time":"2017-07-01T21:54:51", "Uptime":0, "Vcc":3.238, "POWER":"OFF", "Wifi":{"AP":1, "SSID":"paddock", "RSSI":100, "APMac":"08:60:6E:21:CE:98"}}
21:56:13 MQTT: tele/sonoff03/LWT =  (retained)
21:56:13 HTTP: MQTT Host 10.0.0.30, Port 1883, Client SONOFF04_%06X, User sensor-user, Password 331d46d9edc5b11266ef262fb0853fba, Topic sonoff04, FullTopic 
21:56:16 APP: Restarting

 ets Jan  8 2013,rst cause:1, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v09826c6d
~ld
⸮
00:00:00 APP: Project sonoff03 Sonoff03 (Topic sonoff04, Fallback SONOFF04_056054, GroupTopic sonoffs) Version 5.2.3
00:00:00 Wifi: Connecting to AP1 paddock in mode 11N as sonoff04-0084...
00:00:06 Wifi: Connected
00:00:06 mDNS: Initialized
00:00:06 HTTP: Webserver active on sonoff04-0084.local with IP address 10.0.0.16
21:56:24 MQTT: Attempting connection...
21:56:26 mDNS: Query done with 0 mqtt services found
21:56:26 MQTT: Connected
21:56:26 MQTT: tele/sonoff04/LWT = Online (retained)
21:56:26 MQTT: cmnd/sonoff04/POWER = 
21:56:26 MQTT: tele/sonoff04/INFO1 = {"Module":"Sonoff Basic", "Version":"5.2.3", "FallbackTopic":"SONOFF04_056054", "GroupTopic":"sonoffs"}
21:56:26 MQTT: tele/sonoff04/INFO2 = {"WebserverMode":"Admin", "Hostname":"sonoff04-0084", "IPaddress":"10.0.0.16"}
21:56:26 MQTT: tele/sonoff04/INFO3 = {"Started":"Software/System restart"}
21:56:27 MQTT: stat/sonoff04/RESULT = {"POWER":"OFF"}
21:56:27 MQTT: stat/sonoff04/POWER = OFF
21:56:35 MQTT: tele/sonoff04/STATE = {"Time":"2017-07-01T21:56:35", "Uptime":0, "Vcc":3.237, "POWER":"OFF", "Wifi":{"AP":1, "SSID":"paddock", "RSSI":100, "APMac":"08:60:6E:21:CE:98"}}
21:56:49 MQTT: tele/sonoff04/LWT =  (retained)
21:56:49 HTTP: MQTT Host 10.0.0.30, Port 1883, Client SONOFF03_%06X, User sensor-user, Password 331d46d9edc5b11266ef262fb0853fba, Topic sonoff03, FullTopic 
21:56:51 APP: Restarting

 ets Jan  8 2013,rst cause:1, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v09826c6d
~ld

 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v09826c6d
~ld
`

arendst, 2017-07-01:
In the sequence above, did you
- upload the code via serial
- openend a webpage and made changes to the MQTT parameters
- Selected save
- device hangs in wdt reset

If that's the case then it's normal as the device needs at least one reset after it has just been flashed due to esp8266 internal problem.

What happens after you reset the device and try to make the changes to MQTT?

Does it stop after a WDT or does it reboot?

If it reboots what is shown on the serial output?

arendst, 2017-07-01:
What is your fulltopic value?

LordVanguard, 2017-07-01:
Upload - Yes, I uploaded via a serial connection (not OTA or via the web interface)
Web - Yes it was through the web interface that I changed the topic and it reset but did come back up. I changed the Client ID and it reset but did not come back up. Unplugging the device and plugging back in yeilds nothing on serial and nothing via web. I had to reflash to get it to work again.

LordVanguard, 2017-07-01:
Topic was just to add the number after the sonoff name. More than happy to completely reflash and test with another if it helps. I have three of them to play about with.

arendst, 2017-07-01:
Strange. I cannot reproduce. Just used your client_id, user and password and it just reboots and comes online.

Try to increase logging by using command ``seriallog 3``.

arendst, 2017-07-01:
What's the output of command ``status 0``

LordVanguard, 2017-07-01:
Sure thing, 

OK I reflashed the unit to bring it back. Web page and all was available again. I changed the seriallog to 3 and went into the web page. The only thing I changed was changing the Client ID from "SONOFF_%06X" to "SONOFF03_%06X"

Comm output as follows after changing to seriallog 3
`22:18:30 CMND: seriallog 3
22:18:30 MQTT: stat/sonoff03/RESULT = {"SerialLog":"3 (Active 3)"}
22:18:30 Cnfg: Save (1492 bytes) to flash at 79 and count 3
22:18:58 HTTP: Handle root
22:19:01 HTTP: Handle config
22:19:04 HTTP: Handle MQTT config
22:19:14 HTTP: Parameter save
22:19:14 HTTP: MQTT Host 10.0.0.30, Port 1883, Client SONOFF03_%06X, User sensor-user, Password 331d46d9edc5b11266ef262fb0853fba, Topic sonoff03, FullTopic 
22:19:15 Cnfg: Save (1492 bytes) to flash at 78 and count 4
22:19:16 APP: Restarting

 ets Jan  8 2013,rst cause:1, boot mode:(1,6)

 ets Jan  8 2013,rst cause:4, boot mode:(1,6)

wdt reset`

the unit did not loop from this point. It just sits. I have to unplug it and then plug it back in for it to come back, but this time it does come back. When I change the Client ID again it does reset, however this time it comes back up without the need for unplugging and plugging back in. 

More than happy to tinker if there is anything else you would like me to test

LordVanguard, 2017-07-01:
status 0 yeilds

`22:25:03 CMND: status 0
22:25:03 RSLT: DataCb Group 0, Index 1, Type STATUS, Data 0 (0)
22:25:03 MQTT: stat/sonoff03/STATUS = {"Status":{"Module":1, "FriendlyName":"Sonoff03", "Topic":"sonoff03", "ButtonTopic":"0", "Power":0, "PowerOnState":3, "LedState":1, "SaveData":1, "SaveState":1, "ButtonRetain":0, "PowerRetain":0}}
22:25:03 MQTT: stat/sonoff03/STATUS1 = {"StatusPRM":{"Baudrate":115200, "GroupTopic":"sonoffs", "OtaUrl":"http://domus1:80/api/arduino/sonoff03.ino.bin", "Uptime":0, "Sleep":0, "BootCount":3, "SaveCount":7, "SaveAddress":"75000"}}
22:25:03 MQTT: stat/sonoff03/STATUS2 = {"StatusFWR":{"Program":"5.2.3", "BuildDateTime":"2017-07-01T22:15:08", "Boot":31, "Core":"2_3_0", "SDK":"1.5.3(aec24ac9)"}}
22:25:03 MQTT: stat/sonoff03/STATUS3 = {"StatusLOG":{"Seriallog":3, "Weblog":2, "Syslog":0, "LogHost":"sonoff03", "SSId1":"paddock", "SSId2":"scootaloo", "TelePeriod":300}}
22:25:04 MQTT: stat/sonoff03/STATUS4 = {"StatusMEM":{"ProgramSize":468, "Free":20, "Heap":28, "ProgramFlashSize":512, "FlashSize":1024, "FlashMode":2}}
22:25:04 MQTT: stat/sonoff03/STATUS5 = {"StatusNET":{"Host":"sonoff03-0084", "IP":"10.0.0.16", "Gateway":"10.0.0.138", "Subnetmask":"255.255.255.0", "DNSServer":"10.0.0.138", "Mac":"2C:3A:E8:05:60:54", "Webserver":2, "WifiConfig":3}}
22:25:04 MQTT: stat/sonoff03/STATUS6 = {"StatusMQT":{"Host":"10.0.0.30", "Port":1883, "ClientMask":"SONOFF04_%06X", "Client":"SONOFF04_056054", "User":"sensor-user", "MAX_PACKET_SIZE":512, "KEEPALIVE":15}}
22:25:04 MQTT: stat/sonoff03/STATUS7 = {"StatusTIM":{"UTC":"Sat Jul 01 21:25:04 2017", "Local":"Sat Jul 01 22:25:04 2017", "StartDST":"Sun Mar 26 02:00:00 2017", "EndDST":"Sun Oct 29 03:00:00 2017", "Timezone":1}}
22:25:04 MQTT: stat/sonoff03/STATUS10 = {"StatusSNS":{"Time":"2017-07-01T22:25:04"}}
22:25:04 MQTT: stat/sonoff03/STATUS11 = {"StatusSTS":{"Time":"2017-07-01T22:25:04", "Uptime":0, "Vcc":3.237, "POWER":"OFF", "Wifi":{"AP":1, "SSID":"paddock", "RSSI":100, "APMac":"08:60:6E:21:CE:98"}}}`

LordVanguard, 2017-07-01:
I went back in and changed the Client ID again from 04 to 03 and it looped ... results below

`22:25:03 CMND: status 0
22:25:03 RSLT: DataCb Group 0, Index 1, Type STATUS, Data 0 (0)
22:25:03 MQTT: stat/sonoff03/STATUS = {"Status":{"Module":1, "FriendlyName":"Sonoff03", "Topic":"sonoff03", "ButtonTopic":"0", "Power":0, "PowerOnState":3, "LedState":1, "SaveData":1, "SaveState":1, "ButtonRetain":0, "PowerRetain":0}}
22:25:03 MQTT: stat/sonoff03/STATUS1 = {"StatusPRM":{"Baudrate":115200, "GroupTopic":"sonoffs", "OtaUrl":"http://domus1:80/api/arduino/sonoff03.ino.bin", "Uptime":0, "Sleep":0, "BootCount":3, "SaveCount":7, "SaveAddress":"75000"}}
22:25:03 MQTT: stat/sonoff03/STATUS2 = {"StatusFWR":{"Program":"5.2.3", "BuildDateTime":"2017-07-01T22:15:08", "Boot":31, "Core":"2_3_0", "SDK":"1.5.3(aec24ac9)"}}
22:25:03 MQTT: stat/sonoff03/STATUS3 = {"StatusLOG":{"Seriallog":3, "Weblog":2, "Syslog":0, "LogHost":"sonoff03", "SSId1":"paddock", "SSId2":"scootaloo", "TelePeriod":300}}
22:25:04 MQTT: stat/sonoff03/STATUS4 = {"StatusMEM":{"ProgramSize":468, "Free":20, "Heap":28, "ProgramFlashSize":512, "FlashSize":1024, "FlashMode":2}}
22:25:04 MQTT: stat/sonoff03/STATUS5 = {"StatusNET":{"Host":"sonoff03-0084", "IP":"10.0.0.16", "Gateway":"10.0.0.138", "Subnetmask":"255.255.255.0", "DNSServer":"10.0.0.138", "Mac":"2C:3A:E8:05:60:54", "Webserver":2, "WifiConfig":3}}
22:25:04 MQTT: stat/sonoff03/STATUS6 = {"StatusMQT":{"Host":"10.0.0.30", "Port":1883, "ClientMask":"SONOFF04_%06X", "Client":"SONOFF04_056054", "User":"sensor-user", "MAX_PACKET_SIZE":512, "KEEPALIVE":15}}
22:25:04 MQTT: stat/sonoff03/STATUS7 = {"StatusTIM":{"UTC":"Sat Jul 01 21:25:04 2017", "Local":"Sat Jul 01 22:25:04 2017", "StartDST":"Sun Mar 26 02:00:00 2017", "EndDST":"Sun Oct 29 03:00:00 2017", "Timezone":1}}
22:25:04 MQTT: stat/sonoff03/STATUS10 = {"StatusSNS":{"Time":"2017-07-01T22:25:04"}}
22:25:04 MQTT: stat/sonoff03/STATUS11 = {"StatusSTS":{"Time":"2017-07-01T22:25:04", "Uptime":0, "Vcc":3.237, "POWER":"OFF", "Wifi":{"AP":1, "SSID":"paddock", "RSSI":100, "APMac":"08:60:6E:21:CE:98"}}}
22:26:41 HTTP: Handle root
22:26:43 HTTP: Handle config
22:26:49 HTTP: Handle other config
22:26:59 HTTP: Handle config
22:27:04 HTTP: Handle MQTT config
22:27:15 HTTP: Parameter save
22:27:15 HTTP: MQTT Host 10.0.0.30, Port 1883, Client SONOFF03_%06X, User sensor-user, Password 331d46d9edc5b11266ef262fb0853fba, Topic sonoff03, FullTopic 
22:27:16 Cnfg: Save (1492 bytes) to flash at 74 and count 8
22:27:17 APP: Restarting

 ets Jan  8 2013,rst cause:1, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v09826c6d
~ld

 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v09826c6d
~ld

 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v09826c6d
~ld
`

arendst, 2017-07-01:
22:19:16 APP: Restarting
ets Jan 8 2013,rst cause:1, boot mode:(1,6)
ets Jan 8 2013,rst cause:4, boot mode:(1,6)
wdt reset`
the unit did not loop from this point. It just sits. I have to unplug it and then plug it back in for it to come back, but this time it does come back. When I change the Client ID again it does reset, however this time it comes back up without the need for unplugging and plugging back in.

As I tried to explain before: The unit will ALWAYS sit and wait after a flash via serial interface. Common issue with esp8266. ALWAYS perform a reset AFTER a serial flash.

arendst, 2017-07-01:
Got it! You did not select the correct flash size in your IDE. You are using a 512k flash setting. Change it to 1M (no spiffs) and all should be well.

LordVanguard, 2017-07-01:
Will try that right now and feed back.

LordVanguard, 2017-07-01:
Just to let you know that I have made the changes as you suggested and all is indeed working a charm. For some reason I thought it was 512. 

The problem in this instance is situated between my chair and my keyboard ;) 

Thank you again for your time, it is most definitely appreciated !!!

LordVanguard, 2017-07-01:
To others that may have fallen foul of their own doing and are at this thread, 

READ https://github.com/arendst/Sonoff-Tasmota/wiki/Upload 

Thank you again arendst !!

closed 2017-07-01 by LordVanguard

Th30wl, 2018-01-28:
Hi,

I am also having the same issue with T1 3ch, so I'm trying to figure out if I did something wrong or I just have faulty unit.
First, I have flashed Tasmota over the stock firmware using SonOTA and flashing went fine but then it began rebooting and resetting the settings to default just as @LordVangard described. Same thing happens with ESPEasy firmware. Erasing the flash before uploading does not change the situation either. 
It seems that there is something wrong with its WiFi connection, it connects to WiFi and after some time it looses the connection and reboots. I have uploaded a sketch that does not connect to WiFi and it does not reboot at all.

Here is the info about what I am using:
**platformio build flags:** 
-Wl,-Tesp8266.flash.1m0.ld; lib_deps: PubSubClient, NeoPixelBus, IRremoteESP8266, ArduinoJSON; platform: espressif8266; board: esp01_1m; framework: arduino; monitor_baud: 115200; board_flash_mode: dout
**WiFi router settings:**
Wireles mode: Auto, b/g protection
Control channel: Auto
Extension channel: Auto
Authentication method: WPA-Personal
Encryption: AES
[Serial log](https://github.com/arendst/Sonoff-Tasmota/files/1671555/sonoffT13ch.txt)

ChipStewart1, 2018-02-03:
I'm having the same (or similar) problem.  I'm still a bit of a noob and can't seem to figure it out, despite trying everything I could.  I've tried the suggestions in dozens of posts, but keep ending up with the same problem.
I bought 4 Sonoffs and was able to flash 3 of them without (much) incident.  The fourth, however, keep rebooting with the following in serial monitor:

	 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

	wdt reset
	load 0x4010f000, len 1384, room 16 
	tail 8
	chksum 0x2d
	csum 0x2d
	v4ceabea9
	~ld

	00:00:00 Project sonoff Sonoff (Topic sonoff, Fallback DVES_3C0D77, GroupTopic sonoffs) Version
	 5.11.1h
	00:00:00 WIF: Connecting to AP1 The Dog House in mode 11N as sonoff-3447...

Is this just a bad unit?  I've tried changing ports, changing flash size  with and without spiffs, tried different upload speeds, I've used NodeMCU flasher to flash blank files, I've tried PlatformIO, but couldn't seem to get it to work for me and reverted to Arduino IDE, tried changing various parameters in user_config.h . . . in short, everything I could find or think of.
Is there anything else you could recommend?  Or is this one just plain dead?

Th30wl, 2018-02-04:
I have uploaded a simple sketch that just connects to wifi and it seems to work. No boot loop and it stays connected. So there's probably still some compatibility issue with T1 and Tasmota or one of the libraries it uses.

ChipStewart1, 2018-02-04:
What’s the simple sketch?

 

Chip