arendst/Tasmota #572: PulseTime works not everytime correct on Sonoff Dual

scharrrfi, 2017-07-01:
Hi there,

I use a Sonoff Dual to control my roller shutter. This works great so far. But my PulseTime1 and PulseTime 2 values do not work everytime reliable. 
So it is posible, that the relais is still active after the time. I see this on the Sonoff LEDs and I hear no clicking when the relais should open again. The strange on this is, that the Sonoff thinks the relais status is changed. So it gives this information in the GUI and over MQTT. 

Can you please check this behavior? It is a big issue for me. 
When I do not check the relais status manually and would try to switch the second relais, the roller shutter will try to go up and down at the same time. This will kill the motor.

arendst, 2017-07-02:
Did you disable serial logging as it interferes with relay control. See wiki.

scharrrfi, 2017-07-02:
Ah thank you! 
It was on level 2. I disabled it now on all devices and will test, if it solve the issue.

tom211, 2017-08-15:
Hi,
same problem here. 
I am testing a sonoff dual for about 10 weeks to control a roller shutter. I implemented the additional code for the motor block as described in #309 . Additionally I have set the pulsetime according to the runtime of the motor plus a few seconds. I thougt this could be a nice feature, because I dont´t have to send the stop command and the relay is only on for the needed time and I can use retained messages. Also I disabled the serial logging. 

One day I noticed that the red led and the corresponding relay was still on after the shutter was opened. 
The second time this behavior came up was after it was disconnected from the mains and the retained message came in. I thought it could be possible that serial logging is still on after power down and checked it another time. Serial logging is still set to level 0. I haven´t tested the power off behavior another time yet.

Then it worked for about 2 weeks without any problems.
Today the shutter opened and the relay is stuck in the on position. After sending the "on" Message another time, the pulsetime worked correct and the relay was set to off.

I have implemented a "stop" command before an up or down message is generated so that wont kill my motor but a perfect working pulsetime would be great!

Any ideas or suggestions what could be tested to narrow down the problem?

davidelang, 2017-08-15:
start by increasing the logging level to "more debugging" and see what happens 
when you run into problems.

arendst, 2017-08-15:
See my comment in #721 

_Serial interface is being used by relay control. That's also a good reason not to use a Dual for interlocked motor control as you'll someday loose communication between the esp8266 and the relay controller and both relays will be on._

tom211, 2017-08-15:
Thanks for the comments. 
I will increase the logging level and then have to wait until it gets stuck another time.

Weeks ago I thought it could work with the duals but today im thinking about using [this](https://luani.de/projekte/esp8266-hvio/) for the shutter control because I want to use hardware switches as well and I think the header for the switches in the dual are directly connected to the relay controller so that the interlock and pulsetime will be not working for that too. 

I hope that it is not to complicated to define it as a new Module type for the use with sonoff tasmota.

tom211, 2017-09-22:
The dual is working without problems in the last weeks.
I think it has to do with retained messages that are coming in before pulsetime is set after a restart of the sonoff like it is described in #628. I think i have to disable the use of retained messages.

jponte85, 2017-10-18:
Hello, 
From what I could read so far, it is possible to control roller shutter with a sonoff dual, is this correct?
I bought one but so far I didn't receive it. My question is, to controll the roller shutter I need to flash my sonoff with the tasmota firmware and in terms of hardware I don't need anything else, is this correct?

Thank you!

giig982, 2017-10-18:
@jponte85 if you don't have any other button to control roller shutter, that's may be true. But if you want multiple point of command, real configuration may vary. 

In any case Sonoff Dual relays contact can stuck by arc fault. Let me explain this.

An electro-mechanical relay, such as the ones used in the Sonoff, is simply a remote controlled mechanical switch. To open a circuit, the relay just move the working contacts away. If there is a coil in the cut circuit (and roller shutter motor has a coil), a voltage spike appears at the coils ends, this spike propagates to the relay work contacts and causes an electric arc. If strong enough, this electric arc will damage the relay contacts. In the most severe cases the arc might jump across the relay isolation and reach the control contacts, causing a disaster in the control electronics. 

The coil doesn't have to be very big to cause severe surges. For instance a small electro-mechanical actuator can generate quite an high voltage. Indeed, an electro-mechanical actuator is just a coil with a metallic rod sliding across it when the magnetic field is activated. Here is a small actuator, which, when driven with 12V only, can generate a -500V surge when its power source is brutally shut down. 

Fortunately, a small electronic component can help us the solve this issue. The TVS diode (Transient Voltage Suppressor) is used in telecommunication equipments to suppress voltage surges, quite frequent in these applications. These diodes are blocking under a specific voltage, and conductive above. The idea is to short the coils ends with a TVS diode. The current from the voltage spike is then trapped in the coil/diode circuit, and is dissipated in the form of heat and no damage is caused to the rest of the circuit. As bi-directional TVS diode works both ways they can be used with both DC and AC. 
(from http://www.yoctopuce.com/EN/article/relay-and-inductive-loads)

If you want to have multiple poin of command you may have to use other interlocked relay, etc..

jponte85, 2017-10-18:
@giig982 , Thanks for your comments.
I was trying to be proactive because so far I didn't receive the sonoff neither the roller shutter, but are both already ordered. So, it seems I have many variables, I need to check my scenario better. Thanks

giig982, 2017-10-18:
With interblocked command buttons, you can use relays as in this image (that's mine)
https://scontent-mxp1-1.xx.fbcdn.net/v/t1.0-9/21766454_10213169980820171_6167750329674873236_n.jpg?oh=cbe200ebca259158b31b45d6f46a7d1b&oe=5A84CCEC

jponte85, 2017-10-18:
My knowledge with electronic equipments isn't too much, basically I just understand Line/neutral, Voltage/ampere. :) But just one question more, are you using any automation system, like Home assistant, domoticz? 
I don't want to bother anyone else, once I have my equipments, I'll analyse better my scenario.

Grazie mille @giig982 .

giig982, 2017-10-18:
Yes I do. I use Home Assistant on a raspberry pi to control Sonoff Device and other automation systems in my house (Xiaomi Yeelight; Broadlink rm, etc).

If you want, send me an email (you can found it on my profile) to not bother other people about those electronic topics.

Figurati @jponte85 ;)

mberlier, 2018-01-17:
Hi @giig982, I'm trying to do the same with sonoff dual; just new to sonoff, just ordered,  and I'll leverage all the post to learn about. I just wonder if in your scheme could miss something. I mean I guess COM of relay-up should be connected to L; otherwise when you close UP button nothing is sent to NO, am I wrong? thank you!

giig982, 2018-01-17:
Hi @mberlier. You're right, and that connection is present but it collapsed when i drawed the schema, over the relay bottom border: if look closely,  you can see the connections points on COM and the L path on the right of the UP relay.

mberlier, 2018-01-17:
Sorry @giig982 , now sounds good!! Anyway I still miss something. When you press any button sonoff looses L, right? does it switch off?

giig982, 2018-01-18:
That's true. This solution keeps physical button priority over sonoff commands, also for security reason: think about somebody passing by under the shutter when the sonoff is shutting it, this way with the physical button you can always bypass that command.

mberlier, 2018-01-18:
Got it, yes, good for security reasons. Hoping it doesn’t suffer the continuous switch on and off due to tequent use of the buttons. Appreciated your commments @giig983

mberlier, 2018-01-20:
Hi @giig982, hi all. I share my variant. I think it could be better to keep sonoff always swithed on; moreover the original version could not fit well when you have to control multiple shutters at the same time, as in my case. Sonoff Output power could not be enough. So my idea would be to directly connect sonoff L1 and L2 to buttons (to trigger relays). As up relay, up button, has always priority, if sonoff is moving down the shutters and for any reason you need manually to block it, you have just to press the up button (as you would do by instinct). Does it make sense? I hope it can work. Tomorrow I'll work on sonoff for the first time, hope it will be simple to program. thank you Mirko 
[image]

closed 2018-06-03 by stale[bot]