arendst/Tasmota #573: Mqtt Help please

mydakota, 2017-07-02:
Ive beent trying to get the mqtt comands frome home assistant to turn on and off but does not seem to work im very new to mqtt so probibly making mistakes please can some one help me .

[image]

switch:
  - platform: mqtt
    name: "Master Bedroom"
    state_topic: "sonoff"
    command_topic: "sonoff/on"
    qos: 1
    payload_on: "ON"
    payload_off: "OFF"
    retain: true

mydakota, 2017-07-02:
I've tried adding cmnd and stat into topic and command but still nothing .
[]10:04:06 CMND: POWER":"ON
10:04:06 MQTT: switch/sonoff/on/RESULT = {"Command":"Unknown"}
10:04:17 APP: Serial logging disabled
10:04:20 CMND: POWER on
10:04:20 MQTT: switch/sonoff/on/RESULT = {"POWER":"ON"}
10:04:20 MQTT: switch/sonoff/on/POWER = ON
10:04:30 CMND: Power off
10:04:30 MQTT: switch/sonoff/on/RESULT = {"POWER":"OFF"}
10:04:30 MQTT: switch/sonoff/on/POWER = OFF
10:04:33 MQTT: switch/sonoff/on/STATE = {"Time":"2017-07-02T10:04:33", "Uptime":1, "Vcc":3.177, "POWER":"OFF", "Wifi":{"AP":1, "SSID":"ParkwayDrive", "RSSI":66, "APMac":"C0:25:06:DC:E1:E8"}}
10:04:33 MQTT: switch/sonoff/on/SENSOR = {"Time":"2017-07-02T10:04:33", "Switch3":"ON"}
10:04:42 CMND: Power off
10:04:42 MQTT: switch/sonoff/on/RESULT = {"POWER":"OFF"}
10:04:42 MQTT: switch/sonoff/on/POWER = OFF
10:04:55 CMND: Power toggle 
10:04:55 MQTT: switch/sonoff/on/RESULT = {"POWER":"OFF"}
10:04:55 MQTT: switch/sonoff/on/POWER = OFF
10:05:05 CMND: Power toggle 
10:05:05 MQTT: switch/sonoff/on/RESULT = {"POWER":"OFF"}
10:05:05 MQTT: switch/sonoff/on/POWER = OFF[\]

mydakota, 2017-07-03:
**below code does nothing** 
`light:
  - platform: mqtt 
    name: "sonoff"
    state_topic: "stat/sonoff/power"
    command_topic: "cmnd/sonoff/power"
    qos: 1
    payload_on: "on"
    payload_off: "off"
    optimistic: false
    retain: true`

**But in this format bellow I at least get the sonoff to turn on but relay/sonoff stays in the on position but in home assistant the switch turns off after 1-3 seconds after press and cant turn relay off unless I manually turn it of via tasmota's web interface ** 

`light:
  - platform: mqtt 
    name: "sonoff"
    state_topic: "sonoff/stat/power"
    command_topic: "sonoff/cmnd/power"
    qos: 1
    payload_on: "on"
    payload_off: "off"
    optimistic: false
    retain: true`

mydakota, 2017-07-03:
reloaded defaults user_config except for WiFi ssid and PW 
added below config.yaml

`light:
  - platform: mqtt 
    name: "masterbed"
    state_topic: "stat/sonoff/power"
    command_topic: "cmnd/light_masterbed/sonoff/power"
    qos: 1
    payload_on: "on"
    payload_off: "off"
    optimistic: false
    retain: true`

[image]

4 Days on this FW and nothing should be some clear instructions

joecotton, 2017-07-03:
On the Sonoff with Tasmota:
Change the "topic" to "light_masterbed"
Change the Full Topic to the default of "%prefix%/%topic%/"

On your homeassistant:
state_topic: "stat/light_masterbed/power"
command_topic: "cmnd/light_masterbed/power"
(The only difference is the start, 'stat' vs 'cmnd')

The %prefix% is the sonoff's message type [stat,cmnd,tele]
The %topic% is the MQTT topic setting

The 3rd segment is the actual command or state being reported. In this case "power".

mydakota, 2017-07-03:
thank you joecotton,
I was starting to think I was talking to myself lol thank you so much for getting back to me with this info  much appreciated. I have tried what you said and are having same results with switch in Home assistant turning off automatically and only triggering the power on command  on sonoff. before following your instructions I re-flashed the Sonoff with the default unmodified Tasmota firmware  using the Atom software. once online I followed your easy instructions and have the same issue.
I am not sure whats causing it  I am running Mosquito MQTT on my local raspberry pi 3 that also host my Home Assistant software I have no issues with other devises like my Wemos R2 with MQTT.
also my Sonoff unit is the basic single relay version as in and have been flashing the
 (env_default = sonoff) version of Tasmota would you have any ideas what could be causing this thanks.

mydakota, 2017-07-03:
Also note all cmnd and stat work to trigger sonoff via Console 
So I'm picking there's nothing wrong with the sonoff and other devices work correctly 
With my MQTT setup I have also  tried this cofig on a default / fresh install of HASS and 
MQTT just for in case but still having same results.

joecotton, 2017-07-03:
In this case, you should get to debugging. Use something like MQTTspy to examine what's being sent to/from the MQTT server and clients.

mydakota, 2017-07-05:
I've run mqtt through a verities of test and using multiple devices also using the sonoff with arduino coded script and all works 100% 
With commands and status . But as soon as I use tasmota firmware it dose not get status updates .
From what I've read up is basically Tasmota is not compatible with HASS due something to do with retaining power commands or something like that a bit above my head but seems there's a lot of people having this same issue using The firmware on Home Assistant. I wish I had of known it's not compatible with Home Assistant before spending the last week getting a sonoff to function correctly. I guess I'll put it down to a lesson learnt. Has anyone got any ideas on the best firmware for the sonoff and Home Assistant I basically just want a basic on off relay via HASS and either use gpio14 to toggle the power state so I can use it to interface with a Standen light switch

mydakota, 2017-07-05:
Out of interest why does both of these commands below  using the Tasmota web console activate the relay when one is a state the other a command.
Cmnd/sonoff/power on
Stat/sonoff/power on 
Or
Cmnd/sonoff/power off
Stat/sonoff/power off

Am I missing something I would have thought stat would not change the relay on or off

joecotton, 2017-07-05:
Regarding the last comment about the web console: It is not meant to type MQTT topics into the web console. The MQTT message/topic, in the default setup, is composed of 3 parts:
%prefix% is the message type, of which there are 3:

- "cmnd" -> Command
- "stat" -> Status
- "tele" -> Telemetry

The next part is the "topic" which is normally the name of the device as defined in the settings. Default is just "sonoff". (You can also set some other topics for other purposes, but this is the basic.)

The last part in a "command" message is the actual command like "power" or "restart"

When you use the web console, things you type are only commands. What you see in the console window are all messages, including your command and the resulting STAT message and any TELE messages.

When you put the entire MQTT topic string, I suspect that the sonoff is only looking at the last part "power on" or "power off". 

Regarding HomeAssistant: I'm not certain Tasmota is incompatible, but there are many different combinations of QoS and Retain settings for: The Sonoff-Tasmota device; the MQTT broker; and the HomeAssistant server. You must understand what QoS and Retain do, and decide which part of your system you want to be "in charge" of holding and setting the state if any one component drops out. But mostly this comes up only when one part (the MQTT broker, the Sonoff device, or the HomeAssistant software) is restarted or otherwise becomes unavailable temporarily. I suspect you have some other issue, since you aren't dealing with a restart of any of the components.

Looking at and/or sharing with experts the logs from any/all components of your system will be the most help in finding success. Showing configuration pages is okay, but not enough for a good analysis of the problem. Logs show what *really* is happening on each component.

mydakota, 2017-07-06:
Thank you,
for all your help sorry I am a bit of a noob lol
but thanks to you I have learned a lot and have good news to report I think I have it working with the below config yay.

`mqtt:
  broker: 10.0.0.23
  port: 1883
  client_id: home-assistant-1                                     I think this is what has fixed it
  keepalive: 60
  username: fake name
  password: coolpassword
  protocol: 3.1
 
light:
  - platform: mqtt
    name: "masterbed"
    state_topic: "stat/sonoff/POWER"
    command_topic: "cmnd/sonoff/power"
    payload_on: "ON"
    payload_off: "OFF"
    optimistic: false
    retain: true`

lmamakos, 2017-07-08:
I think what fixed it is you changed the `state_topic` as compared to what you posted previously.  The MQTT topic names are case sensitive.  You had `power` before and now `POWER` in the version that works.

mydakota, 2017-07-08:
no i changed the 
state_topic: "stat/sonoff/POWER"
command_topic: "cmnd/sonoff/power"
to many different combinations as the above did not change the issue i was having.
it was either.

1: the optimistic: false
          retain: true
or 
2: client_id: home-assistant-1 
   keepalive: 60

also from my test with commands inside the tasmota OTA console,
both Capitals or lower case commands work fine both POWER and power act the same and the same for ON on
OFF off  
I have also tested both on my working config and they either way works correctly

closed 2017-07-08 by mydakota

brberis, 2017-08-20:
+1 Thanks mydakota.

mixu-, 2018-07-21:
I'm having the exact same issue with _one_ of my Sonoff S20 sockets. Three Sonoff basic relays and another S20 are working flawlessly. It's just this one S20 that refuses to turn off via Home Assistant.

If I send an OFF command with MQTT-spy, the relay works fine. MQTT-spy also never sees an OFF message coming from home assistant. Apparently HA just doesn't send the OFF-command for some reason.

mixu-, 2018-07-21:
For some reason, switching payload to 1 and 0 instead of ON and OFF worked and fixed my problem. On the Sonoff side I used TasmoAdmin to change the State Text: Off/On to 0 and 1 respectively. 

The TasmoAdmin part does seem to be necessary for this to work. If I change the State text setting back to OFF/ON, it stops working again.

  - platform: mqtt
    name: "Socket"
    state_topic: "tasmota/sonoff-socket-2/stat/POWER"
    command_topic: "tasmota/sonoff-socket-2/cmnd/POWER"
    qos: 1
    payload_on: **"1"**
    payload_off: **"0"**
    retain: true