arendst/Tasmota #574: How PulseTime works?

cnieves1, 2017-07-02:
Hi, 
I am testing Tasmota using WemosD1. I followed this webpage and it worked very well:
  https://github.com/tIsGoud/Tasmota-on-WEMOS

I want to have a motion detector connected to GPIO2 This signal is low while motion is detected, and it goes high after 0.5-1s of no motion detected. I'm trying to switch on the relay as soon as motion is detected, and switch off the relay some configurable time after no motion is detected anymore. Noticed that if motion is detected while the timer is running, the timer should be stopped and restarted again in the next low to high transition. 

It's something like "Resettable Single Shot / Watchdog" in: http://www.airotronics.com/AOTpdfs/multifunction/Airotronics-MC363.pdf (but my control input is active low instead of active high as shown in that pdf).

For testing, I have a switch connected from D5 (GPIO2) to ground. I set it up as Button 1 in the configuration page. I use the LED in the board as the relay (GPIO12 is Relay1). The behaviour I get is that LED follows the switch state.

Then, I tried to set PulseTime1, so I did a command using web interface: http://192.168.0.22/cm?cmnd=PulseTime1%2050. I got "RESULT = {"PulseTime1":50}" as the answer.

The behaviour is still the same. The LED follows switch state.
I found no documentation about PulseTime, other than the command list. Did I miss something to get this working?

Thanks in advance,

Carlos

arendst, 2017-07-02:
PulseTime works with switches only, not buttons

cnieves1, 2017-07-04:
I tried using switches (Switch1 on GPIO2, Relay1 on GPIO14) and PulseTime. It works as a timer for maximum ON time if the switch doesn't change, but this is not what I wanted to do... When the relay is on and the switch changes the relay is switched off. 
What I wanted to do is a delay only when switching off, but switching on instantaneously. Is there any other way to do this? 

TIA,

Carlos

stefanbode, 2017-07-04:
Hi Carlos,

normally this is not supported, but there are so called Relay1I the I does mean this is an INVERTED relay. You should give it a try. I assume it will do what you want, but the message on the webbrowser and mqtt could be somewhere confusing.

stefanbode, 2017-07-04:
Hi Carlos,

this is not supported. I was thinking about an inverted relay, but this will also do not the job. The implemented pulse time switched directly and then hold it for a while. Your request is to delay the switch for some time. This completely different.

closed 2018-05-07 by stale[bot]