arendst/Tasmota #575: Additional Sensors for Sonoff POW

Stalkk, 2017-07-03:
Hi,

in the early version (~4.2) was possible to assign additional sensors on Serial GPIO's of POW. Now it's not possible anymore. Does it have any special reason? Is it because of connection between serial and the AC power line? If yes, i would write the warning on the POW site (like now for flashing without AC) just to be sure nobody attach any sensors on it.

Regards,
Stalkk

arendst, 2017-07-03:
It's because of the connection between serial and AC power lines. Pow page just updated.

Stalkk, 2017-07-03:
Thank you.

i would also add text like "Also do not connect any additional sensors to serial pins until you are 100% sure. It can at least destroy yout Pow!" to the page.

closed 2017-07-03 by Stalkk