arendst/Tasmota #576: Help with H801 and RGBW via MQTT

Blue407, 2017-07-03:
New to all this, so please be gentle :)

I have successfuly configured my H801 and reflashed it with Sonoff-Tasmota.

I have tested commands like:-
mosquitto_pub -r -t cmnd/H801-1/PWM4 -m "1000"

Which work fine.

Is there a way to set multiple colour values at the same time in one line?
Or do I have to issue 4 separate colour commands every time?

I plan on using this with Openhab, so any advice specific to that would be appreciated :) :)

arendst, 2017-07-03:
Currently there is only single pwm control.

Blue407, 2017-07-03:
Is there any chance of multiple PWM control at a later date?

Any advice on how best to use it to pick colours based on its current limitation?

cobra85ec, 2017-07-05:
are u using node-red..?

Blue407, 2017-07-05:
No, OpenHAB.

kruebsam, 2017-07-11:
Would also be interested in the above with exactly same config

closed 2018-05-07 by stale[bot]