arendst/Tasmota #579: Time not correct when using "APP_TIMEZONE 99"

AffeP, 2017-07-04:
Compiled with the following code:
#define TIME_DST               Last, Sun, Mar, 2, +120  // Last sunday in march at 02:00 +120 minutes
#define TIME_STD               Last, Sun, Oct, 3, +60   // Last sunday in october 02:00 +60 minutes
#define APP_TIMEZONE           99                // [Timezone] +1 hour (Amsterdam) (-12 .. 12 = hours from UTC, 99 = use TIME_DST/TIME_STD)

But it does not work, gives me TIME_STD even when it should be TIME_DST?

Great work by the way!!
:-)

arendst, 2017-07-04:
What's your current timezo e value (command timezone)

AffeP, 2017-07-04:
{"Timezone":1}

arendst, 2017-07-04:
Change it to 99 and you're on time. No need to fool around with user_config.h. It's so 2016...

AffeP, 2017-07-04:
Worked fine!!

AffeP, 2017-07-04:
Should it not have worked the other way to?
It is ok now when I know the backdoor way.

closed 2017-07-04 by AffeP