arendst/Tasmota #580: MQTT connection drops / is reestablished every 2 seconds

hdodenhof, 2017-07-04:
First off: thanks for your work on this - it's exactly what I've been looking for!

I built v5.2.3 with minimal changes to the configuration (Wifi only) and successfully flashed it to an S20 socket. Using the web interface I set it up to connect to my local MQTT server (on an RPI) without any issues - I'm able to send commands to / receive status updates from the socket.

Problem now is that the MQTT connection drops / is reestablished every 2 seconds and so far I've no idea why. The MQTT server (Mosquitto) logs `Socket error on client SONOFF_F912C4, disconnecting` but I wasn't able to find a possible cause for that yet. Both devices (Sonoff, RPI) are on the same network and are reachable at all times without any package loss.

Any hint on where to look at to get to the bottom of this? Thanks!

Sonoff-Logs (Level 4, web interface):
```
14:21:50 MQTT: Attempting connection...
14:21:51 mDNS: Query done with 0 mqtt services found
14:21:51 MQTT: Connected
14:21:51 MQTT: coffeemaker/tele/sonoff/LWT = Online (retained)
14:21:51 MQTT: coffeemaker/cmnd/sonoff/POWER = 
14:21:51 RSLT: Receive topic coffeemaker/cmnd/sonoff/power, data size 3, data off
14:21:51 RSLT: DataCb Group 0, Index 1, Type POWER, Data off (OFF)
14:21:51 RSLT: coffeemaker/stat/sonoff/RESULT = {"POWER":"OFF"}
14:21:51 RSLT: coffeemaker/stat/sonoff/POWER = OFF

14:21:52 MQTT: Attempting connection...
14:21:53 mDNS: Query done with 0 mqtt services found
14:21:53 MQTT: Connected
14:21:53 MQTT: coffeemaker/tele/sonoff/LWT = Online (retained)
14:21:53 MQTT: coffeemaker/cmnd/sonoff/POWER = 
14:21:53 RSLT: Receive topic coffeemaker/cmnd/sonoff/power, data size 3, data off
14:21:53 RSLT: DataCb Group 0, Index 1, Type POWER, Data off (OFF)
14:21:53 RSLT: coffeemaker/stat/sonoff/RESULT = {"POWER":"OFF"}
14:21:53 RSLT: coffeemaker/stat/sonoff/POWER = OFF

[..]
```

Mosquitto-Logs:
```
1499172984: New connection from 192.168.178.58 on port 1883.
1499172984: New client connected from 192.168.178.58 as SONOFF_F912C4 (c1, k15, usonoff).
1499172984: Sending CONNACK to SONOFF_F912C4 (0)
1499172984: Received PUBLISH from SONOFF_F912C4 (d0, q0, r1, m0, 'coffeemaker/tele/sonoff/LWT', ... (6 bytes))
1499172984: Received PUBLISH from SONOFF_F912C4 (d0, q0, r0, m0, 'coffeemaker/cmnd/sonoff/POWER', ... (0 bytes))
1499172984: Received SUBSCRIBE from SONOFF_F912C4
1499172984: 	coffeemaker/cmnd/sonoff/# (QoS 0)
1499172984: SONOFF_F912C4 0 coffeemaker/cmnd/sonoff/#
1499172984: Sending SUBACK to SONOFF_F912C4
1499172984: Sending PUBLISH to SONOFF_F912C4 (d0, q0, r1, m0, 'coffeemaker/cmnd/sonoff/power', ... (3 bytes))
1499172984: Received SUBSCRIBE from SONOFF_F912C4
1499172984: 	coffeemaker/cmnd/sonoffs/# (QoS 0)
1499172984: SONOFF_F912C4 0 coffeemaker/cmnd/sonoffs/#
1499172984: Sending SUBACK to SONOFF_F912C4
1499172984: Received SUBSCRIBE from SONOFF_F912C4
1499172984: 	cmnd/SONOFF_F912C4/# (QoS 0)
1499172984: Sending SUBACK to SONOFF_F912C4
1499172984: Socket error on client SONOFF_F912C4, disconnecting.

1499172986: New connection from 192.168.178.58 on port 1883.
1499172986: New client connected from 192.168.178.58 as SONOFF_F912C4 (c1, k15, usonoff).
1499172986: Sending CONNACK to SONOFF_F912C4 (0)
1499172986: Received PUBLISH from SONOFF_F912C4 (d0, q0, r1, m0, 'coffeemaker/tele/sonoff/LWT', ... (6 bytes))
1499172986: Received PUBLISH from SONOFF_F912C4 (d0, q0, r0, m0, 'coffeemaker/cmnd/sonoff/POWER', ... (0 bytes))
1499172986: Received SUBSCRIBE from SONOFF_F912C4
1499172986: 	coffeemaker/cmnd/sonoff/# (QoS 0)
1499172986: SONOFF_F912C4 0 coffeemaker/cmnd/sonoff/#
1499172986: Sending SUBACK to SONOFF_F912C4
1499172986: Sending PUBLISH to SONOFF_F912C4 (d0, q0, r1, m0, 'coffeemaker/cmnd/sonoff/power', ... (3 bytes))
1499172986: Received SUBSCRIBE from SONOFF_F912C4
1499172986: 	coffeemaker/cmnd/sonoffs/# (QoS 0)
1499172986: SONOFF_F912C4 0 coffeemaker/cmnd/sonoffs/#
1499172986: Sending SUBACK to SONOFF_F912C4
1499172986: Received SUBSCRIBE from SONOFF_F912C4
1499172986: 	cmnd/SONOFF_F912C4/# (QoS 0)
1499172986: Sending SUBACK to SONOFF_F912C4
1499172986: Socket error on client SONOFF_F912C4, disconnecting.

[..]
```

Subscriber on `coffeemaker/#`:
```
coffeemaker/tele/sonoff/LWT Online
coffeemaker/cmnd/sonoff/POWER (null)
coffeemaker/tele/sonoff/LWT Offline

coffeemaker/tele/sonoff/LWT Online
coffeemaker/cmnd/sonoff/POWER (null)
coffeemaker/tele/sonoff/LWT Offline

[..]
```

ionciubotaru, 2017-07-05:
It looks like a mosquitto instalation problem. First check if you install the latest version:
```
$mosquitto -v
mosquitto version 1.4.12 (build date Thu, 25 May 2017 22:38:16 +0100) 

```
If you have the latest version, try with a public mosquitto server like test.mosquitto.org or cloudmqtt to check if the problem persists.

davidelang, 2017-07-06:
we've seen things like this when you have multiple sonoff devices reporting the 
same name to the mosquitto server (when the second one connects, mosquitto 
thinks it's the first one on a new IP, so it disconnects the first one)

Courty40, 2017-07-07:
also, disable auto-discovery/mDNS on the config, this does allsorts of nasties to my setup including the above

hdodenhof, 2017-07-08:
Ha, that was too easy. Updating my Mosquitto installation seems to have done the trick, both my Tasmota clients now have a stable connection.

Disabling mDNS also sounds like a good idea. While it doesn't seem to cause any issue right now, I've also seen all kinds of weird behavior in other environments.

Thanks again!

closed 2017-07-08 by hdodenhof

LakeCruiser, 2018-02-28:
I had to  clear my EEPROM before I could get mine to work. Then I reloaded Tasmota with the latest version and it worked.