arendst/Tasmota #582: Compatibility with T1

ageorgios, 2017-07-05:
Is Tasmota compatible with Sonoff T1? 
If not, will it ever be?
It is great for motored shades.

https://www.itead.cc/sonoff-t1.html

arendst, 2017-07-05:
See README.md

ageorgios, 2017-07-05:
cool you are great!

timmo001, 2017-08-14:
What's the progress on this? I'm going to buy a 2 gang switch ready and was wondering when I should expect to be able to flash this switch

cputoaster, 2017-08-15:
I already ordered one too. And it looks like people already tinkered with something similar: https://github.com/arendst/Sonoff-Tasmota/issues/495 .

chmihai, 2017-08-16:
@arendst 
First, thank you for the incredible work on Sonoff support! 
Secondly, I got today 2 & 3 channels T1, therefore I am looking forward for the T1 support :)

arendst, 2017-08-16:
Well I received my T1 3CH today. Investigating started...

arendst, 2017-08-16:
Investigation paused. It seems the Sonoff T1-UK 3CH is just a Sonoff 4CH without relay 4. So people wanting to use the T1 with Tasmota can start using it as a Sonoff 4CH.

ferbulous, 2017-08-21:
@arendst Should it be sonoff 4ch or sonoff 4ch pro?
Since it has rf

arendst, 2017-08-21:
It is sonoff t1 1ch , 2ch or 3ch in the latest release.

RobertSzabo, 2017-08-23:
Hi,
I see that t1 1, 2, 3 ch versions are supported (great work THANKS!). Are there any hardware preparation pics available for the T1 variants? I wonder how to enter into flash mode. What is the purpose of the S1 switch?
Thanks

[image]
[image]

cputoaster, 2017-08-23:
I flashed via the OTA method, see my tutorial page at tasmota.

juan11perez, 2017-08-25:
Can anyone point the GPIO0 contact on the above picture. Thank you

juan11perez, 2017-08-25:
Ok Andres, followed your tutorial but can you explain in more detail how to run the "start sonota.py with your standard wifi SSID/PW and PC IP (eg ./sonota.py --wifi-ssid asdfasdf --wifi-password fidjfidfjidjf 192.168.123.123" without omitting details. 

I saved the files in my mac desktop opened termina went to the directory I created try to run as above and got env python 3 no such file or directory.

cputoaster, 2017-08-25:
looks like python3 is not installed correctly. you can install it using homebrew: http://docs.python-guide.org/en/latest/starting/install3/osx/ . btw, its clear that you have to replace ssid, pw and ip with your current ones I hope.

juan11perez, 2017-08-25:
Thank you Andres, I installed from here (https://www.python.org/downloads/ <https://www.python.org/downloads/>) run the command and gets, 

  File "./sonota.py", line 24, in <module>
    import tornado.ioloop
ModuleNotFoundError: No module named 'tornado'

Regards/Saludos
Juan

cputoaster, 2017-08-25:
Indeed you will have to install missing modules with something like "pip3
install tornado"

juan11perez, 2017-08-25:
Ok Andres, so installed all modules, run sonata.py connected to module and get this Value error at bottom

~~ Provisioning completed
** The IP address of <serve_host> (192.168.1.21) is not  assigned to any interface on this machine.
** Please change WiFi network to $ESSID and make  sure %s is being assigned to your WiFi interface.
** This application can be kept running.
......~~ Starting web server
Traceback (most recent call last):
  File "./sonota.py", line 498, in <module>
    main()
  File "./sonota.py", line 486, in main
    "keyfile": "ssl/server.key",
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages/tornado/util.py", line 293, in __new__
    instance.initialize(*args, **init_kwargs)
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages/tornado/httpserver.py", line 167, in initialize
    read_chunk_size=chunk_size)
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages/tornado/tcpserver.py", line 128, in __init__
    self.ssl_options['certfile'])
ValueError: certfile "ssl/server.crt" does not exist
Juans-MacBook-Air:SonOTA Juan$ 

Regards/Saludos
Juan Perez

RobertSzabo, 2017-08-25:
I am halfway through the SonOTA install, but stack before the FinalStage with a 404 error. Any idea how to continue? Log inserted below.
Thanks!

`
<< HTTP POST /dispatch/device
>> /dispatch/device
>> {
    "IP": "192.168.100.43",
    "port": 8443,
    "reason": "ok",
    "error": 0
}
2017-08-25 11:20:19,588 (INFO) 200 POST /dispatch/device (192.168.100.116) 2.18ms
2017-08-25 11:20:19,780 (INFO) 101 GET /api/ws (192.168.100.116) 1.34ms
2017-08-25 11:20:19,780 (DEBUG) << WEBSOCKET OPEN
2017-08-25 11:20:19,791 (DEBUG) << WEBSOCKET INPUT
2017-08-25 11:20:19,791 (DEBUG) << {
    "userAgent": "device",
    "model": "PSF-B04-GL",
    "romVersion": "2.0.0",
    "ts": 744,
    "deviceid": "10001182c9",
    "apikey": "c1f4a468-b659-44a1-b867-917015a15395",
    "version": 2,
    "action": "register"
}
~~~ device sent action request,  acknowledging / answering...
2017-08-25 11:20:19,792 (INFO) We are dealing with a PSF-B04-GL model.
~~~~ register
>> {
    "deviceid": "10001182c9",
    "apikey": "adcf96ce-9e40-46d5-a0c0-fbc53f6888b9",
    "error": 0,
    "config": {
        "hbInterval": 145,
        "hb": 1
    }
}
2017-08-25 11:20:19,806 (DEBUG) << WEBSOCKET INPUT
2017-08-25 11:20:19,807 (DEBUG) << {
    "userAgent": "device",
    "apikey": "adcf96ce-9e40-46d5-a0c0-fbc53f6888b9",
    "deviceid": "10001182c9",
    "action": "date"
}
~~~ device sent action request,  acknowledging / answering...
~~~~ date
>> {
    "date": "2017-08-25T11:20:19.807Z",
    "deviceid": "10001182c9",
    "apikey": "adcf96ce-9e40-46d5-a0c0-fbc53f6888b9",
    "error": 0
}
2017-08-25 11:20:19,820 (DEBUG) << WEBSOCKET INPUT
2017-08-25 11:20:19,823 (DEBUG) << {
    "userAgent": "device",
    "apikey": "adcf96ce-9e40-46d5-a0c0-fbc53f6888b9",
    "deviceid": "10001182c9",
    "params": {
        "switches": [
            {
                "outlet": 0,
                "switch": "off"
            },
            {
                "outlet": 1,
                "switch": "off"
            },
            {
                "outlet": 2,
                "switch": "off"
            },
            {
                "outlet": 3,
                "switch": "off"
            }
        ],
        "staMac": "60:01:94:84:13:B8",
        "configure": [
            {
                "outlet": 0,
                "startup": "off"
            },
            {
                "outlet": 1,
                "startup": "off"
            },
            {
                "outlet": 2,
                "startup": "off"
            },
            {
                "outlet": 3,
                "startup": "off"
            }
        ],
        "fwVersion": "2.0.0",
        "rssi": -59
    },
    "action": "update"
}
~~~ device sent action request,  acknowledging / answering...
~~~~ update
>> {
    "deviceid": "10001182c9",
    "apikey": "adcf96ce-9e40-46d5-a0c0-fbc53f6888b9",
    "error": 0
}
>> {
    "sequence": "1503652819824",
    "userAgent": "app",
    "from": "hackepeter",
    "params": {
        "switch": "off"
    },
    "ts": 0,
    "deviceid": "10001182c9",
    "apikey": "adcf96ce-9e40-46d5-a0c0-fbc53f6888b9",
    "action": "update"
}
>> {
    "sequence": "1503652819825",
    "userAgent": "app",
    "from": "hackepeter",
    "params": {
        "switch": "on"
    },
    "ts": 0,
    "deviceid": "10001182c9",
    "apikey": "adcf96ce-9e40-46d5-a0c0-fbc53f6888b9",
    "action": "update"
}
>> {
    "sequence": "1503652819825",
    "userAgent": "app",
    "from": "hackepeter",
    "params": {
        "switch": "off"
    },
    "ts": 0,
    "deviceid": "10001182c9",
    "apikey": "adcf96ce-9e40-46d5-a0c0-fbc53f6888b9",
    "action": "update"
}
>> {
    "sequence": "1503652819825",
    "userAgent": "app",
    "from": "hackepeter",
    "params": {
        "switch": "on"
    },
    "ts": 0,
    "deviceid": "10001182c9",
    "apikey": "adcf96ce-9e40-46d5-a0c0-fbc53f6888b9",
    "action": "update"
}
>> {
    "sequence": "1503652819825",
    "userAgent": "app",
    "from": "hackepeter",
    "params": {
        "switch": "off"
    },
    "ts": 0,
    "deviceid": "10001182c9",
    "apikey": "adcf96ce-9e40-46d5-a0c0-fbc53f6888b9",
    "action": "update"
}
>> {
    "sequence": "1503652819830",
    "userAgent": "app",
    "params": {
        "binList": [
            {
                "name": "user1.bin",
                "digest": "97fe70d737390bbde7b8d208c79d590e29fc76962623c7bd198ee210210ea616",
                "downloadUrl": "http://192.168.100.43:8080/ota/image_user1-0x01000.bin"
            },
            {
                "name": "user2.bin",
                "digest": "e6d0cbb641542bd0a27e6c4628fc9331f1f5d3ac3afa42c9307569cf4d70fcdc",
                "downloadUrl": "http://192.168.100.43:8080/ota/image_user2-0x81000.bin"
            }
        ],
        "version": "23.42.5",
        "model": "PSF-B04-GL"
    },
    "ts": 0,
    "deviceid": "10001182c9",
    "apikey": "adcf96ce-9e40-46d5-a0c0-fbc53f6888b9",
    "action": "upgrade"
}
2017-08-25 11:20:19,870 (INFO) 200 GET /ota/image_user2-0x81000.bin?deviceid=10001182c9&ts=1270216262&sign=68c11599e5338c21e1fd621faed8d52e9798196c9dc58837a30d014427ff3d99 (192.168.100.116) 12.12ms
2017-08-25 11:20:19,870 (DEBUG) << WEBSOCKET INPUT
2017-08-25 11:20:19,871 (DEBUG) << {
    "sequence": "1503652819830",
    "userAgent": "device",
    "apikey": "adcf96ce-9e40-46d5-a0c0-fbc53f6888b9",
    "error": 404,
    "deviceid": "10001182c9"
}
`

cputoaster, 2017-08-25:
@juan11perez, did you unzip the zip file as mentioned into the same dir das
the sonota script lives? it looks like it cannot find the ssl certs.

cputoaster, 2017-08-25:
@Robert, you should first connect your device with the original ewLink and
upgrade it to the most recent original firmware.

RobertSzabo, 2017-08-25:
@cputoaster, i did that, first used the device with ewLink, then deleted (unregistered) and started the SonOTA upgrade process. It went well until the 404 error. If I restart the device it keeps trying with the same handshake and returns the 404. Any advise? Thanks!

juan11perez, 2017-08-25:
Andrés the link in the tutorial is broken, so i went to your GitHub and downloaded https://github.com/cputoaster/Espressif2Arduino

I then unzipped and put in a folder I called SonOta in my desktop. I also saved the SonOta script in the same folder.

Regards
Juan Perez

cputoaster, 2017-08-25:
I corrected the link in the tutorial, should work now. This is not equivalent to getting the fork.

cputoaster, 2017-08-25:
@RobertSzabo did you upgrade to a new version while using it with ewLink? it normally takes a few minutes for an upgrade message to appear. I just had exactly the same 404 problem when I used the original version, it went away after updating.

juan11perez, 2017-08-25:
Andres, thank you very much for your help. I know you updated the link, I dowloaded it, repeated the install and it’s working now. MQTT and everything.

@Robert, you need to ensure the device is in firmware 1.61 or higher. I had the same error you were showing. I reconnected it to Welink, updated the firmware and sorted.

Thank you

Regards/Saludos
Juan

RobertSzabo, 2017-08-25:
Indeed; just managed to upgrade via ewLink after, which the SonOTA completed successfully. Thanks for all the help and the great firmware! 
Cheers, Robert

arendst, 2017-08-25:
@cputoaster Andres, could you be so kind to update my wiki with information how to download, configure and use your tool?

I've made a chapter about upload tools and some entries for esptool, arduino and platformio. I did add a SonOTA page too. 

Pls update this page with information mentionend here for a noob to use your great tool.

ferbulous, 2017-08-26:
@cputoaster do you have any plans to include other firmware like espurna with your OTA method?

cputoaster, 2017-08-28:
@arendst I wrote it up on the wiki here: https://github.com/arendst/Sonoff-Tasmota/wiki/SonOTA---Espressif2Arduino---Tasmota-without-compiling and placed the link in the tutorial section. I am not sure if its a good idea to call it a "SonOTA" install, as I tried to make things simple just by compiling (unmodified, my changes got merged) Espressif2Arduino, using (unmodified) SonOTA with snakeoil certs and directly linking to the "official" Tasmota builds on the web. Its mostly packaging for noobs.
If you want to, I can copy it to the SonOTA page, but it might not be the "pure" way.

cputoaster, 2017-08-29:
@ferbulous err, did not really try yet. The point of my package was to just fetch some hardcoded URL, which I set to Tasmota. But I guess if the fw you need is somewhere available on the web, it should be simple to compile an E2A binary that fetches that.

matphillips, 2017-08-30:
My T1 arrived today and I managed to install the Tasmots firmware OTA with reasonable easy thanks to the details provided here. The people involved in this project are fantastic!
I have the homebridge-mqtt-switch-tasmota plugin running with Homebridge to control the other Sonoff devices and can't quite get it to work with this three-gang unit. I can switch ON each of the four relays but cannot switch them off. The web interface indicates them as ON but the Home app in iOS thinks they are off and sends an ON command each time.

Different plugin needed or wrong config used? Anyone tried this?

ferbulous, 2017-08-31:
> But I guess if the fw you need is somewhere available on the web, it should be simple to **_compile an E2A binary_** that fetches that

@cputoaster I'm not really sure how to do that 

Another question, how should I key in my wifi ssid if there's space between the ssid name?
For example if wifi ssid is 'house wifi', when add it to this command
`./sonota.py --wifi-ssid house wifi --wifi-password mypassword 192.168.2.190`

The script doesn't run properly and only shows up for a split second

cputoaster, 2017-09-01:
I can compile something for you if you can give me a link of the fw on the web. As for using sonota on windows, I dont really know. I would guess that that you can put the string between quotes (eg --wifi-ssid "house wifi"  --wifi-password .....) but thats just a guess.

cputoaster, 2017-09-01:
@ageorgios you might want to close this issue, as its starting to deviate from the original subject.

ferbulous, 2017-09-01:
@cputoaster for espurna you find it here, https://bitbucket.org/xoseperez/espurna/downloads/
It's espurna.ino if that's what you require

For  sonoff ota method, I'm still stuck at this:

`    "serverName": "192.168.8.100",
    "port": 4223
}
<< {
    "error": 0
}
~~ Provisioning completed
** The IP address of <serve_host> (192.168.8.100) is not  assigned to any interface on this machine.
** Please change WiFi network to $ESSID and make  sure %s is being assigned to your WiFi interface.
** This application can be kept running.
..........~~ Starting web server
~~ Waiting for device to connect`

I'm trying this method on [sonoff t1 clone](https://item.taobao.com/item.htm?spm=a1z09.2.0.0.2ed6e797dsJDBm&id=554217191590&_u=22avq0do7c8a) (psf-b85), also this one has firmware version 2.0.1

ferbulous, 2017-09-02:
@cputoaster some updates, I successfully flashed using sonota method on my sonoff t1 clone
Only issue I'm having is that it's not saving anything that I modified on the settings
If I add my home wifi/change module, once the device restarts it's still on the default settings

davidelang, 2017-09-02:
it sounds like you flashed the be minimal version, that version does't let you 
save anything, it only exists to then let you flash the full version.

cputoaster, 2017-09-04:
@ferbulous yes, you should probably update by using the urls on this page on the web interface: https://github.com/arendst/Sonoff-Tasmota/wiki/Upgrade, most probably http://sonoff.maddox.co.uk/tasmota/sonoff.bin

jeylites, 2017-09-26:
Why is the following keep appearing, Is the syntax wrong?

JK-Air:SonOTA JK$ ./sonota.py --wifi-ssid asus --wifi-password 055555 192.168.1.79
usage: sonota.py [-h] [--serving-host SERVING_HOST] [--no-prov]
                 [--wifi-ssid WIFI_SSID] [--wifi-password WIFI_PASSWORD]
                 [--no-check-ip] [--legacy]
sonota.py: error: unrecognized arguments: 192.168.1.79
JK-Air:SonOTA JK$

lgnd33, 2017-09-26:
Arent you missing --serving-host in front of IP address? :)

//_

jeylites, 2017-09-26:
got it to load ./sonota.py    but it's stock at discovering the Sonoff T1

JK:~ JK$ ./sonota.py
-bash: ./sonota.py: No such file or directory
JK:~ JK$ cd ~/src/SonOTA
JK:SonOTA JK$ ./sonota.py
Select IP address of the WiFi interface:
    0: 192.168.1.79
Select IP address [0]: 0
WiFi SSID: asus
WiFi Password: 55555555

Using the following configuration:
	Server IP Address: 192.168.1.79
	WiFi SSID: asus
	WiFi Password: 55555555
** Now connect via WiFi to your Sonoff device.
** Please change into the ITEAD WiFi network (ITEAD-100001XXXX). The default password is 12345678.
To reset the Sonoff to defaults, press the button for 7 seconds and the light will start flashing rapidly.
** This application should be kept running and will wait until connected to the Sonoff...
.........................................................................................................................................................................................................

jeylites, 2017-09-27:
My device is currently stuck at Final Stage, I see it show up on AP list but I can’t seem to do anything with it. Any ideas, also how to do a hardware flash with OTA. I did some research online and found a few steps but I can’t seem to get it flash mode

jscurtu, 2017-10-12:
@jeylites same with me , i read the 4ch pro is supported and upgraded the firmware before but now I am stuck at FinalStage. 
I see the AP, any luck on fixing this?

jeylites, 2017-10-12:
@jscurtu I feel your frustration :) The only way around will be to take it apart for a manual upgrade.

jscurtu, 2017-10-12:
@jeylites aight ... thx.. :-)

LurchiStromberg, 2017-11-08:
Hello!
Today i got my 3 Gang Touch from Banggood...

But it looks like they change the Board Layout again.

Can somebody help me about the Connectors to flash Tasmota?

1st Question would be, what is the right Port? J1 or J3 ?

And what connector is what? 

[image]
[image]

davidelang, 2017-11-08:
it sounds like you are the first person to try this module, so you will have to 
trace leads or get a datasheet to figure out what the correct port would be.

juan11perez, 2017-11-09:
Suggest using SonOTA
https://github.com/arendst/Sonoff-Tasmota/wiki/SonOTA---Espressif2Arduino---Tasmota-without-compiling

Works an no need to solder anything. 

Regards
Juan

LurchiStromberg, 2017-11-09:
ty Juan! I found that already too and its really good working.
Was some tricky, because to get T1 into pairing AND ALSO Wifi Mode you need to hold the left Button for about 15-20 sec., till the Wifi LED flashes fast contiuesly...
Then you can also find the T1 Sonoff Wifi at your Computer...

After flashing its also the left Button you need to push as fast as you can 5 Times to get it into WPS Mode....

codedmind, 2017-11-09:
After flash can you use the two buttons? I only can get one to work even after change module type to T1 2Ch. Is needed do any more change? Thanks
Mine is this model https://pt.aliexpress.com/item/Work-with-Amazon-Alexa-Smart-Wi-Fi-Switch-Glass-Panel-1gang-2gang-3gang-EU-Touch-Light/32818148422.html

ferbulous, 2017-11-09:
@codedmind
Try flashing to t1 3ch firmware, see if the 2nd button responds after that

codedmind, 2017-11-09:
@ferbulous flashing or change type?

ferbulous, 2017-11-09:
@codedmind 
Chane the type should be sufficient

codedmind, 2017-11-09:
@ferbulous I will try that, thanks

LurchiStromberg, 2017-11-09:
I have the 3 Gang Switch.... Works without problems.....

After flashing you should update the firmware with Webinterface...

SONota flashes an older Version of Arends Tasmota....... Just go to the Wiki and download the newest *.bin

[image]
[image]

ageorgios, 2017-11-10:
@codedmind 
I saw from the link to aliexpress that this is not from iTead. Is this 2gang EU socket also compatible with Sonoff? Is the build quality good?

codedmind, 2017-11-10:
@ferbulous change to t1 3ch works ok :)
@LurchiStromberg after update with the latest firmware from tasmota it works well with t1 2ch
@ageorgios yes, is a "clone" i can upload the firmware with sonota and it works. The build material is good, but the relay isn't ssd, so it make noise when change state. Other think i notice is that exists a certain delay toggle the switch(in both cases in the switch and via tasmota) but because sonoff don't sell an EU version this is a acceptable alternative

ferbulous, 2017-11-10:
@ageorgios
As long as it's using itead chipset/ewelink app
You should be able to flash it
I bought this one which is pretty much a cheaper version of sonoff t1 with more color option

https://m.intl.taobao.com/detail/detail.html?id=554114857768&spm=a2141.7631563.1.detail&ttid=212200%40htao_android_4.1.4

codedmind, 2017-11-10:
@ferbulous the relay in that version are ssd ?

davidelang, 2017-11-10:
if it uses an ESP8266 you should be able to use Tasmota on it, it doesn't 
matter what firmware is currently on it.

[17 more replies]

closed 2018-03-14 by ageorgios