arendst/Tasmota #584: Custom mqtt payload

solucionesmipc, 2017-07-05:
hi:

which file should i modify to custom mqttt payload?

thanks 

luis

davidelang, 2017-07-06:
that depends what you are wanting to do. A lot of the files create parts of the 
mqtt payload.

explain a bit what you are trying to do.

solucionesmipc, 2017-07-06:
Hi

I need extra data in published messages.

Ex: {"POWER":"ON", "TOPIC":"54555531"}

I need topic published in payload for stat and tele messages

Thanks a lot

davidelang, 2017-07-06:
that sort of modification would be in sonoff.ino

solucionesmipc, 2017-07-06:
Thanks for your answer

i got extra data in mqtt messages, but i still dont get the "topic" variable.

ex:

snprintf_P(svalue, ssvalue, PSTR("%s, \"AP\":%d, \"SSID\":\"%s\", \"RSSI\":%d, \"APMac\":\"%s\", \"Topic\":\"%s\"}"),
    svalue, sysCfg.sta_active +1, sysCfg.sta_ssid[sysCfg.sta_active], WIFI_getRSSIasQuality(WiFi.RSSI()), WiFi.BSSIDstr().c_str(), "what variable should i use here");

thanks again

davidelang, 2017-07-06:
look at the code that sends the stat and tele messages, see the variables that 
are used there to form the message header.
kk

solucionesmipc, 2017-07-06:
i got it.
sysCfg.mqtt_topic
thanks a lot

solved.

closed 2017-07-06 by solucionesmipc