arendst/Tasmota #585: Button  activate Relay

Smoker2604, 2017-07-06:
Hi, I have the firmware in the latest version on my Sonoff dual flasht.
Through the web interface I can switch the device without problems.
But I have a problem, if I over the two buttons on the board try to switch the relays. Then usually only the first relay is switched, no matter whether Button 0 or Button 1. It also occurs times, the ESonoff the first Relay then switches the second and then again the first with only one-time the button 0. I have the serial Log disabled, as it is in the manual.

arendst, 2017-07-06:
Only the single button on the Dual is supported. The header for Button 0 and 1 is NOT supported (as has been discussed a zillion times in other issues....).

Smoker2604, 2017-07-06:
92/5000
OK Thanks for your quick reply. Is it planned to make it usable in the future?

sillyfrog, 2017-08-20:
It won't be possible in the future because of the issues outlined here: http://tinkerman.cat/smart-wall-switches-and-push-buttons/

Game01Master, 2017-11-03:
Sonoff Dual with original firmware work with buttons. Push switch must be open type. 

https://youtu.be/VX_eV_HKb3s

Joeyhza, 2017-11-08:
I've had OTA issues flashing my Duals, in retrospect I'm glad i did for now

I have 2 of my duals installed behind wall light switches each controlling 2 lights on Ewelink as well as each physical wall switch connected to But 0 and But 1 on the Sonoff. Using the original firmware im able to turn on and off each light from ewelink and the all switch. 

Only problem with the wall switch is i have to flip it up and then down to turn the light on and off. So 2 flips to turn on and then 2 flips to turn off.

I have a few other sonoff basics running lights for other parts of the house where a dual switch wasn't needed. These are all Tasmota flashed with the physical wall switch for each running to GPIO14. Works flawlessly.

I'm still not entirely sure why the custom firmware can't be made to work the same as the orginal firmware for the dual when it comes to attaching physical switches to But 1 and But 2.
So.....for now I'm running some of my house on ewelink and someof my house on Domoticz using MQQT. If custom firmware can make the dual behave like the orginal when it comes to wiring up a switch to But 1 and But 2 - I'll be the first to flash Tasmota :)

Probably better - not having all my eggs in one basket and all :)

Thanks for all the great dev work people!!! Love Tasmota

davidelang, 2017-11-08:
change the switchmode for the switches.

Joeyhza, 2017-11-08:
Changing switchmode isn't an available option on the original ITEAD firmware

And flashing Tasmota on the dual would mean I wouldn't have manual switch functionality for both switches which the original ITEAD firmware sort of accommodates.

davidelang, 2017-11-08:
Well, if you aren't running Tasmota, nobody here is going to be able to help 
you, you will have to ask ITEAD to add features to their system.

Joeyhza, 2017-11-08:
Wasnt asking for help - merely sharing my experience in relation to the original post. Did you perhaps read my entire post detailing my current setup?

davidelang, 2017-11-08:
I didn't go back and read the entire issue, I'm usually just looking at the most 
recent post.

closed 2018-05-07 by stale[bot]