arendst/Tasmota #586: Sonoff Pow seems to have died

mihalski, 2017-07-07:
Or so it seems.. Was working fine one moment and then everything that was running through it lost power. There's no heavy load here.

Tried changing the fuse, and tried powering it directly via the header pins. Nada. Light never comes on or flashes.

Is there anything else I could check?

ionciubotaru, 2017-07-08:
I had the same problem withe one of my POW's 
The ESP works if the FTDI is connected, but it doesn't work when connected to main. The fuse is OK
I am sure that the failure is inside the power supply. 
Recently I had the same problem with 2 Sonoffs Touch - the first has a power supply broken, the second has the esp8266 broken, so I moved the touchPCB  from the 2-nd to the first and recovered one of them.

On any mass production there are 1-2% failures.

mihalski, 2017-07-08:
I couldn't get it working with FTDI either. Not sure what failed on it.. I guess the saving grace is that they are so cheap. Maybe I can recover some components from it.

mkh595, 2017-08-07:
Have same experience with POW. This weekend two died. 
One was working 5 minutes and then I saw red led slowly switch off. I am able to flash it again. Powered from FTDI relay click but red led do nothing. When powered from main on output is main power independend of status ON/OFF. Current, power and  voltage show nonsensical values.
Second was working one hour and after flashing to firmware 5.5.1 died completly.

I am afraid that POW is not well designed.

closed 2018-05-07 by stale[bot]