arendst/Tasmota #588: Sonoff SC Baudrate

swilalaa, 2017-07-07:
Hello,

I'm working with Arduino to config your awesome software.

I managed to upload the software to the SC but I'm stucking with the Baudrate.

You're saying in the tutorial "After flashing you need to set the baudrate to 19200. "

I'm a total noob, can someone help me how to do that? 

Thanks in advance!

Lennyz1988, 2017-07-09:
It does not say that anywhere. You set it to 19200 BEFORE flashing.

swilalaa, 2017-07-23:
Thanks @Lennyz1988 but Arduino just gives me 9600 or 57600 as the lowest upload speed option. I know with platform.io that it had this option, but with the latest phyton issues I can't get platform.io on atom running, so I have to use Arduino

Lennyz1988, 2017-07-24:
Yes, I was not paying attention. 19200 does not exists.

Just set upload speed at 115200/9600. It doesn't really matter what you choose. It's just the speed for uploading the sketch, nothing else. The lower the speed the longer it will take to upload.

swilalaa, 2017-07-25:
Got that, uploading wasnt the problem but in the http console it keeps saying set baudrate to 19200. So how do I do that afterwards with Arduino?

jakass202, 2018-02-20:
Apologies i have the same problem as swilalaa. On the instructions;
Note! After flashing you need to set the baudrate to 19200. Don't forget to reconnect the TX jumper after flashing ;)
I have Atom and have flashed the Sonoff SC successfull, what is the next step pls?

mdal21, 2018-02-25:
Same here. I have uploaded Sonoff SC firmware but dont know how to set the baud rate to 19200 after that. No communication with ATMega.
Any tips?

mdal21, 2018-02-28:
OK found it. It can be done via browser under Configure/Configure Module/ and selecting the correct model type (21 Sonoff SC)