arendst/Tasmota #592: Flash Sonoff Tasmota

sujitrp, 2017-07-08:
Arduino: 1.8.3 (Windows 7), Board: "Generic ESP8266 Module, 80 MHz, 40MHz, DIO, 115200, 1M (64K SPIFFS), ck, Disabled, None"

exec: "C:\Users\spp07561\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-gcc": file does not exist
Error compiling for board Generic ESP8266 Module.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

Preferences.txt
board=generic
boardsmanager.additional.urls=http://arduino.esp8266.com/versions/2.3.0/package_esp8266com_index.json
build.verbose=false
build.warn_data_percentage=75
compiler.cache_core=true
compiler.warning_level=none
console=true
console.auto_clear=true
console.error.file=stderr.txt
console.length=500
console.lines=4
console.output.file=stdout.txt
custom_CpuFrequency=generic_80
custom_Debug=generic_Disabled
custom_DebugLevel=generic_None____
custom_FlashFreq=generic_40
custom_FlashMode=generic_dio
custom_FlashSize=generic_512K64
custom_ResetMethod=generic_ck
custom_UploadSpeed=generic_115200
editor.antialias=true
editor.auto_close_braces=true
editor.caret.blink=true
editor.code_folding=false
editor.divider.size=2
editor.external=false
editor.font=Monospaced,plain,12
editor.indent=true
editor.invalid=false
editor.keys.alternative_cut_copy_paste=true
editor.keys.home_and_end_beginning_end_of_doc=false
editor.keys.shift_backspace_is_delete=true
editor.languages.current=
editor.linenumbers=false
editor.save_on_verify=true
editor.tabs.expand=true
editor.tabs.size=2
editor.update_extension=true
editor.window.height.default=600
editor.window.height.min=290
editor.window.width.default=500
editor.window.width.min=400
export.applet.separate_jar_files=false
export.application.fullscreen=false
export.application.platform=true
export.application.stop=true
export.delete_target_folder=true
gui.scale=auto
last.folder=D:\OS\Raspberrypi3\Sonoff\sonoff-final\sonoff\sonoff.ino
last.ide.1.8.2.daterun=1499396806
last.ide.1.8.2.hardwarepath=C:\Program Files (x86)\Arduino\hardware
last.ide.1.8.3.daterun=1499420435
last.ide.1.8.3.hardwarepath=D:\OS\arduino-1.8.3-windows\arduino-1.8.3\hardware
last.screen.height=768
last.screen.width=1366
last.serial.location=627,100,669,378
last.sketch.count=2
last.sketch.default.location=433,84,500,600,-1,0
last.sketch.default.path=C:\Users\spp07561\AppData\Local\Temp\untitled2061132596.tmp\sketch_jul07a\sketch_jul07a.ino
last.sketch0.location=437,73,500,600,-1,0
last.sketch0.path=D:\OS\Raspberrypi3\Sonoff\sonoff-final\sonoff\sonoff.ino
last.sketch1.location=-8,-8,1382,744,536,6
last.sketch1.path=D:\OS\Raspberrypi3\Sonoff\sonoff1vwitherro\sonoff1vwitherro.ino
last.sketch2.location=-8,-8,1382,744,536,6
last.sketch2.path=C:\Users\spp07561\Documents\Arduino\Sujit\FadeRGB\FadeRGB.ino
last.sketch3.location=-8,-8,1382,744,536,6
last.sketch3.path=C:\Program Files (x86)\Arduino\examples\03.Analog\AnalogInOutSerial\AnalogInOutSerial.ino
platform.auto_file_type_associations=true
preproc.color_datatype=true
preproc.enhanced_casting=true
preproc.imports.list=java.applet.,java.awt.Dimension,java.awt.Frame,java.awt.event.MouseEvent,java.awt.event.KeyEvent,java.awt.event.FocusEvent,java.awt.Image,java.io.,java.net.,java.text.,java.util.,java.util.zip.,java.util.regex.*
preproc.output_parse_tree=false
preproc.save_build_files=false
preproc.substitute_floats=true
preproc.substitute_unicode=true
preproc.web_colors=true
programmer=arduino:avrisp
proxy.manual.hostname=
proxy.manual.password=
proxy.manual.port=
proxy.manual.type=HTTP
proxy.manual.username=
proxy.pac.url=
proxy.type=auto
recent.sketches=D:\OS\Raspberrypi3\Sonoff\sonoff-final\sonoff\sonoff.ino,D:\OS\Raspberrypi3\Sonoff\sonoff1vwitherro\sonoff1vwitherro.ino,D:\OS\Raspberrypi3\Sonoff\Sonoff-Tasmota\Sonoff-Tasmota-5.2.0\sonoff1vwithirerror\sonoff1vwithirerror.ino,C:\Users\spp07561\Desktop\sonoff\sonoff.ino,C:\Program Files (x86)\Arduino\examples\03.Analog\AnalogInOutSerial\AnalogInOutSerial.ino,D:\OS\Raspberrypi3\Sonoff\sonoff\sonoff.ino,D:\OS\Raspberrypi3\Sonoff\Sonoff-Tasmota-5.2.0\Sonoff-Tasmota-5.2.0\sonoff\sonoff.ino,D:\OS\Raspberrypi3\Sonoff\Sonoff-Tasmota\Sonoff-Tasmota-5.2.0\Sonoff-Tasmota-5.2.0\sonoff\sonoff.ino,C:\Users\spp07561\Desktop\Blink\Blink.ino,C:\Program Files (x86)\Arduino\libraries\SD\examples\CardInfo\CardInfo.ino,C:\Program Files (x86)\Arduino\examples\04.Communication\Dimmer\Dimmer.ino,C:\Users\spp07561\Documents\Arduino\Sujit\FadeRGB\FadeRGB.ino,C:\Users\spp07561\Downloads\FV9QZY5HZLYRIU8\FV9QZY5HZLYRIU8.ino
run.display=1
run.options=
run.options.memory=false
run.options.memory.initial=64
run.options.memory.maximum=256
run.present.bgcolor=#666666
run.present.exclusive=false
run.present.stop.color=#cccccc
serial.databits=8
serial.debug_rate=19200
serial.line_ending=1
serial.parity=N
serial.port=COM12
serial.port.file=COM12
serial.port.iserial=null
serial.stopbits=1
sketchbook.path=C:\Users\spp07561\Documents\Arduino
software=ARDUINO
target_package=esp8266
target_platform=esp8266
update.check=true
update.id=-966520695278570248
upload.using=bootloader
upload.verbose=false
upload.verify=true

arendst, 2017-07-08:
Enable option "Show verbose output during compilation" via File > Preferences and compile again.

Study the result and fix your paths.

sujitrp, 2017-07-09:
Hi 
Pls find the

Arduino: 1.8.3 (Windows 7), Board: "Generic ESP8266 Module, 80 MHz, 40MHz, DIO, 115200, 1M (64K SPIFFS), ck, Disabled, None"

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\spp07561\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\spp07561\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\spp07561\Documents\Arduino\libraries -fqbn=esp8266:esp8266:generic:CpuFrequency=80,FlashFreq=40,FlashMode=dio,UploadSpeed=115200,FlashSize=1M64,ResetMethod=ck,Debug=Disabled,DebugLevel=None____ -ide-version=10803 -build-path C:\Users\spp07561\AppData\Local\Temp\arduino_build_66113 -warnings=none -build-cache C:\Users\spp07561\AppData\Local\Temp\arduino_cache_90297 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.esptool.path=C:\Users\spp07561\AppData\Local\Arduino15\packages\esp8266\tools\esptool\0.4.9 -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=C:\Users\spp07561\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2 -prefs=runtime.tools.mkspiffs.path=C:\Users\spp07561\AppData\Local\Arduino15\packages\esp8266\tools\mkspiffs\0.1.2 -verbose D:\OS\01782017\Sonoff-Tasmota-master\Sonoff-Tasmota-master\sonoff\sonoff.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\spp07561\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\spp07561\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\spp07561\Documents\Arduino\libraries -fqbn=esp8266:esp8266:generic:CpuFrequency=80,FlashFreq=40,FlashMode=dio,UploadSpeed=115200,FlashSize=1M64,ResetMethod=ck,Debug=Disabled,DebugLevel=None____ -ide-version=10803 -build-path C:\Users\spp07561\AppData\Local\Temp\arduino_build_66113 -warnings=none -build-cache C:\Users\spp07561\AppData\Local\Temp\arduino_cache_90297 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.esptool.path=C:\Users\spp07561\AppData\Local\Arduino15\packages\esp8266\tools\esptool\0.4.9 -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=C:\Users\spp07561\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2 -prefs=runtime.tools.mkspiffs.path=C:\Users\spp07561\AppData\Local\Arduino15\packages\esp8266\tools\mkspiffs\0.1.2 -verbose D:\OS\01782017\Sonoff-Tasmota-master\Sonoff-Tasmota-master\sonoff\sonoff.ino
Using board 'generic' from platform in folder: C:\Users\spp07561\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0
Using core 'esp8266' from platform in folder: C:\Users\spp07561\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0
Detecting libraries used...
"C:\Users\spp07561\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Users\spp07561\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0/tools/sdk/include" "-IC:\Users\spp07561\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0/tools/sdk/lwip/include" "-IC:\Users\spp07561\AppData\Local\Temp\arduino_build_66113/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=80000000L -DLWIP_OPEN_SRC   -DARDUINO=10803 -DARDUINO_ESP8266_ESP01 -DARDUINO_ARCH_ESP8266 -DARDUINO_BOARD="ESP8266_ESP01"  -DESP8266 "-IC:\Users\spp07561\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266" "-IC:\Users\spp07561\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\variants\generic" "C:\Users\spp07561\AppData\Local\Temp\arduino_build_66113\sketch\sonoff.ino.cpp" -o "nul"
"C:\Users\spp07561\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Users\spp07561\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0/tools/sdk/include" "-IC:\Users\spp07561\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0/tools/sdk/lwip/include" "-IC:\Users\spp07561\AppData\Local\Temp\arduino_build_66113/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=80000000L -DLWIP_OPEN_SRC   -DARDUINO=10803 -DARDUINO_ESP8266_ESP01 -DARDUINO_ARCH_ESP8266 -DARDUINO_BOARD="ESP8266_ESP01"  -DESP8266 "-IC:\Users\spp07561\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266" "-IC:\Users\spp07561\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\variants\generic" "-IC:\Users\spp07561\Documents\Arduino\libraries\PubSubClient\src" "C:\Users\spp07561\AppData\Local\Temp\arduino_build_66113\sketch\sonoff.ino.cpp" -o "nul"
"C:\Users\spp07561\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Users\spp07561\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0/tools/sdk/include" "-IC:\Users\ [... cut ...]
Using library IRremoteESP8266 at version 2.1.0 in folder: C:\Users\spp07561\Documents\Arduino\libraries\IRremoteESP8266 
Using library NeoPixelBus at version 2.2.8 in folder: C:\Users\spp07561\Documents\Arduino\libraries\NeoPixelBus 
Using library SPI at version 1.0 in folder: C:\Users\spp07561\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\SPI 
exec: "C:\\Users\\spp07561\\AppData\\Local\\Arduino15\\packages\\esp8266\\tools\\xtensa-lx106-elf-gcc\\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-gcc": file does not exist
Error compiling for board Generic ESP8266 Module.

davidelang, 2017-07-09:
On Sat, 8 Jul 2017, sujitrp wrote:

> exec: "C:\\Users\\spp07561\\AppData\\Local\\Arduino15\\packages\\esp8266\\tools\\xtensa-lx106-elf-gcc\\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-gcc": file does not exist
> Error compiling for board Generic ESP8266 Module.

This error says you don't have the esp8266 extensions for the arduino IDE 
installed properly.

sujitrp, 2017-07-09:
thanks David . post re-installation https://github.com/esp8266/Arduino issue solved.

closed 2017-07-09 by sujitrp