arendst/Tasmota #593: Update SSId<x> and Password<x>

ionciubotaru, 2017-07-08:
```

Password<x> | <pwd> | Set APx Wifi password (64 chars max) and restart
SSId<x>     | <ssid> | Set APx Wifi SSId (32 chars max) and restart

```

Updating SSID or Password via MQTT restart Sonoff. 
After SSID update sonoff will boot with a wrong ssid/password configuration

Please fire restart only after SaveData command

arendst, 2017-07-09:
I will consider JSON input in the future allowing more commands at once.

cobra85ec, 2017-07-10:
good call.... i would love this also

0n3-70uch, 2017-07-11:
Yeah good idea :-)

Sunrise17, 2017-07-12:
Really good idea, during AP mode (192.168.4.1), i want Sonoff device to restart one time only after adjusting all following configuration below, it will be perfect.

http://192.168.4.1/cm?cmnd=friendlyname%20Ceiling%20Light
http://192.168.4.1/cm?cmnd=grouptopic%20Kitchen
http://192.168.4.1/cm?cmnd=ssid2%20AsusRouter
http://192.168.4.1/cm?cmnd=password2%20x5x5x5x
http://192.168.4.1/cm?cmnd=mqtthost%20192.168.1.4
Save data
Restart

arendst, 2017-07-14:
Did some research and ended up not using JSON because:
- it only allows for unique keys which makes the use of the same command on one line unusable:
``Backlog {"status":2, "status":4}`` will only process command "status 4"
- it needs additional typing ;-)
- it takes +1232 bytes of precious code space

So I decided for the old fashionend semicolon separated command line like: ``Backlog status 2; status 4``

Will be implemented in next release.

lmamakos, 2017-07-16:
You don't have to just use a single dictionary item; the response could be a list of independent responses:

```
[ { "cmd":"friendlyname, "status":"OK},
  { "cmd":"grouptopic, "status":"OK},
  { "cmd":"foo", status:"WTF?"} ]
```
of course that doesn't address your code space concern..

arendst, 2017-07-16:
@ionciubotaru since the latest release 5.3.0 try command 

backlog ssid1 yourssid; password1 yourpassword

No need to enter restart 1 as it is postponed until the last command from the backlog is executed.

0n3-70uch, 2017-07-16:
Hi,

I tested the new function... works very nice. Thank you.
One of the best new features in the last months :-).

Greetings
Fabi

PS: I close the issue.

closed 2017-07-16 by ionciubotaru

Sunrise17, 2017-07-16:
Dear @arendst,
Thanks for the useful update, it works so well. I have one issue after update. Before 5.3.0, i was using v.5.1.3 and my group topic could accept space between words. However, with 5.3.0, its adding "_" to spaces between words. I have checked same thing in FriendlyName, there is no such problem in that.
[image]

Greetings
Koray