arendst/Tasmota #594: HC-SR501 PIR Motion Detection - Switch (9)

mydakota, 2017-07-08:
Hello Guys hopefully I can get  some help I have flashed my wemos and have set it to 
WeMos D1 mini Module in the online interface now from reading the switch/button sensor type guide it says 
for a motion detectors / PIR use switch 9. But there is no switch 9 it only goes up to switch4 am I missing something .  or reading below wrong.
HC-SR501 PIR Motion Detection - Switch (9)

mydakota, 2017-07-08:
lol OK i think Ive read it wrong switch 9 is actual ( 09switch 1 )

mydakota, 2017-07-08:
Or would I do something like this to detect motion or change in the switch state.

cmnd/sonoff1/POWER

If so how if I am using multiple sensors
Do I select between sensors 1-8 
Thanks

davidelang, 2017-07-08:
cmnd/ commands tell the device to switch a relay on or off. defining a switch 
generates stats/ and tele/ messages reporting the state of the switch

Tasmota allows for up to 4 switches, which will generate stat/sonoff/POWER1-4 
messages

you cannot have more than 4 switch states in the current version of tasmota.

mydakota, 2017-07-08:
Thank you for your response it helps a lot with my understanding as I'm New is to Tasmota 
I'm a bit gutted about only being able to us 4 switch would be a useful option I think anyways
Hopefully in the future this will be added.
Would it be easy for me to edit the config file to add more than 4 or would this be a major coding challenge. Also out of interest are all devices limited by 4 for example maximum number of 4 relays 
Like the switchs. Would it be possible for me to assign 4 switches and use 4 buttons or is it a a limitation of the MQTT reporting

davidelang, 2017-07-09:
right now, it's limited to 4 of each thing. look at sonoff_template.ino and see 
how things are currently defined. If you increase the number of items, you may 
have to change some loop definitions in sonoff.ino

by default, buttons and switches do the same thing, but you can set switchtopic 
to have the switches report differently than the buttons. note that buttons are 
always momentary toggles, there is no buttonmode that's the equivalent of 
switchmode.

mydakota, 2017-07-09:
Thanks for the supper fast reply ,
Ok I might have a play see what I can do if all else fails I'll just write a basic arduino code 
For my alarm interface. Hopefully The Tasmota Firmware will add the ability to used more than 4 my preference would be 8 specifically for interfacing with a existing alarm systems PIRs rather than installing more PIRs on top of the ones I already have. If anyone else goes down this path keep in mind mist alarms use EOL resistors and takes a bit of nutting out. For me there was two ways around the EOL issue 1. Is put a PGM expander card in and program the PGMs to follow zone open and close .
But the way I'm doing it mainly to save money is adding a relay card between PIRs and inputs of alarm system either disable EOL resistors or putting them in the relays themselves kind of pointless doing this unless you don't have access to reprogramming alarm.  So PIRs activate the individual relays  N/C contacts go into alarm panel zones and functions the same as they did before adding the relay card in between I then use the N/O contact from the relay outputs to trigger inputs on my wemos .
Works perfectly just need some code to drive the MQTT status.

closed 2017-07-09 by mydakota