arendst/Tasmota #595: Yealink/Tuya

pedroOliveiraptc, 2017-07-09:
My friends do you know if it is possible control this touch panel with tasmota?
I have flash it with success but i don't have communication with the console, but manual it works.

Many thanks,

ionciubotaru, 2017-07-10:
It is possible, but you have to dig a little bit inside code.
Sonoff Dual use TX/RX to comunicate with the IC controlling the relays.
Tuya do the same, but with different protocol.

In sonoff.ino locate this code:
```

void setRelay(uint8_t rpower)
..........................................
  if ((SONOFF_DUAL == sysCfg.module) || (CH4 == sysCfg.module)) {
    Serial.write(0xA0);
    Serial.write(0x04);
    Serial.write(rpower);
    Serial.write(0xA1);
    Serial.write('\n');
    Serial.flush();
  }

```

and try to replace with:

```
          Serial.write(0x55); // header 55AA
          Serial.write(0xAA);
          Serial.write(0x00); // version 00
          Serial.write(0x06); // command 06 - send order
          Serial.write(0x00);
          Serial.write(0x05); // following data length 0x05
          Serial.write( (btnnum + 1) ); // relay number 1,2,3
          Serial.write(0x01); // ?
          Serial.write(0x00); // ?
          Serial.write(0x01); // ?
          Serial.write( state ); // status
          Serial.write((13 + btnnum + state)); // checksum:sum of all bytes in packet mod 256
          Serial.flush();
```

enesbcs, 2017-12-02:
Sort of... but rpower in Tasmota contains every relay, and the Tuya API sends separated data packets from every relay, and has to be send separately... so if first button pressed the btnnum is 0 (added 1, because Tuya starts numbering from 1). State can be 0/1 (off/on). 
So the rpower has to be preprocessed before this code can be used, someone needs to be code it. :)

Original plugin source: https://github.com/enesbcs/ESPEasyPluginPlayground/blob/master/_P165_SerSwitch.ino
Used Tuya SDK: 
[serial.zh-CN.en.pdf](https://github.com/arendst/Sonoff-Tasmota/files/1524168/serial.zh-CN.en.pdf)

closed 2018-05-07 by stale[bot]