arendst/Tasmota #596: Alexa not finding device

simo26246, 2017-07-10:
Hi all,

I'm**totally** new to this whole home automation thing, so please bear with me. I've scanned the forum and done all the suggested things to try and get it to work, but it still isn't...

I've purchased some of the sonoff basic models, and flashed them with the Tasmota firmware without issue, but now that I've done that, Alexa doesn't seem to want to discover them (and stupid me didn't check whether or not ewelink actually found them before flashing tasmota (I'm using an alexapi and don't have a genuine hue bridge - I don't know if this makes a difference?)

I wasn't able to work out how to enable the emulation when I flashed the firmware so I did this using the web interface after the fact

This is my HTTP config in the user_config.h file as flashed (which doesn't actually activate the emulation - anyone know what I should do differently to enable emulation?
`// -- HTTP ----------------------------------------
#define USE_WEBSERVER                            // Enable web server and wifi manager (+62k code, +8k mem) - Disable by //
  #define WEB_SERVER           2                 // [WebServer] Web server (0 = Off, 1 = Start as User, 2 = Start as Admin)
  #define WEB_PORT             80                // Web server Port for User and Admin mode
  #define WEB_USERNAME         "USERNAME"           // Web server Admin mode user name
  #define WEB_PASSWORD         "PASSWORD"                // [WebPassword] Web server Admin mode Password for WEB_USERNAME (empty string = Disable)
  #define FRIENDLY_NAME        "Sonoff"          // [FriendlyName] Friendlyname up to 32 characters used by webpages and Alexa
  #define USE_EMULATION                          // Enable Belkin WeMo and Hue Bridge emulation for Alexa (+11k code, +2k mem)
  #define EMULATION            EMUL_HUE         // [Emulation] Select Belkin WeMo (single relay/light) or Hue Bridge emulation (multi relay/light) (EMUL_NONE, EMUL_WEMO or EMUL_HUE)
`

I've setup my HA instance and it finds it and it appears as a switch controllable by mqtt, but yea, I kinda want Alexa to be able to control them (cos you know, voice is so much easier than pressing a switch...).

If anyone has any idea how to do this, the help would be much appreciated.

Courty40, 2017-07-10:
Try setting the emulation to 'Belkin WeMo', leave the Sonoff on for 20 mins (make sure its on the same network as the Alexa) and then go to the Alexa app, 'smart home' devices and hit discover, try a few times and the device will appear. once you have the device, set it up as a group (even if its only one), Groups appear a little more robust and you can add other devices later as well
I have three devices in two groups and it world great

CommodoreWhite, 2017-07-15:
On a related issue, I have electro dragon devices with two switches each. Can I give a friendly name to each so I can control each with Alexa?

smadds, 2017-07-28:
I'm also having a problem detecting. I'm using the latest 3.4 firmware and set emulation to "Hue Bridge multi devices", but Alexa doesn't see it. 
Do you need to have a Dot or Echo on the network? (my Alexa is on a Fire TV box)
I also notice that the Wemo skills are not available on amazon.co.uk (which my account is linked to). Would this affect it?

davidelang, 2017-07-29:
the latest firmware is 5.x, so if you are using 3.4 you need to upgrade to 
something more current.

smadds, 2017-08-01:
Sorry, no idea why I typed 3.4! It's 5.5, but not being detected

simo26246, 2017-08-01:
I fixed my issue by adding a genuine Dot to my setup and saying "alexa, discover my devices" (clicking discover in the app didn't work for me).

buserror, 2017-10-01:
I have the same problem. a Touch with wemo emulation isn't picked up by my Echo Dot. I have other emulated wemo switches via node-red (9 of them) that work fine... I tried stopping node-red and re-trying the discovery, no luck.
Is there a way to 'debug' wemo discovery? is that mdns?

zerog2k, 2017-11-13:
+1 - hue emulation works for me (alexa discovers), but wemo does not. Unsure how to debug further...

CODeRUS, 2018-01-14:
wemo emulation works for me only when changing `WEB_PORT` to some port from wemo: (49153, 49152, 49154, 49151, 49155)

buserror, 2018-05-15:
Well I tried fairly recently and it still didn't work -- I'm going to have another pop at it with a recent git later on. I also tried @CODeRUS trick and that didn't seem to work either...

buserror, 2018-05-21:
I tried again -- updated to 5.14, tried both Wemo and Hue Bridge, restarted, erased all devices in Echo page, rescan, and it doesn't find it.

Does anyone manage to get this to work? I mean, I installed Tasmota *just* for that feature really, and it never worked here.

buserror, 2018-07-05:
Well given the level of interest in this bug, I've switched to another firmware for my devices anyway, so so do close away unfixed bugs, bot...

luisfpinto, 2018-08-13:
Hello, 
I hope I'm not too late and this is an useful information.
You can control your tasmota devices with Alexa using the Yeti mobile app. 

https://getyeti.co/posts/controlling-your-sonoff-tasmota-with-yeti

Once you installed the app and you set up all your tasmota devices with Yeti, you only need to create charms and then install the Yeti Skill. In that guide in the "What's next section" you will see how to do that.

Let me know if you have any question or issue

ascillato, 2018-08-23:
@buserror 

The problem you are facing is explained in the wiki at: https://github.com/arendst/Sonoff-Tasmota/wiki/Theo's-Tasmota-Tips#20180523---relation-tasmota-and-esp8266arduino-core-version

the solution is at: https://github.com/arendst/Sonoff-Tasmota/issues/1639#issuecomment-360218536

Sorry that all that was missed in your search.

closed 2018-08-23 by ascillato2

slayrj, 2020-12-28:
I had the same problem and found that my Access Point had a broadcast filter, so as soon as I disabled it it found the tasmotas